v923z/micropython-ulab #389: median on empty array crashes

kevinjwalters, 2021-05-13:
`ulab.numerical.median()` on an empty `ndarray` crashes in CircuitPython 6.2.0. This occurred in an application I'm writing under certain conditions and it the CLUE board did the hard crash handler thing.

```python
Adafruit CircuitPython 6.2.0 on 2021-04-05; Adafruit Feather nRF52840 Express with nRF52840
>>> import ulab
>>> empty = ulab.array([])
>>> ulab.numerical.median(empty)
```

For comparison, numpy returns `nan`.

v923z, 2021-05-13:
@kevinjwalters This is one of those corner cases, on which you find absolutely no documentation. It is related to these two open issues: https://github.com/v923z/micropython-ulab/issues/304, and https://github.com/v923z/micropython-ulab/issues/301.

Is it a reasonable assumption that `numpy` functions called with empty arrays return a `nan`?

kevinjwalters, 2021-05-13:
I tested `numpy` and it gives `nan` which seems reasonable to me.

This was a real world case for me, I'm filtering an array for voltages above 50mV and I had a case where there were none above that value and array became zero length. My CLUE crashed when it did some stats on that array. I just had to reboot my desktop as it got confused over the usb devices due to those hard crashes!

v923z, 2021-05-13:
@kevinjwalters 
This is really a bit of a mess:

```python
def test_function(fun):
    print(fun([]))

test_function(std)
test_function(sum)
test_function(median)
test_function(sin)
```
prints 

```
nan
0.0
nan
[]
```

kevinjwalters, 2021-05-13:
```
Adafruit CircuitPython 6.2.0 on 2021-04-05; Adafruit CLUE nRF52840 Express with nRF52840
>>>
>>> import ulab
>>> ulab.numerical.mean(ulab.array([]))
1.08433e-19
```

!

v923z, 2021-05-13:
@kevinjwalters It is clear that this needs to be fixed, but I don't see at the moment, what kind of system `numpy` uses. `sum`, `median`, and `sin` return three different types, if you pass an empty array.

kevinjwalters, 2021-05-15:
Is that weird value out of mean a separate bug?

v923z, 2021-05-16:
> Is that weird value out of `mean` a separate bug?

No, I think we can keep it here. The treatment of empty arrays has to be overhauled, and I don't want to open a new issue for all functions that will be affected. I am going to do this in one sweep, as soon as I can figure out, how `numpy` behaves.

kevinjwalters, 2021-05-17:
`numpy.sin` is different to the rest as it iterates over the array and returns a value for each member. It's reasonable for that to return empty array if given an empty array.

v923z, 2021-05-18:
@kevinjwalters The latest code should not crash, if you initialise with an empty iterable. `mean`, `sum`, `std`, `sin` etc. also work. I will have to check out the `median`.

v923z, 2021-05-22:
@kevinjwalters https://github.com/v923z/micropython-ulab/pull/393 fixes the issue. If you are satisfied with the solution, I will create a new tag, so that `circuitpython` can include the changes in the nightly build.

kevinjwalters, 2021-05-24:
I had a quick glance and it looks ok. Are you using `sqrt` to get a `nan` value? I see this approach elsewhere in the code.

https://github.com/adafruit/circuitpython/blob/7df5d74d951f41347502087ef851c19e76c7a2f7/py/parsenum.c#L236

v923z, 2021-05-24:
> I had a quick glance and it looks ok. Are you using `sqrt` to get a `nan` value? 

What speaks against it?

v923z, 2021-05-24:
@kevinjwalters I have took `sqrt(-1.0)`, because the CI failed on mac with the gcc constant `NaN`. Curiously, while `sqrt(-1.0)` passed the CI checks, it created another problem, as indicated in https://github.com/v923z/micropython-ulab/issues/394, so I will try with your suggestion now.

v923z, 2021-06-04:
I believe, https://github.com/v923z/micropython-ulab/pull/395 is correct, so I am closing the issue.

closed 2021-06-04 by v923z