v923z/micropython-ulab #394: Can't build firmware

FrantisekPostl, 2021-05-23:
Hi, I'm on MacOs 11.1 , python 3.9.5. I'm trying to build Micropython with ulab 2.7.1 for Pico. I'm new to Python and Micropython. In last step - make USER_C_MODULES=/path/to/ulab/code/micropython.cmake I'm getting this error.

```
[  4%] Generating genhdr/qstr.i.last
In file included from /Users/frantisekpostl/projects/pico/micropython/py/obj.h:31,
                 from /Users/frantisekpostl/projects/pico/ulab/code/numpy/numerical/numerical.c:17:
/Users/frantisekpostl/projects/pico/ulab/code/numpy/numerical/numerical.c:993:83: error: pasting ")" and "F" does not give a valid preprocessing token
  993 |         return mp_obj_new_float(MICROPY_FLOAT_CONST(MICROPY_FLOAT_C_FUN(sqrt)(-1.0)));
      |                                                                                   ^
/Users/frantisekpostl/projects/pico/micropython/py/mpconfig.h:715:32: note: in definition of macro 'MICROPY_FLOAT_CONST'
  715 | #define MICROPY_FLOAT_CONST(x) x##F
      |                                ^
Traceback (most recent call last):
  File "/Users/frantisekpostl/projects/pico/micropython/py/makeqstrdefs.py", line 188, in <module>
    preprocess()
```

In readme is USER_C_MODULE, in build script is USER_C_MODULES.

khalidsgh, 2021-05-24:
Hi, i got the same error too while trying to build the firmware for ESP32.
My environment:
- Ubuntu 20.04
- python 3.6
- CMake 3.20

v923z, 2021-05-24:
@FrantisekPostl @khalidsgh Thanks for reporting the issue! I _think_ https://github.com/v923z/micropython-ulab/pull/395 should be a fix. Could you, please, check out the latest code, and try to compile the firmware?

FrantisekPostl, 2021-05-24:
Hi, it's working :), super fast fix. In firmware compilation steps for pico is probably typo, USER_C_MODULE should be USER_C_MODULES.

v923z, 2021-05-24:
@FrantisekPostl Thanks for the feedback!

> In firmware compilation steps for pico is probably typo, USER_C_MODULE should be USER_C_MODULES.

I will change the text. However, you can also just run the build script.

khalidsgh, 2021-05-25:
Hi @FrantisekPostl, it's working, thank you very much.

v923z, 2021-05-25:
@khalidsgh Thanks for the feedback! Since both cases seem to be resolved, I am closing the issue now. 
@FrantisekPostl

closed 2021-05-25 by v923z