v923z/micropython-ulab #399: Compile errors for TinyPICO on macOS

mdaeron, 2021-06-01:
Based on the instructions in the readme, I'm attempting to compile `micropython` + `ulab` for a TinyPICO, from macOS Catalina (10.15.7), using the following script:

````
#!/bin/bash

export BUILD_DIR=$(pwd)

git clone https://github.com/v923z/micropython-ulab.git ulab
git clone https://github.com/micropython/micropython.git

cd $BUILD_DIR/micropython/

git clone -b v4.0.2 --recursive https://github.com/espressif/esp-idf.git

cd esp-idf
./install.sh
. ./export.sh

cd $BUILD_DIR/micropython/mpy-cross
make
cd $BUILD_DIR/micropython/ports/esp32
make submodules

cd $BUILD_DIR/micropython/ports/esp32
echo 'BOARD = UM_TINYPICO' > /tmp/mf.txt
echo 'USER_C_MODULES = $(BUILD_DIR)/ulab/code/micropython.cmake' >> /tmp/mf.txt
cat Makefile >> /tmp/mf.txt
mv /tmp/mf.txt Makefile

make
````

This fails with the following errors:

````
[ 97%] Generating ../../genhdr/mpversion.h
[ 97%] Generating ../../frozen_content.c
[ 97%] Building C object esp-idf/main/CMakeFiles/__idf_main.dir/__/__/frozen_content.c.obj
[ 97%] Linking CXX static library libmain.a
[ 97%] Built target __idf_main
[ 97%] Generating ld/esp32.project.ld
Traceback (most recent call last):
  File "/Users/mathieu/ShareBox/μPy/firmware/tinypico+ulab+i2s/micropython/esp-idf/tools/ldgen/generation.py", line 592, in add_sections_info
    results = parser.parseString(first_line)
  File "/Users/mathieu/.espressif/python_env/idf4.0_py3.7_env/lib/python3.7/site-packages/pyparsing.py", line 1814, in parseString
    raise exc
  File "/Users/mathieu/.espressif/python_env/idf4.0_py3.7_env/lib/python3.7/site-packages/pyparsing.py", line 1804, in parseString
    loc, tokens = self._parse( instring, 0 )
  File "/Users/mathieu/.espressif/python_env/idf4.0_py3.7_env/lib/python3.7/site-packages/pyparsing.py", line 1548, in _parseNoCache
    loc,tokens = self.parseImpl( instring, preloc, doActions )
  File "/Users/mathieu/.espressif/python_env/idf4.0_py3.7_env/lib/python3.7/site-packages/pyparsing.py", line 3722, in parseImpl
    loc, exprtokens = e._parse( instring, loc, doActions )
  File "/Users/mathieu/.espressif/python_env/idf4.0_py3.7_env/lib/python3.7/site-packages/pyparsing.py", line 1552, in _parseNoCache
    loc,tokens = self.parseImpl( instring, preloc, doActions )
  File "/Users/mathieu/.espressif/python_env/idf4.0_py3.7_env/lib/python3.7/site-packages/pyparsing.py", line 3472, in parseImpl
    raise ParseException(instring, loc, self.errmsg, self)
pyparsing.ParseException: Expected end of line (at char 35), (line:1, col:36)

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/Users/mathieu/ShareBox/μPy/firmware/tinypico+ulab+i2s/micropython/esp-idf/tools/ldgen/ldgen.py", line 150, in <module>
    main()
  File "/Users/mathieu/ShareBox/μPy/firmware/tinypico+ulab+i2s/micropython/esp-idf/tools/ldgen/ldgen.py", line 108, in main
    sections_infos.add_sections_info(dump)
  File "/Users/mathieu/ShareBox/μPy/firmware/tinypico+ulab+i2s/micropython/esp-idf/tools/ldgen/generation.py", line 594, in add_sections_info
    raise ParseException("Parsing sections info for library " + sections_info_dump.name + " failed. " + p.message)
  File "/Users/mathieu/.espressif/python_env/idf4.0_py3.7_env/lib/python3.7/site-packages/pyparsing.py", line 272, in __getattr__
    raise AttributeError(aname)
AttributeError: message
make[3]: *** [esp-idf/esp32/ld/esp32.project.ld] Error 1
make[2]: *** [esp-idf/esp32/CMakeFiles/__ldgen_output_esp32.project.ld.dir/all] Error 2
make[1]: *** [all] Error 2
make failed with exit code 2
make: *** [all] Error 2
````

Is there anything obvious I'm missing?

mdaeron, 2021-06-01:
@v923z, you're right, leaving `ulab` out of the script results in the same error. I'll try to understand what's happening, sorry for the noise.

v923z, 2021-06-01:
> leaving `ulab` out of the script results in the same error. I'll try to understand what's happening, sorry for the noise.

No problem.

v923z, 2021-06-01:
@teuler Thomas, do you see similar errors on windows/linux? We might have to update the build script.

v923z, 2021-06-19:
@mdaeron 

There doesn't seem to be too much activity here, so I am closing the issue. Open a new one, if you still face difficulties.

closed 2021-06-19 by v923z