v923z/micropython-ulab #405: [FEATURE REQUEST]scipy-compatibility of cho_solve

v923z, 2021-06-03:
@vikas-udupa 

Vikas, according to the documentation, https://docs.scipy.org/doc/scipy/reference/generated/scipy.linalg.cho_solve.html, the first argument of `cho_solve` is a tuple, while https://github.com/v923z/micropython-ulab/pull/391 implements it with a single matrix. I overlooked the issue at the time, but I think I would like to have `scipy`-compatibility. Do you want to look into this, or should I?

vikas-udupa, 2021-06-08:
For the moment, I implemented `cho_solve` such that it is compatible with `numpy.linalg.cholesky` function that has already been implemented. But in the long run, it is right to follow what is given in the documentation, just like you mentioned.
However, I am unable to work on it at the moment due to other commitments. It would be great if you could go ahead and fix this. If not, I can fix this after a couple of weeks.

v923z, 2021-06-08:
@vikas-udupa OK, thanks for the feedback! I might take a stab at it.