v923z/micropython-ulab #409: [FEATURE REQUEST] Support for 3- and 4-D arrays

JamesTev, 2021-06-11:
It would be very helpful to allow support for higher order tensors (3, 4 dimensions). Has this functionality been introduced since v2.6.0? I'm using 2.6.0 and there is currently only 2D support. 

```maximum number of dimensions:  2.6.0-2D```

I may have missed it but haven't spotted any changes in more recent releases.

v923z, 2021-06-11:
@JamesTev 4 dimensions are supported. You can set it in https://github.com/v923z/micropython-ulab/blob/161a7288482200c884623ed3cb59dd91f53be91a/code/ulab.h#L41-L43, as described in https://micropython-ulab.readthedocs.io/en/latest/ulab-intro.html#customising-the-firmware.

JamesTev, 2021-06-11:
Oh, great. Sorry about that!

closed 2021-06-11 by JamesTev

v923z, 2021-06-11:
@JamesTev Absolutely no problem.