v923z/micropython-ulab #410: [BUG] scalar-matrix multiplication is not commutative

JamesTev, 2021-06-11:
**Description**

When pre-multiplying a matrix by a scalar, a `TypeError` is raised. Post-multiplication works as expected.

**To Reproduce**

```python
from ulab import numpy as np
print(ulab.__version__)

k = 2

print(np.eye(2)*k) # post-multiplication works fine
print(k*np.eye(2)) # pre-mult. raises an error
```
Outputs: 
```
2.6.0-2D
array([[2.0, 0.0],
       [0.0, 2.0]], dtype=float32)
Traceback (most recent call last):
  File "<stdin>", line 7, in <module>
TypeError: unsupported types for __mul__: 'int', 'ndarray'
```

**Expected behaviour**
```
2.6.0-2D
array([[2.0, 0.0],
       [0.0, 2.0]], dtype=float32)
array([[2.0, 0.0],
       [0.0, 2.0]], dtype=float32)
```

v923z, 2021-06-11:
@JamesTev I don't see this. Here is my output:

```python
>>> from ulab import numpy as np
>>> import ulab
>>> k = 2

>>> print(np.eye(2) * k)
>>> print(k * np.eye(2))
>>> print(ulab.__version__)

array([[2.0, 0.0],
       [0.0, 2.0]], dtype=float64)
array([[2.0, 0.0],
       [0.0, 2.0]], dtype=float64)
3.0.0-2D
```

JamesTev, 2021-06-11:
Strange... must be a version issue. I will try updating to the latest firmware.

v923z, 2021-06-19:
@JamesTev Is this still a valid issue, or we can close it?

v923z, 2022-02-01:
I guess, this issue can be closed now.

closed 2022-02-01 by v923z