v923z/micropython-ulab #413: [BUG]

thalesmaoa, 2021-06-22:
I can't compile ulab for esp32.

```bash
$ make submodules & make all
[1] 13182
Use make V=1 or set BUILD_VERBOSE in your environment to increase build verbosity.
Including User C Module from /home/thales/Downloads/ulab-ESP32/ulab/code
Use make V=1 or set BUILD_VERBOSE in your environment to increase build verbosity.
Including User C Module from /home/thales/Downloads/ulab-ESP32/ulab/code
Building with ESP IDF v3
Building with ESP IDF v3
Updating submodules: lib/berkeley-db-1.xx
CC /home/thales/Downloads/ulab-ESP32/ulab/code/scipy/linalg/linalg.c
/home/thales/Downloads/ulab-ESP32/ulab/code/scipy/linalg/linalg.c: In function 'solve_triangular':
/home/thales/Downloads/ulab-ESP32/ulab/code/scipy/linalg/linalg.c:85:57: error: implicit conversion from 'float' to 'double' to match other operand of binary expression [-Werror=double-promotion]
         if (MICROPY_FLOAT_C_FUN(fabs)(get_A_ele(A_arr)) < TOLERANCE)
                                                         ^
cc1: all warnings being treated as errors
../../py/mkrules.mk:77: recipe for target 'build-GENERIC/code/scipy/linalg/linalg.o' failed
make: *** [build-GENERIC/code/scipy/linalg/linalg.o] Error 1
[1]+  Concluído              make submodules

```

I'm using the following procedure:

```
#!/bin/bash

export BUILD_DIR=$(pwd)

git clone https://github.com/v923z/micropython-ulab.git ulab
git clone https://github.com/micropython/micropython.git

cd $BUILD_DIR/micropython/
git checkout tags/v1.14

git submodule update --init
cd ./mpy-cross && make # build cross-compiler (required)

cd $BUILD_DIR/micropython/ports/esp32
make ESPIDF= # will display supported ESP-IDF commit hashes
# output should look like: """
# ...
# Supported git hash (v3.3): 9e70825d1e1cbf7988cf36981774300066580ea7
# Supported git hash (v4.0) (experimental): 4c81978a3e2220674a432a588292a4c860eef27b

ESPIDF_VER=9e70825d1e1cbf7988cf36981774300066580ea7

mkdir $BUILD_DIR/micropython/esp32

cd $BUILD_DIR/micropython/esp32
git clone https://github.com/espressif/esp-idf.git esp-idf
cd $BUILD_DIR/micropython/esp32/esp-idf
git checkout $ESPIDF_VER
git submodule update --init --recursive # get idf submodules
pip install -r ./requirements.txt # install python reqs

curl https://dl.espressif.com/dl/xtensa-esp32-elf-linux64-1.22.0-80-g6c4433a-5.2.0.tar.gz | tar xvz

cd $BUILD_DIR/micropython/ports/esp32
# temporarily add esp32 compiler to path
export PATH=$BUILD_DIR/micropython/esp32/esp-idf/xtensa-esp32-elf/bine:$PATH
export ESPIDF=$BUILD_DIR/micropython/esp32/esp-idf
export BOARD=GENERIC # board options are in ./board
export USER_C_MODULES=$BUILD_DIR/ulab # include ulab in firmware

make submodules & make all
```

I was able to compile for PYBOARD without a problem. Maybe I should upgrade cmake to use it?

v923z, 2021-06-22:
Have you tried https://github.com/v923z/micropython-ulab/blob/master/build/esp32-cmake.sh?

v923z, 2021-06-22:
@thalesmaoa  Thanks for reporting the issue! I believe, I found the root of the problem. Could you, please, check out https://github.com/v923z/micropython-ulab/pull/414 ?

thalesmaoa, 2021-06-22:
> Have you tried https://github.com/v923z/micropython-ulab/blob/master/build/esp32-cmake.sh?

Yes. I also suggest an update.

```
#!/bin/bash

export BUILD_DIR=$(pwd)

git clone https://github.com/v923z/micropython-ulab.git 
git clone https://github.com/micropython/micropython.git

cd $BUILD_DIR/micropython/

git clone -b v4.0.2 --recursive https://github.com/espressif/esp-idf.git
  
cd esp-idf
git checkout v4.2
git submodule update --init --recursive

./install.sh
source ./export.sh # Not sure, perhaps is just . ./export.sh

cd $BUILD_DIR/micropython/mpy-cross
make -C
cd $BUILD_DIR/micropython/ports/esp32
make submodules

echo "BOARD = GENERIC" > $BUILD_DIR/micropython/ports/esp32/makefile
echo "USER_C_MODULES = \$(BUILD_DIR)/ulab/code/micropython.cmake" >> $BUILD_DIR/micropython/ports/esp32/makefile
echo -e "\n\ninclude Makefile" >> $BUILD_DIR/micropython/ports/esp32/makefile

cd $BUILD_DIR/micropython/ports/esp32

make
```

However, I needed to update cmake. I will try again and let you know.

thalesmaoa, 2021-06-22:
> @thalesmaoa Thanks for reporting the issue! I believe, I found the root of the problem. Could you, please, check out #414 ?

I confirm that this branch fix the problem. I will try with cmake now.

However, I do have a question. Why it did compile for pyboard?

thalesmaoa, 2021-06-22:
I confirm that cmake work as well.

```
#!/bin/bash
export BUILD_DIR=$(pwd)

git clone https://github.com/v923z/micropython-ulab.git 
git clone https://github.com/micropython/micropython.git

cd $BUILD_DIR/micropython/

git clone -b v4.0.2 --recursive https://github.com/espressif/esp-idf.git
  
cd esp-idf
git checkout v4.2
git submodule update --init --recursive

./install.sh
. ./export.sh

cd $BUILD_DIR/micropython/mpy-cross
make
cd $BUILD_DIR/micropython/ports/esp32
make submodules

echo "BOARD = GENERIC" > $BUILD_DIR/micropython/ports/esp32/makefile
echo "USER_C_MODULES = \$(BUILD_DIR)/micropython-ulab/code/micropython.cmake" >> $BUILD_DIR/micropython/ports/esp32/makefile
echo -e "\n\ninclude Makefile" >> $BUILD_DIR/micropython/ports/esp32/makefile

cd $BUILD_DIR/micropython/ports/esp32

make
```

```
#!/bin/bash

export BUILD_DIR=$(pwd)

git clone https://github.com/v923z/micropython-ulab.git ulab
git clone https://github.com/micropython/micropython.git

cd $BUILD_DIR/micropython/
git checkout tags/v1.14

git submodule update --init
cd ./mpy-cross && make # build cross-compiler (required)

cd $BUILD_DIR/micropython/ports/esp32
make ESPIDF= # will display supported ESP-IDF commit hashes
# output should look like: """
# ...
# Supported git hash (v3.3): 9e70825d1e1cbf7988cf36981774300066580ea7
# Supported git hash (v4.0) (experimental): 4c81978a3e2220674a432a588292a4c860eef27b

ESPIDF_VER=9e70825d1e1cbf7988cf36981774300066580ea7

mkdir $BUILD_DIR/micropython/esp32

cd $BUILD_DIR/micropython/esp32
git clone https://github.com/espressif/esp-idf.git esp-idf
cd $BUILD_DIR/micropython/esp32/esp-idf
git checkout $ESPIDF_VER
git submodule update --init --recursive # get idf submodules
pip install -r ./requirements.txt # install python reqs

curl https://dl.espressif.com/dl/xtensa-esp32-elf-linux64-1.22.0-80-g6c4433a-5.2.0.tar.gz | tar xvz

cd $BUILD_DIR/micropython/ports/esp32
# temporarily add esp32 compiler to path
export PATH=$BUILD_DIR/micropython/esp32/esp-idf/xtensa-esp32-elf/bin:$PATH
export ESPIDF=$BUILD_DIR/micropython/esp32/esp-idf
export BOARD=GENERIC # board options are in ./board
export USER_C_MODULES=$BUILD_DIR/ulab # include ulab in firmware

make submodules & make all
```

v923z, 2021-06-22:
> However, I do have a question. Why it did compile for pyboard?

I know only that the ESP compiler is a bit pickier about the floating point numbers.

v923z, 2021-06-22:
> I confirm that cmake work as well.

@thalesmaoa Many thanks for the testing! 

Do you want to make a PR with your changes? People who do so always get brownie points: https://github.com/v923z/micropython-ulab/graphs/contributors

thalesmaoa, 2021-06-23:
Thanks. I will probably do.

closed 2021-06-23 by thalesmaoa