v923z/micropython-ulab #416: [BUG] ESP32 keep reseting

thalesmaoa, 2021-06-25:
**BUG**
After compiling and flash ESP32 i get the following error:

```
rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
```
and the board keeps reseting itself.

According to https://github.com/espressif/esp-idf/issues/113#issuecomment-268698113 it seems to be a wrong flash encryption settings.

v923z, 2021-06-25:
@thalesmaoa Does the section https://github.com/v923z/micropython-ulab#what-to-do-if-the-firmware-is-too-large help? What you can also do is exclude functions from the firmware https://micropython-ulab.readthedocs.io/en/latest/ulab-intro.html#customising-the-firmware till it is small enough to fit on the flash. 

In the long term, I think the fix must come from espressif, because this is not a `ulab` issue. So, you could just try to upvote the issue that you linked to.

thalesmaoa, 2021-06-26:
@v923z, I'm not an expert, and I really want to get involved with your project, but I still have a lot to lean.
Small flash size is an option, but I'm confused. I'm using a TTGO ESP32 16MB. Besides, the firmware that I compiled the end of last year works perfectly. I was just trying to updated it. I follow the same procedure to compile.

v923z, 2021-06-26:
> Small flash size is an option, but I'm confused. I'm using a TTGO ESP32 16MB. Besides, the firmware that I compiled the end of last year works perfectly. I was just trying to updated it. I follow the same procedure to compile.

What I meant is that the error https://github.com/espressif/esp-idf/issues/113#issuecomment-268698113 does not seem to involve `ulab`, it seems to be a somewhat generic problem. From the comments, I could find out, whether it has been fixed (I see that the issue is closed, but that might not mean anything). I am more than happy to modify the code, so that it can be compiled for ESP32, but I would need feedback.

thalesmaoa, 2021-07-05:
Probably Espressif problem as you mentioned. I was able to compile and run it today without a problem.

closed 2021-07-05 by thalesmaoa

v923z, 2021-07-05:
Thanks for the feedback!