v923z/micropython-ulab #417: THD feature request

thalesmaoa, 2021-06-29:
Hi there, I'm not sure if this is the clever way to perform it, but I'm measuring sine from AC grid. I calculate RMS and THD

My feeling is that I'm allocating to much memory
```
def rms(array):
    return np.sqrt(np.mean(array ** 2))

def THD(array):
    """Perform FFT"""
    L = len(array)
    R, I = np.fft.fft(array)
    P2 = np.array( [0]*L )
    for i in range(L):
        P2[i] = np.sqrt(R[i]**2 + I[i]**2)/L
    P1 = P2[:L//2+1]*2
    THD = 100*np.sqrt( np.sum(P1**2) - np.max(P1)**2 ) / np.max(P1)
    return THD
```

v923z, 2021-06-30:
> My feeling is that I'm allocating to much memory

You don't need to calculate the FFT, and initialise the array (`P2`), but should replace this 

```
    L = len(array)
    R, I = np.fft.fft(array)
    P2 = np.array( [0]*L )
     for i in range(L):
        P2[i] = np.sqrt(R[i]**2 + I[i]**2)/L
```

by a single call to `scipy.signal.spectrogram`: https://micropython-ulab.readthedocs.io/en/latest/scipy-signal.html#spectrogram 

If there is a similar `numpy/scipy` function, we could think about implementing it, though, I don't think it would be necessary.

thalesmaoa, 2021-07-01:
I could only found variations, but not an implementation. spectrogram simplified my allocation. Thank you

v923z, 2021-07-01:
@thalesmaoa OK, thanks for the feedback!

thalesmaoa, 2021-07-02:
@v923z, sorry to open. After changing to `scipy.signal.spectrogram` I'm having more allocation errors:

```
MemoryError: memory allocation failed, allocating 4096 bytes
```

This not a bug, neither a request. Not sure if here is the correct place to ask for help.

I've tried pre-allocating, but none worked.

I work with three phase ac. I receive the data over UART, fill three arrays (Va, Vb, Vc). After that I convert them to `np.array` to perform RMS, THD and publish over MQTT.

It, sometime, complain about:

```
def THD(array):
    """Calculate the THD"""
    # global Y
    Y = spy.signal.spectrogram(array)

    # Fundamental 60Hz
    Vfun = 2*Y[20]/size_ringbuffer
    
    # Sum all the amplitudes 
    h_max = 25 # Max harmonic order
    sum_abs = 0
    for i_h in np.arange(40,20*(h_max+1),20):
        sum_abs += (Y[i_h]/size_ringbuffer)*2

    THD = 100 * np.sqrt( sum_abs ) / Vfun
    return THD
``` 
```
Traceback (most recent call last):
  File "main.py", line 226, in <module>
  File "uasyncio/core.py", line 1, in run
  File "uasyncio/core.py", line 1, in run_until_complete
  File "uasyncio/core.py", line 1, in run_until_complete
  File "main.py", line 43, in main
  File "main.py", line 198, in THD
MemoryError: memory allocation failed, allocating 4096 bytes
MicroPython v1.14 on 2021-03-21; ESP32 module with ESP32
```
I know that memory allocation is difficult to debug, but I'm not sure even about how to start.

v923z, 2021-07-02:
> @v923z, sorry to open. After changing to `scipy.signal.spectrogram` I'm having more allocation errors:
> 
> ```
> MemoryError: memory allocation failed, allocating 4096 bytes
> ```
> 
> This not a bug, neither a request. Not sure if here is the correct place to ask for help.

You could try the forum.micropython.org/, there are always people who are glad to help, and know more about the subject than myself. 

However, I still have a suggestion: 

```
def THD(array):
    """Calculate the THD"""
    # global Y
    Y = None
    gc.collect()
    Y = spy.signal.spectrogram(array)

    # Fundamental 60Hz
    Vfun = 2*Y[20]/size_ringbuffer
    
    # Sum all the amplitudes 
    h_max = 25 # Max harmonic order
    sum_abs = 0
    for i_h in np.arange(40,20*(h_max+1),20):
        sum_abs += (Y[i_h]/size_ringbuffer)*2

    THD = 100 * np.sqrt( sum_abs ) / Vfun
    return THD
```

> I know that memory allocation is difficult to debug, but I'm not sure even about how to start.

I would also recommend Jim's excellent exposition on how the garbage collector in `micropython` works. https://www.youtube.com/watch?v=H_xq8IYjh2w

I hope this helps.

thalesmaoa, 2021-07-05:
Thanks @v923z. I've went through every single doc and video that I could find over this weekend. I've done a lot of code optimization. However, I did not perform the same task for `mqtt_as.py`. 
The problem did persist.
I've changed to fft again and I got no more error with memory.

I will let it for a while and let you know:

```
def THD(array):
    """Calculate the THD"""
    R = None
    I = None
    gc.collect()
    R, I = np.fft.fft(array)
    Vfun = 2*np.sqrt( R[20]**2 + I[20]**2 )/sr
    # print('THD RAM free {} alloc {}'.format(gc.mem_free(), gc.mem_alloc()))
    h_max = 25
    sum_abs = 0
    for i_h in np.arange(40,20*(h_max+1),20):
        sum_abs += ( np.sqrt( R[i_h]**2 + I[i_h]**2 ) /sr)*2

    return 100 * np.sqrt( sum_abs ) / Vfun
```

v923z, 2021-07-05:
> Thanks @v923z. I've went through every single doc and video that I could find over this weekend. I've done a lot of code optimization. However, I did not perform the same task for `mqtt_as.py`.
> The problem did persist.
> I've changed to fft again and I got no more error with memory.

I find this  a strange, because `fft`, and `spectrometer` call the same underlying function. I will look into it.

thalesmaoa, 2021-07-05:
I'm not sure if this is a ulab problem as well. 

```
Calibra RAM free 55216 alloc 55952
Calibra RAM free 42304 alloc 68864
Calibra RAM free 42464 alloc 68704
Calibra RAM free 55152 alloc 56016
Calibra RAM free 42240 alloc 68928
Calibra RAM free 38256 alloc 72912
Traceback (most recent call last):
  File "main.py", line 260, in <module>
  File "uasyncio/core.py", line 1, in run
  File "uasyncio/core.py", line 1, in run_until_complete
  File "uasyncio/core.py", line 1, in run_until_complete
  File "main.py", line 21, in main
MemoryError: memory allocation failed, allocating 4096 bytes
MicroPython v1.14 on 2021-07-05; ESP32 module with ESP32
Type "help()" for more information.
```
Looks my heap is totally mess up. I have a lot of room, but can't alloc 4096. Probably ESP32.

v923z, 2021-07-05:
> I'm not sure if this is a ulab problem as well.
>
> ```
> Calibra RAM free 55216 alloc 55952
> Calibra RAM free 42304 alloc 68864
> Calibra RAM free 42464 alloc 68704
> Calibra RAM free 55152 alloc 56016 [...]

Is the call to `fft` in `main.py`?

thalesmaoa, 2021-07-06:
No. It indicates line 21 in main, only because I've put @micropython.native decoration in my function. I was trying different things.

v923z, 2021-07-07:
> No. It indicates line 21 in main, only because I've put @micropython.native decoration in my function. I was trying different things.

OK, but I think we should try to produce a minimal version of the code that still triggers the error. Otherwise, we really don't stand a chance.

thalesmaoa, 2021-07-07:
I've opened a question in the forum:
https://forum.micropython.org/viewtopic.php?f=2&t=10786

Jimmo suggestion improved the performance, however it is still crashing.
I will wait his reply. Thank you.

v923z, 2022-01-02:
@thalesmaoa Is this issue still relevant, or we can close it?

closed 2022-01-02 by thalesmaoa