v923z/micropython-ulab #419: [BUG] CircuitPython documentation appears in wrong location (ulab.X vs ulab.numpy.X)

jepler, 2021-07-08:
**Describe the bug**
In CircuitPython main branch, which is based off 161a7288482200c884623ed3cb59dd91f53be91a (3.0.0), many items appear in the wrong location in the documentation.  For instance, the documentation shows that the `array` constructor is at `ulab.array` rather than `ulab.numpy.array`.

**To Reproduce**
View the documentation on [readthedocs](https://circuitpython.readthedocs.io/en/latest/shared-bindings/ulab/index.html#ulab.array) or use `./build-cp.sh` locally and look at the generated docs

**Expected behavior**
Items are organized in the documentation as they are in the implementation

v923z, 2021-07-08:
Is there an easy way to generate a comparison between what it is, and what it should be?

jepler, 2021-07-09:
No easy way to generate lists like that comes to mind.

I just know that almost nothing is directly in `ulab` anymore, and almost everything is down in submodules like `ulab.numpy`.

v923z, 2021-07-09:
I don't want to combine everything that doesn't belong to sub-modules into a single file, but would something like this work?

```
ulab
|----numpy
|    |----poly
|    |     |----poly.c
|    |     |----poly.h
|    |----numerical
|    |     |----numerical.c
|    |     |----numerical.h
|    |----transform
|    |     |----transform.c
|    |     |----transform.h
|    |----approx
|    |     |----approx.c
|    |     |----approx.h
|    |----modules
|         |----fft
|         |     |----fft.c
|         |     |----fft.h
|         |----linalg
|               |----linalg.c
|               |----linalg.h
|
|----scipy
|    |----modules
|         |----signal
|         |----linalg
|         |----optimize
|
|----utils
|
|----user
```
An alternative could be 

```
ulab
|----numpy
|    |----core
|    |     |----poly.c
|    |     |----poly.h
|    |     |----transform.c
|    |     |----transform.h
|    |     |----approx.c
|    |     |----approx.h
|    |----modules
|         |----fft
|         |     |----fft.c
|         |     |----fft.h
|         |----linalg
|               |----linalg.c
|               |----linalg.h
|
|----scipy
|    |----modules
|         |----signal
|         |----linalg
|         |----optimize
|
|----utils
|
|----user
```
i.e., we could dump all files that do not belong to a particular module into `numpy/core`, and then the doc script would have to treat everything that is under `numpy/modules`, and `scipy/modules` as modules, and the stubs from the rest could be combined. Or is the problem that, if we do this, the functions are still not alphabetically ordered?

I am sure we can find a solution here, you just have to tell what is reasonable as far as the docs generation is concerned.

jepler, 2021-07-10:
Suppose that you have some source file `micropython-ulab/code/A/B/Z.c` that contains some documentation prefixed by `//|`.  As far as I know, that is going to be documented as being in module ulab.A.B. This pretty much dictates the folder/directory structure required. I'm pretty sure that means that intermediate folder names like "modules" and "core" will give undesired results.

v923z, 2021-07-10:
> Suppose that you have some source file `micropython-ulab/code/A/B/Z.c` that contains some documentation prefixed by `//|`. As far as I know, that is going to be documented as being in module [...]

I think the main difficulty is that most of the functions are simply in top-level `numpy`, and I can't possibly imagine that we put all those functions in a single file. But we could do the following: we can write a single script that strips all stubs from the source, and puts them into separate files under `docs/`, and then you generate the documentation from there. I believe, this is a relatively clean solution, and would keep the actual documentation in the source. I think that was Scott's request.

jepler, 2021-07-10:
Notice how the filename itself (`Z.c`) doesn't figure in the location of the documentation (`ulab.A.B`), so the general idea that CircuitPython follows is to split a single Python _module_ into _files_ that make sense, but they all reside in the same _folder_.

v923z, 2021-07-10:
OK, so you are saying that 

```
ulab
|----numpy
|    |----poly.c
|    |----poly.h
|    |----transform.c
|    |----transform.h
|    |----approx.c
|    |----approx.h
|    |----fft
|         |----fft.c
|         |----fft.h
|    |----linalg
|          |----linalg.c
|          |----linalg.h
|
|----scipy
|     |----signal
|     |----linalg
|     |----optimize
|
|----utils
|
|----user
```
should work, right? I think this would be an acceptable change.

jepler, 2021-07-10:
I think so, and if I understand what CircuitPython is doing correctly.  Do you want to start by just re-organizing a few files and make sure the items move as I have said they would?

v923z, 2021-07-10:
> Do you want to start by just re-organizing a few files and make sure the items move as I have said they would?

I can do that, no problem. I can push this up on my priority list, but I would still need a couple of days.

jepler, 2021-07-10:
Absolutely fine😀

v923z, 2021-07-14:
@jepler I have taken a stab at it: https://github.com/v923z/micropython-ulab/pull/423 The firmware can be compiled, but I get a strange error, when generating the docs for `circuitpython`. I can look into it tomorrow, if you don't have time.

PR #426 (2021-07-22): Circuitpython doc move
This builds on #423 and also works together with https://github.com/adafruit/circuitpython/pull/5028 to complete the doc move. (If we take this PR, then #423 is moot)

However, this build will come up in red for the circuitpython builds because without https://github.com/adafruit/circuitpython/pull/5028 (which renames some use of ulab.ndarray to ulab.numpy.ndarray within our own pyi/docs) it is still not OK.

I didn't find a way for both things to work together, so the possibility I see is to (approximately in order):

- [ ] verify that https://github.com/adafruit/circui [... cut]
closed 2021-07-22 by jepler