v923z/micropython-ulab #420: Q: How to iterate over all elements of a 2D matrix in unspecified order

nsajko, 2021-07-08:
I have no experience with ulab (nor numpy, for that matter), and I couldn't find a discussion forum dedicated to ulab, so i hope it's OK to ask the question here.

So I have an ndarray like this:

```
a = np.array([[1, 2],[7, 5]], dtype = np.float)
```

and I want to iterate over all of its elements, so over the set `{1, 2, 5, 7}`.

I think that in numpy one would use `a.flat` for that purpose, but that doesn't seem to be available in ulab.

What is the intended way to solve this problem?

v923z, 2021-07-08:
> I have no experience with ulab (nor numpy, for that matter), and I couldn't find a discussion forum dedicated to ulab, so i hope it's OK to ask the question here.

It's OK. 

> 
> So I have an ndarray like this:
> 
> ```
> a = np.array([[1, 2],[7, 5]], dtype = np.float)
> ```
> 
> and I want to iterate over all of its elements, so over the set `{1, 2, 5, 7}`.
> 
> I think that in numpy one would use `a.flat` for that purpose, but that doesn't seem to be available in ulab.

You have at least two options. One is the [.flatten method](https://micropython-ulab.readthedocs.io/en/latest/ulab-ndarray.html?highlight=flatten#flatten), the other is [.reshape](https://micropython-ulab.readthedocs.io/en/latest/ulab-ndarray.html?highlight=flatten#reshape).

nsajko, 2021-07-08:
It's very possible I'm wrong, but I was under the impression that `.flat` avoids allocations that would happen with `.flatten`?

Is there a way to iterate over all elements of a matrix in a single loop, but without copying the backing storage of the matrix?

v923z, 2021-07-08:
> It's very possible I'm wrong, but I was under the impression that `.flat` avoids allocations that would happen with `.flatten`?
> 
> Is there a way to iterate over all elements of a matrix in a single loop, but without copying the backing storage of the matrix?

`.reshape` does not copy anything, if the matrix is dense. It will return a view of the underlying data. 

But, if you really want to have a flat iterator, it is definitely doable, and it's not even that hard. If you are not happy with `.reshape`, I could add the flat iterator in a week or so, perhaps, even sooner.

nsajko, 2021-07-08:
Don't do it just for me :)

Thank you, I will look into reshape, then.

v923z, 2021-07-08:
> Don't do it just for me :)

The issue of the flat iterator has come up in the past, and I think it is a useful feature. It just somehow slipped off my TODO list.

v923z, 2021-07-13:
@nsajko Do you think you could check out https://github.com/v923z/micropython-ulab/pull/422, and see if it is OK. I haven't implemented slicing and indexing, but you can do something like this: https://github.com/v923z/micropython-ulab/blob/flatiter/docs/manual/source/ulab-ndarray.rst#flat .

nsajko, 2021-07-13:
Judging by the documentation that you pointed to, this is perfect for me, but I won't be trying this out just yet, as I'm actually using Micropython as part of the Omega calculator (but note that the change that adds Ulab to Omega actually isn't merged yet: [omega#547](https://github.com/Omega-Numworks/Omega/pull/547)).

v923z, 2021-07-13:
Then would it be OK, if we merged this? We can always extend this later, if you need slicing or indexing.

v923z, 2021-07-14:
https://github.com/v923z/micropython-ulab/pull/422 closes the issue.

closed 2021-07-14 by v923z