v923z/micropython-ulab #424: [BUG] Build error with ATSAMD on MicroPython.

water5, 2021-07-15:
micropython/ports/samd/ # `make BOARD=ADAFRUIT_ITSYBITSY_M4_EXPRESS USER_C_MODULES=../../../ulab all`

> Use make V=1 or set BUILD_VERBOSE in your environment to increase build verbosity.
> Including User C Module from ../../../ulab/code
> mkdir -p build-ADAFRUIT_ITSYBITSY_M4_EXPRESS/genhdr
> GEN build-ADAFRUIT_ITSYBITSY_M4_EXPRESS/genhdr/mpversion.h
> GEN build-ADAFRUIT_ITSYBITSY_M4_EXPRESS/genhdr/moduledefs.h
> GEN build-ADAFRUIT_ITSYBITSY_M4_EXPRESS/genhdr/qstr.i.last
> In file included from ../../../ulab/code/numpy/ndarray/ndarray_iter.c:20:
../../../ulab/code/numpy/ndarray/ndarray_iter.h:21:10: fatal error: ulab.h: No such file or directory
   21 | #include "ulab.h"
      |          ^~~~~~~~
> compilation terminated.
> In file included from ../../../ulab/code/ndarray_properties.c:22:
../../../ulab/code/numpy/ndarray/ndarray_iter.h:21:10: fatal error: ulab.h: No such file or directory
   21 | #include "ulab.h"
      |          ^~~~~~~~
> compilation terminated.
> In file included from ../../../ulab/code/ndarray_properties.h:23,
                 from ../../../ulab/code/ulab.c:25:
../../../ulab/code/numpy/ndarray/ndarray_iter.h:21:10: fatal error: ulab.h: No such file or directory
   21 | #include "ulab.h"
      |          ^~~~~~~~
> compilation terminated.

MicroPython 1.16 (2021-07-16)

I use same environment to build STM port passed, although ATSAMD is most new port on MicroPython, I curious what's wrong happened, ATSAMD port need different build configure for `ulab`?

v923z, 2021-07-16:
@water5 Thanks for reporting the issue. Could you, please, check out https://github.com/v923z/micropython-ulab/pull/425. I could locally compile with that. If you OK it, I would merge it immediately.

water5, 2021-07-16:
Use `samd-fix` branch, although compile passed, but not appear like STM32 port compile message below:

> CC ../../../ulab/code/scipy/linalg/linalg.c
> CC ../../../ulab/code/scipy/optimize/optimize.c
> CC ../../../ulab/code/scipy/signal/signal.c
> CC ../../../ulab/code/scipy/special/special.c
> CC ../../../ulab/code/ndarray_operators.c
> CC ../../../ulab/code/ulab_tools.c
> CC ../../../ulab/code/ndarray.c
> CC ../../../ulab/code/numpy/ndarray/ndarray_iter.c
> CC ../../../ulab/code/ndarray_properties.c
> CC ../../../ulab/code/numpy/approx/approx.c
> CC ../../../ulab/code/numpy/compare/compare.c
> CC ../../../ulab/code/ulab_create.c
> CC ../../../ulab/code/numpy/fft/fft.c
> CC ../../../ulab/code/numpy/fft/fft_tools.c
> CC ../../../ulab/code/numpy/filter/filter.c
> CC ../../../ulab/code/numpy/linalg/linalg.c
> CC ../../../ulab/code/numpy/linalg/linalg_tools.c
> CC ../../../ulab/code/numpy/numerical/numerical.c
> CC ../../../ulab/code/numpy/poly/poly.c
> CC ../../../ulab/code/numpy/stats/stats.c
> CC ../../../ulab/code/numpy/transform/transform.c
> CC ../../../ulab/code/numpy/vector/vector.c
> CC ../../../ulab/code/numpy/numpy.c
> CC ../../../ulab/code/scipy/scipy.c
> CC ../../../ulab/code/user/user.c
> CC ../../../ulab/code/utils/utils.c
> CC ../../../ulab/code/ulab.c

And I have n't SAMD board currently to confirm `ulab` compiled into the firmware correct actually.
I need more time to do that.

v923z, 2021-07-16:
@water5 I think you are right, `ulab` is definitely not in the firmware. I have raised an issue in `micropython`: https://github.com/micropython/micropython/issues/7545

closed 2021-07-22 by v923z