v923z/micropython-ulab #436: [FEATURE REQUEST]

labodezao, 2021-08-30:
Hi, 
I wonder if it's possible to implement the hilbert transform with micropython ulab.
Best regards

v923z, 2021-08-31:
I assume, you want the implementation of https://docs.scipy.org/doc/scipy/reference/generated/scipy.fftpack.hilbert.html. However, before we add that, I wanted to ask, whether you could already calculate the Hilbert transform via the Fourier transform by applying the relation: [Relationship with the Fourier transform](https://en.wikipedia.org/wiki/Hilbert_transform#Relationship_with_the_Fourier_transform).

labodezao, 2022-01-02:
Hi, this is the right feature.
It's certainly possible to code this manually, yes, but some of my tries where hazardous :)

v923z, 2022-01-02:
OK, but the question is, whether this has to be implemented, if the same functionality can be gotten with the already available tools. In order to keep the flash size small, I would rather implement things with some `python` wrapper. https://github.com/v923z/micropython-ulab/tree/master/snippets

labodezao, 2022-01-02:
Hmm yes i did not think about it :)

v923z, 2022-01-29:
@labodezao I am not sure, where this issue stands now. Do you want to implement this functionality as a snippet, or should we close it?

labodezao, 2022-01-29:
This could be a good idea, but I don't have the knoloedge to modify the code :)

v923z, 2022-01-29:
> This could be a good idea, but I don't have the knowledge to modify the code :)

But if it is supposed to be a snippet, then you would write the code in `python`, so you would not have to change anything in the code base. That's the beauty of it.

labodezao, 2022-01-29:
How can I do that ?

---------------------------------------------------------------
http://labodezao.fr/
---------------------------------------------------------------
Ewen d'Aviau
9 rue fernand de Magellan
44600 Saint-Nazaire
06-02-37-53-22

v923z, 2022-01-29:
You have access to the Fourier transform, and then you use what is written in https://en.wikipedia.org/wiki/Hilbert_transform#Relationship_with_the_Fourier_transform. You can calculate the phase of your signal by calling `np.atan2`, and then shift the phase by +/-90 degrees. This all can be done in `python`.