v923z/micropython-ulab #437: Building on the Ti Nspire

jkcoxson, 2021-09-01:
**Describe the bug**
There do not seem to be universal instructions on how to build this module. I am attempting to build this module for the Ti Nspire [Micropython](https://github.com/Vogtinator/micropython) port but having little luck. 

The Nspire port has its own make file, which will use the Nspire board config and nspire-gcc cross compiler. Specifying ulab as a C module when calling the make command has no effect on the project building.

Building the objects in nspire-gcc with the -c flag gives many errors with building, leading me to think that it must be built with Micropython, not into object files that can be imported.

```nspire-gcc -c -I../../ -I../../nspire ulab.c
In file included from ulab_create.h:16,
                 from ulab.c:23:
ndarray.h:157:44: error: expected ‘)’ before ‘mp_obj_t’
  157 | mp_obj_t ndarray_binary_op(mp_binary_op_t , mp_obj_t , mp_obj_t );
      |                                            ^~~~~~~~~
      |                                            )
ndarray.h:158:42: error: expected ‘)’ before ‘mp_obj_t’
  158 | mp_obj_t ndarray_unary_op(mp_unary_op_t , mp_obj_t );
      |                                          ^~~~~~~~~
      |                                          )
ulab.c:94:6: error: ‘mp_obj_type_t’ {aka ‘const struct _mp_obj_type_t’} has no member named ‘flags’
   94 |     .flags = MP_TYPE_FLAG_EXTENDED
      |      ^~~~~
ulab.c:98:5: error: request for member ‘name’ in something not a structure or union
   98 |     .name = MP_QSTR_ndarray,
      |     ^
In file included from ulab_create.h:16,
                 from ulab.c:23:
ulab.c:110:17: error: ‘ndarray_unary_op’ undeclared here (not in a function)
  110 |     .unary_op = ndarray_unary_op,
      |                 ^~~~~~~~~~~~~~~~
ndarray.h:56:38: note: in definition of macro ‘MP_TYPE_EXTENDED_FIELDS’
   56 | #define MP_TYPE_EXTENDED_FIELDS(...) __VA_ARGS__
      |                                      ^~~~~~~~~~~
ulab.c:113:18: error: ‘ndarray_binary_op’ undeclared here (not in a function)
  113 |     .binary_op = ndarray_binary_op,
      |                  ^~~~~~~~~~~~~~~~~
ndarray.h:56:38: note: in definition of macro ‘MP_TYPE_EXTENDED_FIELDS’
   56 | #define MP_TYPE_EXTENDED_FIELDS(...) __VA_ARGS__
      |                                      ^~~~~~~~~~~
ulab.c:177:1: warning: data definition has no type or storage class
  177 | MP_REGISTER_MODULE(MP_QSTR_ulab, ulab_user_cmodule, MODULE_ULAB_ENABLED);
      | ^~~~~~~~~~~~~~~~~~
ulab.c:177:1: warning: type defaults to ‘int’ in declaration of ‘MP_REGISTER_MODULE’ [-Wimplicit-int]
ulab.c:177:1: warning: parameter names (without types) in function declaration```

Any assistance with this would be greatly appreciated, thanks your your work on this project.

jkcoxson, 2021-09-01:
This is my current Makefile for building Micropython for the Ti Nspire
```include ../py/mkenv.mk
-include mpconfigport.mk

CC := nspire-gcc
LD := arm-none-eabi-ld.gold

ZEHNFLAGS = --name "micropython" --author "micropython team, ported by Fabian Vogt" --notice "Interactive python console" --version 09 --compress

# define main target
PROG = micropython

# qstr definitions (must come before including py.mk)
QSTR_DEFS = qstrdefsport.h

# OS name, for simple autoconfig
UNAME_S := $(shell uname -s)

# Disable stripping of the main executable, relocation info is needed by genzehn
NOSTRIP := 1
# DEBUG := 1

# include py core make definitions
include ../py/py.mk

INC =  -I.
INC += -I..
INC += -I../py
INC += -I$(BUILD)

# Debugging/Optimization
ifdef DEBUG
COPT = -O0 -g
else
COPT = -Ofast -fomit-frame-pointer -flto
endif

# compiler settings
CWARN = -Wall -Wno-error=cpp -Wno-error=format 
CFLAGS =  $(INC) $(CWARN) -ansi -std=gnu99 $(CFLAGS_MOD) $(COPT) $(CFLAGS_EXTRA) -USER_C_MODULES=/home/camelus/ndless/micropython/micropython-ulab/code

LDFLAGS = $(LDFLAGS_MOD) -lm $(LDFLAGS_EXTRA) -Wl,--nspireio

# source files
SRC_C = $(shell find . -name \*.c)

SRC_QSTR += $(SRC_C)

OBJ = $(PY_O) $(addprefix $(BUILD)/, $(SRC_C:.c=.o))

include ../py/mkrules.mk

all: $(PROG).tns

$(PROG).tns: $(PROG)
	+genzehn --input $^ --output $@.zehn $(ZEHNFLAGS)
	+make-prg $@.zehn $@
	+rm $@.zehn
```

v923z, 2021-09-01:
I have just realised that the repository that you linked to hasn't been updated for about 3 years. It is really unlikely that you will be able to compile `ulab` with that, because user modules were added around two years ago. I would really start with a fresh clone of `micropython`, and try to add the modifications of https://github.com/Vogtinator/micropython/tree/next-next/nspire.

jkcoxson, 2021-09-01:
Ah that would make sense why it doesn't try to compile the module. Thanks for the info!