v923z/micropython-ulab #438: Trying to compile Ulab for RP2 using a MBP M1

alport, 2021-10-04:
**Describe the bug**
A clear and concise description of what the bug is. Give the `ulab` version

When I execute the supplied script: RP2.sh, I get the following error:
[ -d build-PICO ] || cmake -S . -B build-PICO -DPICO_BUILD_DOCS=0 -DMICROPY_BOARD=PICO -DUSER_C_MODULES=/Users/mike/ulab/code/micropython.cmake
/bin/sh: cmake: command not found
make: *** [all] Error 127

Any ideas?
Alternately, it would really help if someone could supply the final binary which I could upload directly to the RP2 using Thonny.

**To Reproduce**
./RP2.sh

**Expected behavior**
Should compile the RP2 firmware to the micropython/ports/rp2/build directory.

**Context**
I get the following when I execute the supplied script.
(base) mike@MikesM1 ~ % ./RP2.sh
Cloning ulab
Cloning into 'ulab'...
remote: Enumerating objects: 6853, done.
remote: Counting objects: 100% (1482/1482), done.
remote: Compressing objects: 100% (940/940), done.
remote: Total 6853 (delta 653), reused 722 (delta 532), pack-reused 5371
Receiving objects: 100% (6853/6853), 2.74 MiB | 349.00 KiB/s, done.
Resolving deltas: 100% (4583/4583), done.
Cloning MicroPython
Cloning into 'micropython'...
remote: Enumerating objects: 90401, done.
remote: Total 90401 (delta 0), reused 0 (delta 0), pack-reused 90401
Receiving objects: 100% (90401/90401), 49.50 MiB | 85.00 KiB/s, done.
Resolving deltas: 100% (65449/65449), done.
Submodule 'lib/asf4' (https://github.com/adafruit/asf4) registered for path 'lib/asf4'
Submodule 'lib/axtls' (https://github.com/micropython/axtls.git) registered for path 'lib/axtls'
Submodule 'lib/berkeley-db-1.xx' (https://github.com/pfalcon/berkeley-db-1.xx) registered for path 'lib/berkeley-db-1.xx'
Submodule 'lib/btstack' (https://github.com/bluekitchen/btstack.git) registered for path 'lib/btstack'
Submodule 'lib/libffi' (https://github.com/atgreen/libffi) registered for path 'lib/libffi'
Submodule 'lib/libhydrogen' (https://github.com/jedisct1/libhydrogen.git) registered for path 'lib/libhydrogen'
Submodule 'lib/lwip' (https://github.com/lwip-tcpip/lwip.git) registered for path 'lib/lwip'
Submodule 'lib/mbedtls' (https://github.com/ARMmbed/mbedtls.git) registered for path 'lib/mbedtls'
Submodule 'lib/mynewt-nimble' (https://github.com/apache/mynewt-nimble.git) registered for path 'lib/mynewt-nimble'
Submodule 'lib/nrfx' (https://github.com/NordicSemiconductor/nrfx.git) registered for path 'lib/nrfx'
Submodule 'lib/nxp_driver' (https://github.com/hathach/nxp_driver.git) registered for path 'lib/nxp_driver'
Submodule 'lib/pico-sdk' (https://github.com/raspberrypi/pico-sdk.git) registered for path 'lib/pico-sdk'
Submodule 'lib/stm32lib' (https://github.com/micropython/stm32lib) registered for path 'lib/stm32lib'
Submodule 'lib/tinyusb' (https://github.com/hathach/tinyusb) registered for path 'lib/tinyusb'
Cloning into '/Users/mike/micropython/lib/asf4'...
Cloning into '/Users/mike/micropython/lib/axtls'...
Cloning into '/Users/mike/micropython/lib/berkeley-db-1.xx'...
Cloning into '/Users/mike/micropython/lib/btstack'...
Cloning into '/Users/mike/micropython/lib/libffi'...
Cloning into '/Users/mike/micropython/lib/libhydrogen'...
Cloning into '/Users/mike/micropython/lib/lwip'...
Cloning into '/Users/mike/micropython/lib/mbedtls'...
Cloning into '/Users/mike/micropython/lib/mynewt-nimble'...
Cloning into '/Users/mike/micropython/lib/nrfx'...
Cloning into '/Users/mike/micropython/lib/nxp_driver'...
Cloning into '/Users/mike/micropython/lib/pico-sdk'...
Cloning into '/Users/mike/micropython/lib/stm32lib'...
Cloning into '/Users/mike/micropython/lib/tinyusb'...
Submodule path 'lib/asf4': checked out 'd270f79aa16dd8fd4ae3b6c14544283dcb992e9c'
Submodule path 'lib/axtls': checked out '531cab9c278c947d268bd4c94ecab9153a961b43'
Submodule path 'lib/berkeley-db-1.xx': checked out '35aaec4418ad78628a3b935885dd189d41ce779b'
Submodule path 'lib/btstack': checked out 'c8b9823f68c6af0fa52e2c4e009aba4dbf257232'
Submodule path 'lib/libffi': checked out 'e9de7e35f2339598b16cbb375f9992643ed81209'
Submodule path 'lib/libhydrogen': checked out '5c5d513093075f7245ea522101b17c50aa579af2'
Submodule path 'lib/lwip': checked out '159e31b689577dbf69cf0683bbaffbd71fa5ee10'
Submodule path 'lib/mbedtls': checked out '1bc2c9cb8b8fe4659bd94b8ebba5a4c02029b7fa'
Submodule path 'lib/mynewt-nimble': checked out '97ce3eacaaa79e8ed6cf71717149ced4f5328ee7'
Submodule path 'lib/nrfx': checked out '7a4c9d946cf1801771fc180acdbf7b878f270093'
Submodule path 'lib/nxp_driver': checked out 'b618cb1d521cc9e133bdcd0fca154dee2d925dfe'
Submodule path 'lib/pico-sdk': checked out 'bfcbefafc5d2a210551a4d9d80b4303d4ae0adf7'
Submodule path 'lib/stm32lib': checked out '302c52794d2f579903f4e49cbad1f5d3a7f401ad'
Submodule path 'lib/tinyusb': checked out 'd49938d0f5052bce70e55c652b657c0a6a7e84fe'
Use make V=1 or set BUILD_VERBOSE in your environment to increase build verbosity.
mkdir -p build/genhdr
GEN build/genhdr/mpversion.h
GEN build/genhdr/moduledefs.h
GEN build/genhdr/qstr.i.last
GEN build/genhdr/qstr.split
GEN build/genhdr/qstrdefs.collected.h
QSTR updated
GEN build/genhdr/qstrdefs.generated.h
mkdir -p build/py
mkdir -p build/shared/runtime
CC ../py/mpstate.c
CC ../py/nlr.c
CC ../py/nlrx86.c
CC ../py/nlrx64.c
CC ../py/nlrthumb.c
CC ../py/nlraarch64.c
CC ../py/nlrpowerpc.c
CC ../py/nlrxtensa.c
CC ../py/nlrsetjmp.c
CC ../py/malloc.c
CC ../py/gc.c
CC ../py/pystack.c
CC ../py/qstr.c
CC ../py/vstr.c
CC ../py/mpprint.c
CC ../py/unicode.c
CC ../py/mpz.c
CC ../py/reader.c
CC ../py/lexer.c
CC ../py/parse.c
CC ../py/scope.c
CC ../py/compile.c
CC ../py/emitcommon.c
CC ../py/emitbc.c
CC ../py/asmbase.c
CC ../py/asmx64.c
CC ../py/emitnx64.c
CC ../py/asmx86.c
CC ../py/emitnx86.c
CC ../py/asmthumb.c
CC ../py/emitnthumb.c
CC ../py/emitinlinethumb.c
CC ../py/asmarm.c
CC ../py/emitnarm.c
CC ../py/asmxtensa.c
CC ../py/emitnxtensa.c
CC ../py/emitinlinextensa.c
CC ../py/emitnxtensawin.c
CC ../py/formatfloat.c
CC ../py/parsenumbase.c
CC ../py/parsenum.c
CC ../py/emitglue.c
CC ../py/persistentcode.c
CC ../py/runtime.c
CC ../py/runtime_utils.c
CC ../py/scheduler.c
CC ../py/nativeglue.c
CC ../py/pairheap.c
CC ../py/ringbuf.c
CC ../py/stackctrl.c
CC ../py/argcheck.c
CC ../py/warning.c
CC ../py/profile.c
CC ../py/map.c
CC ../py/obj.c
CC ../py/objarray.c
CC ../py/objattrtuple.c
CC ../py/objbool.c
CC ../py/objboundmeth.c
CC ../py/objcell.c
CC ../py/objclosure.c
CC ../py/objcomplex.c
CC ../py/objdeque.c
CC ../py/objdict.c
CC ../py/objenumerate.c
CC ../py/objexcept.c
CC ../py/objfilter.c
CC ../py/objfloat.c
CC ../py/objfun.c
CC ../py/objgenerator.c
CC ../py/objgetitemiter.c
CC ../py/objint.c
CC ../py/objint_longlong.c
CC ../py/objint_mpz.c
CC ../py/objlist.c
CC ../py/objmap.c
CC ../py/objmodule.c
CC ../py/objobject.c
CC ../py/objpolyiter.c
CC ../py/objproperty.c
CC ../py/objnone.c
CC ../py/objnamedtuple.c
CC ../py/objrange.c
CC ../py/objreversed.c
CC ../py/objset.c
CC ../py/objsingleton.c
CC ../py/objslice.c
CC ../py/objstr.c
CC ../py/objstrunicode.c
CC ../py/objstringio.c
CC ../py/objtuple.c
CC ../py/objtype.c
CC ../py/objzip.c
CC ../py/opmethods.c
CC ../py/sequence.c
CC ../py/stream.c
CC ../py/binary.c
CC ../py/builtinimport.c
CC ../py/builtinevex.c
CC ../py/builtinhelp.c
CC ../py/modarray.c
CC ../py/modbuiltins.c
CC ../py/modcollections.c
CC ../py/modgc.c
CC ../py/modio.c
CC ../py/modmath.c
CC ../py/modcmath.c
CC ../py/modmicropython.c
CC ../py/modstruct.c
CC ../py/modsys.c
CC ../py/moduerrno.c
CC ../py/modthread.c
CC ../py/vm.c
CC ../py/bc.c
CC ../py/showbc.c
CC ../py/repl.c
CC ../py/smallint.c
CC ../py/frozenmod.c
CC main.c
CC gccollect.c
CC ../shared/runtime/gchelper_generic.c
LINK mpy-cross
__TEXT	__DATA	__OBJC	others	dec	hex
311296	16384	0	4295016448	4295344128	10005c000
rm: build: No such file or directory
[ -d build-PICO ] || cmake -S . -B build-PICO -DPICO_BUILD_DOCS=0 -DMICROPY_BOARD=PICO -DUSER_C_MODULES=/Users/mike/ulab/code/micropython.cmake
/bin/sh: cmake: command not found
make: *** [all] Error 127

v923z, 2021-10-04:
@alport As far as I understand, this issue has nothing to do with `ulab`: it seems to me that you don't have `cmake` on your system. You can find some pointers here: https://cmake.org/install/

alport, 2021-10-05:
Thanks, but need some more help:

I installed Cmake for the "Mac 10.13 or later". It seems like I had then to configure it for a Command Line Use, by executing the Cmake app, and following the instructions:

```
After copying CMake.app into /Applications (or a custom location), run it and follow the “How to Install For Command Line Use” menu item for instructions to make the command-line tools (e.g. cmake) available in the PATH.

One may add CMake to the PATH:

 PATH="/Applications/CMake.app/Contents/bin":"$PATH"
```

which is what I did - and then repeated the ./RP2.sh, which gave me the following error log:

[ -d build-PICO ] || cmake -S . -B build-PICO -DPICO_BUILD_DOCS=0 -DMICROPY_BOARD=PICO -DUSER_C_MODULES=/Users/mike/ulab/code/micropython.cmake
PICO_SDK_PATH is /Users/mike/micropython/lib/pico-sdk
Defaulting PICO_PLATFORM to rp2040 since not specified.
Defaulting PICO platform compiler to pico_arm_gcc since not specified.
PICO compiler is pico_arm_gcc
CMake Error at /Users/mike/micropython/lib/pico-sdk/cmake/preload/toolchains/find_compiler.cmake:28 (message):
  Compiler 'arm-none-eabi-gcc' not found, you can specify search path with
  "PICO_TOOLCHAIN_PATH".
Call Stack (most recent call first):
  /Users/mike/micropython/lib/pico-sdk/cmake/preload/toolchains/pico_arm_gcc.cmake:20 (pico_find_compiler)
  /Applications/CMake.app/Contents/share/cmake-3.21/Modules/CMakeDetermineSystem.cmake:124 (include)
  CMakeLists.txt:50 (project)

CMake Error: CMake was unable to find a build program corresponding to "Unix Makefiles".  CMAKE_MAKE_PROGRAM is not set.  You probably need to select a different build tool.
CMake Error: CMAKE_C_COMPILER not set, after EnableLanguage
CMake Error: CMAKE_CXX_COMPILER not set, after EnableLanguage
-- Configuring incomplete, errors occurred!
make: *** [all] Error 1

I am also trying a parallel install on another Intel based MBP - but get the same error.

Again, your help would be appreciated...!
Thanks,

v923z, 2021-10-06:
@alport Would https://github.com/pimoroni/pimoroni-pico/releases/download/v0.1.0/pimoroni-pico-micropython.uf2 work for you? I don't know, whether later versions of the firmware include `ulab`. Based on https://github.com/pimoroni/pimoroni-pico/blob/main/micropython/modules/micropython.cmake, I would assume that `ulab` is included in all versions after 0.1: https://github.com/pimoroni/pimoroni-pico/releases But, on the other hand, you can simply try it out, and if you can't import it, then it is not there. Have you seen https://github.com/pimoroni/pimoroni-pico/blob/main/setting-up-the-pico-sdk.md?

@Gadgetoid Philip, is this something that should be fixed here in the long run? It seems to me that the compiler is not set properly in the `cmake` file, but I can't really comment on the issue.

Gadgetoid, 2021-10-06:
> @Gadgetoid Philip, is this something that should be fixed here in the long run? It seems to me that the compiler is not set properly in the `cmake` file, but I can't really comment on the issue.

I don't think so. It just looks like gcc-arm-none-eabi isn't installed or available in PATH.

The crux of the error is:

```
CMake Error at /Users/mike/micropython/lib/pico-sdk/cmake/preload/toolchains/find_compiler.cmake:28 (message):
Compiler 'arm-none-eabi-gcc' not found, you can specify search path with
"PICO_TOOLCHAIN_PATH".
```

@alport I don't have any experience building from macOS, but I'd recommend getting the basic Pico Examples building against the SDK first before upping the ante to MicroPython + Modules. There's a guide here, but I can't vouch for its accuracy: https://blog.smittytone.net/2021/02/02/program-raspberry-pi-pico-c-mac/

Other than that, I think you're on the right track.

v923z, 2021-11-04:
@alport Mike, can we close the issue, or do you still have difficulties with the compilation?

closed 2021-11-17 by v923z