v923z/micropython-ulab #441: New compile problems

mdaeron, 2021-10-17:
I'm trying to use the `esp32-cmake.sh` script. I am on MacOS 10.15.7 with cmake version 3.20.5. Everything seems fine until the final `make` command, which throws the error `make: *** No targets.  Stop.` Is there something obvious I'm missing? Script output (without the `git clone` parts) below.

````
Installing ESP-IDF tools
Installing tools: xtensa-esp32-elf, esp32ulp-elf, openocd-esp32
Skipping xtensa-esp32-elf@esp-2020r3-8.4.0 (already installed)
Skipping esp32ulp-elf@2.28.51.20170517 (already installed)
Skipping openocd-esp32@v0.10.0-esp32-20200709 (already installed)
Installing Python environment and packages
Installing Python packages from /Users/mathieu/ShareBox/uPy/firmware/tinypico_with_ulab/micropython/esp-idf/requirements.txt
Requirement already satisfied: setuptools in /Users/mathieu/.espressif/python_env/idf4.0_py3.9_env/lib/python3.9/site-packages (from -r /Users/mathieu/ShareBox/uPy/firmware/tinypico_with_ulab/micropython/esp-idf/requirements.txt (line 4)) (57.4.0)
Requirement already satisfied: click>=5.0 in /Users/mathieu/.espressif/python_env/idf4.0_py3.9_env/lib/python3.9/site-packages (from -r /Users/mathieu/ShareBox/uPy/firmware/tinypico_with_ulab/micropython/esp-idf/requirements.txt (line 8)) (8.0.3)
Requirement already satisfied: pyserial>=3.0 in /Users/mathieu/.espressif/python_env/idf4.0_py3.9_env/lib/python3.9/site-packages (from -r /Users/mathieu/ShareBox/uPy/firmware/tinypico_with_ulab/micropython/esp-idf/requirements.txt (line 9)) (3.5)
Requirement already satisfied: future>=0.15.2 in /Users/mathieu/.espressif/python_env/idf4.0_py3.9_env/lib/python3.9/site-packages (from -r /Users/mathieu/ShareBox/uPy/firmware/tinypico_with_ulab/micropython/esp-idf/requirements.txt (line 10)) (0.18.2)
Requirement already satisfied: cryptography>=2.1.4 in /Users/mathieu/.espressif/python_env/idf4.0_py3.9_env/lib/python3.9/site-packages (from -r /Users/mathieu/ShareBox/uPy/firmware/tinypico_with_ulab/micropython/esp-idf/requirements.txt (line 11)) (35.0.0)
Requirement already satisfied: pyparsing<2.4.0,>=2.0.3 in /Users/mathieu/.espressif/python_env/idf4.0_py3.9_env/lib/python3.9/site-packages (from -r /Users/mathieu/ShareBox/uPy/firmware/tinypico_with_ulab/micropython/esp-idf/requirements.txt (line 12)) (2.3.1)
Requirement already satisfied: pyelftools>=0.22 in /Users/mathieu/.espressif/python_env/idf4.0_py3.9_env/lib/python3.9/site-packages (from -r /Users/mathieu/ShareBox/uPy/firmware/tinypico_with_ulab/micropython/esp-idf/requirements.txt (line 13)) (0.27)
Requirement already satisfied: cffi>=1.12 in /Users/mathieu/.espressif/python_env/idf4.0_py3.9_env/lib/python3.9/site-packages (from cryptography>=2.1.4->-r /Users/mathieu/ShareBox/uPy/firmware/tinypico_with_ulab/micropython/esp-idf/requirements.txt (line 11)) (1.15.0)
Requirement already satisfied: pycparser in /Users/mathieu/.espressif/python_env/idf4.0_py3.9_env/lib/python3.9/site-packages (from cffi>=1.12->cryptography>=2.1.4->-r /Users/mathieu/ShareBox/uPy/firmware/tinypico_with_ulab/micropython/esp-idf/requirements.txt (line 11)) (2.20)
WARNING: You are using pip version 21.2.4; however, version 21.3 is available.
You should consider upgrading via the '/Users/mathieu/.espressif/python_env/idf4.0_py3.9_env/bin/python -m pip install --upgrade pip' command.
All done! You can now run:

  . ./export.sh

Setting IDF_PATH to '/Users/mathieu/ShareBox/uPy/firmware/tinypico_with_ulab/micropython/esp-idf'
Adding ESP-IDF tools to PATH...
Checking if Python packages are up to date...
Python requirements from /Users/mathieu/ShareBox/uPy/firmware/tinypico_with_ulab/micropython/esp-idf/requirements.txt are satisfied.
Added the following directories to PATH:
  /Users/mathieu/ShareBox/uPy/firmware/tinypico_with_ulab/micropython/esp-idf/components/esptool_py/esptool
  /Users/mathieu/ShareBox/uPy/firmware/tinypico_with_ulab/micropython/esp-idf/components/espcoredump
  /Users/mathieu/ShareBox/uPy/firmware/tinypico_with_ulab/micropython/esp-idf/components/partition_table/
  /Users/mathieu/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin
  /Users/mathieu/.espressif/tools/esp32ulp-elf/2.28.51.20170517/esp32ulp-elf-binutils/bin
  /Users/mathieu/.espressif/tools/openocd-esp32/v0.10.0-esp32-20200709/openocd-esp32/bin
  /Users/mathieu/.espressif/python_env/idf4.0_py3.9_env/bin
  /Users/mathieu/ShareBox/uPy/firmware/tinypico_with_ulab/micropython/esp-idf/tools
Done! You can now compile ESP-IDF projects.
Go to the project directory and run:

  idf.py build

Use make V=1 or set BUILD_VERBOSE in your environment to increase build verbosity.
mkdir -p build/genhdr
GEN build/genhdr/mpversion.h
GEN build/genhdr/moduledefs.h
GEN build/genhdr/qstr.i.last
GEN build/genhdr/qstr.split
GEN build/genhdr/qstrdefs.collected.h
QSTR updated
GEN build/genhdr/qstrdefs.generated.h
mkdir -p build/py
mkdir -p build/shared/runtime
CC ../py/mpstate.c
CC ../py/nlr.c
CC ../py/nlrx86.c
CC ../py/nlrx64.c
CC ../py/nlrthumb.c
CC ../py/nlraarch64.c
CC ../py/nlrpowerpc.c
CC ../py/nlrxtensa.c
CC ../py/nlrsetjmp.c
CC ../py/malloc.c
CC ../py/gc.c
CC ../py/pystack.c
CC ../py/qstr.c
CC ../py/vstr.c
CC ../py/mpprint.c
CC ../py/unicode.c
CC ../py/mpz.c
CC ../py/reader.c
CC ../py/lexer.c
CC ../py/parse.c
CC ../py/scope.c
CC ../py/compile.c
CC ../py/emitcommon.c
CC ../py/emitbc.c
CC ../py/asmbase.c
CC ../py/asmx64.c
CC ../py/emitnx64.c
CC ../py/asmx86.c
CC ../py/emitnx86.c
CC ../py/asmthumb.c
CC ../py/emitnthumb.c
CC ../py/emitinlinethumb.c
CC ../py/asmarm.c
CC ../py/emitnarm.c
CC ../py/asmxtensa.c
CC ../py/emitnxtensa.c
CC ../py/emitinlinextensa.c
CC ../py/emitnxtensawin.c
CC ../py/formatfloat.c
CC ../py/parsenumbase.c
CC ../py/parsenum.c
CC ../py/emitglue.c
CC ../py/persistentcode.c
CC ../py/runtime.c
CC ../py/runtime_utils.c
CC ../py/scheduler.c
CC ../py/nativeglue.c
CC ../py/pairheap.c
CC ../py/ringbuf.c
CC ../py/stackctrl.c
CC ../py/argcheck.c
CC ../py/warning.c
CC ../py/profile.c
CC ../py/map.c
CC ../py/obj.c
CC ../py/objarray.c
CC ../py/objattrtuple.c
CC ../py/objbool.c
CC ../py/objboundmeth.c
CC ../py/objcell.c
CC ../py/objclosure.c
CC ../py/objcomplex.c
CC ../py/objdeque.c
CC ../py/objdict.c
CC ../py/objenumerate.c
CC ../py/objexcept.c
CC ../py/objfilter.c
CC ../py/objfloat.c
CC ../py/objfun.c
CC ../py/objgenerator.c
CC ../py/objgetitemiter.c
CC ../py/objint.c
CC ../py/objint_longlong.c
CC ../py/objint_mpz.c
CC ../py/objlist.c
CC ../py/objmap.c
CC ../py/objmodule.c
CC ../py/objobject.c
CC ../py/objpolyiter.c
CC ../py/objproperty.c
CC ../py/objnone.c
CC ../py/objnamedtuple.c
CC ../py/objrange.c
CC ../py/objreversed.c
CC ../py/objset.c
CC ../py/objsingleton.c
CC ../py/objslice.c
CC ../py/objstr.c
CC ../py/objstrunicode.c
CC ../py/objstringio.c
CC ../py/objtuple.c
CC ../py/objtype.c
CC ../py/objzip.c
CC ../py/opmethods.c
CC ../py/sequence.c
CC ../py/stream.c
CC ../py/binary.c
CC ../py/builtinimport.c
CC ../py/builtinevex.c
CC ../py/builtinhelp.c
CC ../py/modarray.c
CC ../py/modbuiltins.c
CC ../py/modcollections.c
CC ../py/modgc.c
CC ../py/modio.c
CC ../py/modmath.c
CC ../py/modcmath.c
CC ../py/modmicropython.c
CC ../py/modstruct.c
CC ../py/modsys.c
CC ../py/moduerrno.c
CC ../py/modthread.c
CC ../py/vm.c
CC ../py/bc.c
CC ../py/showbc.c
CC ../py/repl.c
CC ../py/smallint.c
CC ../py/frozenmod.c
CC main.c
CC gccollect.c
CC ../shared/runtime/gchelper_generic.c
LINK mpy-cross
__TEXT	__DATA	__OBJC	others	dec	hex
307200	4096	0	4295000076	4295311372	10005400c
git submodule update --init ../../lib/berkeley-db-1.xx
Submodule 'lib/berkeley-db-1.xx' (https://github.com/pfalcon/berkeley-db-1.xx) registered for path '../../lib/berkeley-db-1.xx'
Cloning into '/Users/mathieu/ShareBox/uPy/firmware/tinypico_with_ulab/micropython/lib/berkeley-db-1.xx'...
Submodule path '../../lib/berkeley-db-1.xx': checked out '35aaec4418ad78628a3b935885dd189d41ce779b'
make: *** No targets.  Stop.
````

v923z, 2021-10-17:
@mdaeron Can you build the firmware without `ulab`? Not necessarily with the script here, but with any script. It seems to me that the script hangs, when you want to `make` the `berkeley` submodule. I don't know at the moment, why that would happen.

mdaeron, 2021-10-17:
OK now it compiles but the resulting firmware can't import `ulab` (ImportError). I believe it is because `echo "USER_C_MODULES = \$(BUILD_DIR)/ulab/code/micropython.cmake"` should come earlier in the MakeFile. But when I chang that by hand I get multiple compile errors. I'll report back after I find more time to look at the errors.

mdaeron, 2021-10-18:
Here is what worked for me in the end:

```bash
#!/bin/bash

export BUILD_DIR=$(pwd)

echo "--- CLONING ULAB ---"
git clone --depth 1 https://github.com/v923z/micropython-ulab.git ulab

echo "--- CLONING MICROPYTHON ---"
git clone --depth 1 https://github.com/micropython/micropython.git

echo "--- CLONING ESP-IDF ---"
cd $BUILD_DIR/micropython/
git clone --depth 1 -b v4.0.2 --recursive https://github.com/espressif/esp-idf.git

echo "--- INSTALL ESP-IDF ---"
cd $BUILD_DIR/micropython/esp-idf
./install.sh
. ./export.sh

echo "--- MPY-CROSS ---"
cd $BUILD_DIR/micropython/mpy-cross
make

echo "--- ESP32 SUBMODULES ---"
cd $BUILD_DIR/micropython/ports/esp32
make submodules

echo "--- PATCH MAKEFILE ---"
cp $BUILD_DIR/micropython/ports/esp32/MakeFile $BUILD_DIR/micropython/ports/esp32/MakeFileOld
echo "BOARD = UM_TINYPICO" > $BUILD_DIR/micropython/ports/esp32/MakeFile
echo "USER_C_MODULES = \$(BUILD_DIR)/ulab/code/micropython.cmake" >> $BUILD_DIR/micropython/ports/esp32/MakeFile
cat $BUILD_DIR/micropython/ports/esp32/MakeFileOld >> $BUILD_DIR/micropython/ports/esp32/MakeFile

echo "--- MAKE ---"
make
```