v923z/micropython-ulab #443: [FEATURE REQUEST] Enable versions in ReadTheDocs documentation

RobertoRoos, 2021-11-04:
**Describe the solution you'd like**

In short: can different older tagged versions be enabled in the RTD project: https://readthedocs.org/projects/micropython-ulab

In the admin panel you can enable which branches and tags you want built and available. I would like to access the docs of some older versions, because we have a old fork of this repo running somewhere.

I understand this is quite niche, but considering how little effort it takes I hope this can be considered!

RobertoRoos, 2021-11-04:
In the meantime I made a fresh fork with my own RTD project, this is what it would then look like: https://micropython-ulab-robert.readthedocs.io/en/0.36.0

v923z, 2021-11-04:
@RobertoRoos Thanks for raising the issue. I also think that this feature would be useful. I'll fix the RTD side later today.

v923z, 2021-11-04:
@RobertoRoos I have activated the older versions on RTD, but all prior to 2.1.2 fail, because at that time we stipulated that the requirements file should be used, and that does not exist for the old tags. I don't want to modify those versions on github. Do you know, whether we could fix this in RTD itself? I would be OK with a one-time setting of version-specific build options, if that is possible.

RobertoRoos, 2021-11-05:
Hm, in my fork it seems only the v1.* versions fail, v0.* do work: https://readthedocs.org/projects/micropython-ulab-robert/builds/ (but I haven't enabled many). I kept all the default settings in RTD (so no requirements.txt path).

This is why the [.readthedocs.yaml](https://docs.readthedocs.io/en/stable/config-file/index.html) file is convenient, over the online settings.

We could commit the current RTD settings to the YAML file for current and future settings, and tweak the online settings to work with legacy versions?
Actually, you could probably reset the online settings too after those builds are completed.

But I could understand this might be more effort than it's worth. Thanks for versions that did built, that might be enough for me!

v923z, 2021-11-05:
> Hm, in my fork it seems only the v1.* versions fail, v0.* do work: https://readthedocs.org/projects/micropython-ulab-robert/builds/ (but I haven't enabled many). I kept all the default settings in RTD (so no requirements.txt path).

We had to add the requirements at one point, because the defaults didn't load the autoapi. 

> 
> This is why the [.readthedocs.yaml](https://docs.readthedocs.io/en/stable/config-file/index.html) file is convenient, over the online settings.

That's right, but the yaml file is also part of the repository, so how should we add that to the older versions? I completely agree with your point, but I never thought that we would need anything beyond the bare minimum. We should have established the yaml file at the very beginning.

> We could commit the current RTD settings to the YAML file for current and future settings, and tweak the online settings to work with legacy versions? Actually, you could probably reset the online settings too after those builds are completed.

The problem is, I think, that the requirements are needed for versions beyond 1.1, I believe, so we can't just simply remove that. 

> But I could understand this might be more effort than it's worth. Thanks for versions that did built, that might be enough for me!

I am really not against enabling building the documentation for all versions, but I don't see at the moment, how we could pull that off.

RobertoRoos, 2021-11-05:
Right. I think in my example the build was technically complete but some pages must be incorrect.

One approach would be to add new old releases. So e.g. create a branch `release/v0.54` on top of tag `v0.54.5` and add an appropriate YAML in that branch with a new commit. Then tag that commit with e.g. `v0.54.6` and enable that tag in RTD.
We would have to do the same for each existing version that we want built in RTD.

v923z, 2021-11-05:
> Right. I think in my example the build was technically complete but some pages must be incorrect.
>
> One approach would be to add new old releases. So e.g. create a branch `release/v0.54` on top of [...]

I would be OK with that. Do you want to add PRs with that?
Just out of curiosity, why do you need the older versions? The last version is orders of magnitude better than the old ones, and I am quite certain that bugs won't even be fixed in those.

RobertoRoos, 2021-11-19:
Sorry I took a while.
Pull requests:

- Master version (for current and future releases): #450

v923z, 2021-11-19:
@RobertoRoos Thanks again for contributing. I have merged all your PRs, so I think this issue can now be closed.

closed 2021-11-19 by v923z