v923z/micropython-ulab #445: [FEATURE REQUEST] scipy.signal.butter

wired8, 2021-11-06:
Please implement:
scipy.signal.butter

scipy.signal.butter(N, Wn, btype='low', analog=False, output='ba', fs=None)
Design an Nth-order digital or analog Butterworth filter and return the filter coefficients.
https://docs.scipy.org/doc/scipy/reference/generated/scipy.signal.butter.html

v923z, 2021-11-06:
@wired8 I think this should be implemented in `python`, and not C. At least, as far as I see, this is what happens in `scipy` itself: https://github.com/scipy/scipy/blob/master/scipy/signal/_filter_design.py

For such cases, we have the [snippets](https://github.com/v923z/micropython-ulab/tree/master/snippets), and I would be more than happy to add that there. Do you think you can implement the function in `python`?

wired8, 2021-11-07:
@v923z looking at the filter source I believe I can.   When I get it working I'll send a PR for snippets.  Thanks!

v923z, 2021-11-07:
@wired8 One more comment: people, who need filters on micropython design the filter on the PC, and then copy only the coefficients.

wired8, 2021-11-07:
I'm aware.  I've built a noise meter and now want to calculate 1/3 octaves.

closed 2022-01-14 by v923z