v923z/micropython-ulab #454: [BUG] Shape of empty array doesn't match Numpy CPython

JorgeGMarques, 2021-11-20:
Shape of empty array returns empty turple `()`  (length 0) instead of `(0,)`  (length 1).

Build from commit c6bcc058b2e2af86fd3291cd956e2bcb2aeb46d2 .

**To Reproduce**
micropython ulab 
```python
from ulab import numpy as np
a = np.zeros(0)
s = a.shape
print (s, len(s))
```
prints
```
() 0
```

**Expected behavior**
cpython numpy
```python
from ulab import numpy as np
a = np.zeros(0)
s = a.shape
print (s, len(s))
```
prints
```
(0,) 1
```

**Additional context**
This little difference can be problematic when stacking arrays like in [here](https://github.com/JorgeGMarques/scikit-fuzzy/blob/fedc7a60f18884d0573550587aaa271f6c1a7f9c/skfuzzy/cluster/_cmeans.py#L238).

v923z, 2021-11-20:
@JorgeGMarques Many thanks for reporting the issue! I think https://github.com/v923z/micropython-ulab/pull/455 should be the fix. Could you, please, check, whether it does what you expect?

JorgeGMarques, 2021-11-24:
Yes! It is now consistent:
```Python
from ulab import numpy as np
a = np.zeros(0)
s1 = a.shape
print (s1, len(s1))

b = np.array([])
s2 = b.shape
print(s2,len(s2))
```
both cases print:
```Python
(0,) 1
(0,) 1
```

matching numpy behavior:
```Python
import numpy as np
a = np.zeros(0)
s1 = a.shape
print (s1, len(s1))

b = np.array([])
s2 = b.shape
print(s2,len(s2))
```
which prints:
```Python
(0,) 1
(0,) 1
```

Sorry for taking a bit to reply, I worked around this issue in my questionable port of [scikit-fuzzy](https://github.com/JorgeGMarques/scikit-fuzzy/tree/micropython) to micropython and I hadn't had time until now to properly test it.

closed 2021-11-24 by v923z