v923z/micropython-ulab #457: [BUG] ulab.numpy.sum(ulab.array(..., dtype=bool)) inaccurate

jepler, 2021-11-30:
**Describe the bug**
`ulab.numpy.sum` for integer argument types gives inaccurate results

```
$ git describe --tags
3.3.1-36-gcd70c91
```

**To Reproduce**
For embedded platforms with mp_float_t being 32 bits, it can be reproduced easily:
```
>>> ulab.numpy.sum(ulab.numpy.array([True, True, False], dtype=ulab.numpy.bool))
1
```
(should print 2).  The behavior is the same/similar for u/int8/16 types.

For builds with 64-bit floats (e.g., the micropython/circuitpython version used for running tests during CI), the smallest array I found reproducing the problem was

```
>>> ulab.numpy.sum(ulab.numpy.array([False] + [True] * 13, dtype=ulab.numpy.bool))
12
```
(should print 13)

**Expected behavior**
The sum should be accurate.

**Additional context**
`numerical_sum_mean_std_ndarray` maintains a running average `M`, then back-multiplies it to get the sum. This can lose accuracy, more quickly for embedded and less quickly for desktop. However, just revising the conversion to int to use `MICROPY_FLOAT_C_FUN(round)(M * ndarray->len)` might make it 'accurate enough' over a wider range of values, so that it matches the builtin sum() over an array of bools or integers.

Originally reported by @mrdalgaard as
 * https://github.com/adafruit/circuitpython/issues/5609

v923z, 2021-11-30:
@jepler Thanks for bringing this issue to the fore!

> **Additional context** `numerical_sum_mean_std_ndarray` maintains a running average `M`, then back-multiplies it to get the sum. This can lose accuracy, more quickly for embedded and less quickly for desktop. 

Actually, I think this calculation of the sum/mean is more accurate than direct summation, and the problem arises only in the cast to `int32`. If you print out the value of `M * ndarray->len`, it is always correct to printing precision.

> However, just revising the conversion to int to use `MICROPY_FLOAT_C_FUN(round)(M * ndarray->len)` might make it 'accurate enough' over a wider range of values, so that it matches the builtin sum() over an array of bools or integers.

Since the error is definitely smaller than 0.5, I can't really imagine that your suggested solution could ever fail. Thanks for pointing it out! I will patch up the code, and create a tag for you.

v923z, 2021-11-30:
https://github.com/v923z/micropython-ulab/pull/458 fixes the issue.

closed 2021-11-30 by v923z