v923z/micropython-ulab #462: [FEATURE REQUEST] Is it have plan to add 3 dimensions or more of numpy.array support?

water5, 2021-12-18:
Or add numpy.array's dtype support object?

v923z, 2021-12-18:
Four dimensions are supported. You just have to set it in https://github.com/v923z/micropython-ulab/blob/a99e0b98787266369eaa33bdae271de761eca05b/code/ulab.h#L44

The `dtype` object is also supported, you can enable it in https://github.com/v923z/micropython-ulab/blob/a99e0b98787266369eaa33bdae271de761eca05b/code/ulab.h#L76

water5, 2021-12-18:
It seems only handle numerical objects, 
```python
a = [1, 2, 3]
an = np.array([a, a, a])
```
or
```python
a = (1, 2, 3)
an = np.array([a, a, a])
```

> array([[1.0, 2.0, 3.0],
>       [1.0, 2.0, 3.0],
>       [1.0, 2.0, 3.0]], dtype=float32)

But not work with other type of objects:
```python
a = ['c7', 'M']
an = np.array([a, a, a])
```

>Traceback (most recent call last):
>  File "stdin", line 1, in module
>ValueError: too many dimensions

Is it possible likes numpy handle complex objects?
```python
a = {'num': 1, 'list_str': ['abc', 'E'], 'tup_': (2, 5), 'set_': {'u1', 'u2'}}
an = np.array([a, a, a])
```

>array([{'num': 1, 'list_str': ['abc', 'E'], 'tup_': (2, 5), 'set_': {'u1', 'u2'}},
>       {'num': 1, 'list_str': ['abc', 'E'], 'tup_': (2, 5), 'set_': {'u1', 'u2'}},
>       {'num': 1, 'list_str': ['abc', 'E'], 'tup_': (2, 5), 'set_': {'u1', 'u2'}}],
>      dtype=object)

v923z, 2021-12-18:
> It seems only handle numerical objects,
>
> ```python
> a = [1, 2, 3]
> an = np.array([a, a, a])
> ```
>
> or
>
> ```python
> a = (1, 2, 3)
> an = np.array([a, a, a])
> ```
>
>
> But not work with other type of objects: [...]

That's correct. I have probably misinterpreted your original question. As stated in the https://github.com/v923z/micropython-ulab/#readme, we support only numerical values of integer and float types. Complex is on the way: https://github.com/v923z/micropython-ulab/pull/456

> Is it possible likes numpy handle complex objects?
>
> ```python
> a = {'num': 1, 'list_str': ['abc', 'E'], 'tup_': (2, 5), 'set_': {'u1', 'u2'}}
> an = np.array([a, a, a])
> ```

Honestly, I don't see too much use for this on a microcontroller, and the overhead of implementing this is enormous. The problem is that some of the microcontrollers have virtually no RAM or flash, and implementing what you suggest in a way that can be disentangled, so that small microcontrollers would still have access to the numerical arrays is a huge undertaking. If only two people need it in the whole world, then I would spend my resources on features that have a broader use case. If you need such advanced facilities, then `numpy` is the way to go. `ulab` is just too small for that.

closed 2021-12-19 by water5