v923z/micropython-ulab #463: [FEATURE REQUEST] numpy.insert()

water5, 2021-12-19:
[https://numpy.org/doc/1.21/reference/generated/numpy.insert.html](https://numpy.org/doc/1.21/reference/generated/numpy.insert.html)

```python
b = np.array([[1, 2, 3], [4, 5, 6]])
```
>array([[1, 2, 3],
>       [4, 5, 6]])

```python
np.insert(b, 1, [7, 8, 9])
```
>array([1, 7, 8, 9, 2, 3, 4, 5, 6])

```python
np.insert(b, 1, [7, 8, 9], axis = 0)
```
>array([[1, 2, 3],
>       [7, 8, 9],
>       [4, 5, 6]])

v923z, 2022-01-19:
@water5 Wouldn't this basically duplicate the functionality of `concatenate'? https://micropython-ulab.readthedocs.io/en/latest/ulab-ndarray.html?highlight=concatenate#concatenate

water5, 2022-01-19:
But that can't specify where to insert, I consider [numpy.append](https://numpy.org/doc/stable/reference/generated/numpy.append.html) before post this request.

v923z, 2022-01-19:
Obviously, you first have to split your original array into two with slicing (that is, where you specify where you want to insert the values), and then `concatenate` the first slice, the value, and the second slice.

```
a = array([1, 2, 3, 4, 5, 6])
value = array([10, 11])
concatenate(a[:3], value, a[3:])
```

v923z, 2022-01-19:
Append is similar

```
a = array([1, 2, 3, 4, 5, 6])
b = array([10, 20, 30])
c = zeros(len(a) + len(b))
c[:len(a)] = a
c[len(a):] = b
```

water5, 2022-01-19:
On 2D array, 
```python
a = np.arange(25).reshape(5, 5)
a
```
>array([[ 0,  1,  2,  3,  4],
>       [ 5,  6,  7,  8,  9],
>       [10, 11, 12, 13, 14],
>       [15, 16, 17, 18, 19],
>       [20, 21, 22, 23, 24]])
```python
b = np.arange(25, 31)
b
```
>array([25, 26, 27, 28, 29])
```python
np.insert(a, 1, b, axis = 0)
```
>array([[ 0,  1,  2,  3,  4],
>       [25, 26, 27, 28, 29],
>       [ 5,  6,  7,  8,  9],
>       [10, 11, 12, 13, 14],
>       [15, 16, 17, 18, 19],
>       [20, 21, 22, 23, 24]])
```python
np.insert(a, 1, b, axis = 1)
```
>array([[ 0, 25,  1,  2,  3,  4],
>       [ 5, 26,  6,  7,  8,  9],
>       [10, 27, 11, 12, 13, 14],
>       [15, 28, 16, 17, 18, 19],
>       [20, 29, 21, 22, 23, 24]])
```python
np.concatenate(a[ : 1], b, a[1 : ])
```
>Traceback (most recent call last):
>  File "\<stdin\>", line 1, in \<module\>
>  File "\<\_\_array_function\_\_ internals\>", line 180, in concatenate
>TypeError: only integer scalar arrays can be converted to a scalar index

v923z, 2022-01-19:
```
a = array(range(25)).reshape((5,5))
b = np.arange(25, 30).reshape((1,5))
np.concatenate((a[:1], b, a[1:]))
```

water5, 2022-01-19:
Right, that seems could implement most `numpy.insert` function.
I close this issue directly, or make a snippet?

v923z, 2022-01-19:
> Right, that seems could implement most `numpy.insert` function. I close this issue directly, or make a snippet?

You could definitely write a small snippet, if you feel like it. In that case, add the `snippet` label to the issue, and keep it open till you are done, otherwise, close it.