v923z/micropython-ulab #464: [BUG] CP docs don't have ulab.numpy.zeros(), .ones()

jepler, 2021-12-29:
.. reported by todbot on Discord.

We expect it to appear here:

https://circuitpython.readthedocs.io/en/latest/shared-bindings/ulab/numpy/index.html

It can be tested locally with `./build-cp.sh` followed by viewing the local file `.../circuitpython/_build/html/shared-bindings/ulab/numpy/index.html` in browser.

v923z, 2021-12-30:
I think all functions of https://github.com/v923z/micropython-ulab/blob/master/code/ulab_create.h are missing from the documentation. While we are at it, I was wondering, whether we could move this particular file to https://github.com/v923z/micropython-ulab/tree/master/code/numpy. I would like to have the root clean, as much as possible.

v923z, 2021-12-30:
@jepler I have tried to fix the file containing the functions creating the arrays. When I run `./build-cp.sh`, `index.rst` is generated, and it has the documentation for the functions in question, but I get an error 
```
Warning, treated as error:
/home/v923z/sandbox/ulab/ulab/circuitpython/shared-bindings/ulab/numpy/approx/index.rst:document isn't included in any toctree
```
so the html file doesn't see the light of day. 

In my infinite wisdom, I implemented the changes in the `complex2` branch. Sorry for that. I really plan on releasing the new features in a day or two, so this should not lead to any delays.

jepler, 2021-12-30:
Not sure, but in circuitpython's `shared-bindings/index.rst` try changing `2` to `3`, since there are now modules of depth 3 that need documentation...
```
Modules
---------

.. toctree::
    :glob:
    :maxdepth: 2

    ../shared-bindings/*/index
    help
```

v923z, 2022-01-01:
@jepler A related question, in the big clean-up: where should the methods (like `.reshape`, `.tobytes` etc.) of an `ndarray` be documented? These are implemented in https://github.com/v923z/micropython-ulab/blob/master/code/ndarray.c, but there is no documentation there. This question most probably pertains to the attributes/properties, like `.shape`, `.itemsize`, and similar, too.

jepler, 2022-01-01:
Probably under `class ndarray` in `ulab.numpy`?

v923z, 2022-01-06:
@jepler I am not sure I get what I am supposed to get. I have changed 
```
.. toctree::
    :glob:
    :maxdepth: 2

    ../shared-bindings/*/index
    help
```
to 

```
.. toctree::
    :glob:
    :maxdepth: 3

    ../shared-bindings/*/index
    help
```
and that does not fix the problem. Moreover, `ls circuitpython/_build/html/shared-bindings/ulab` returns 
```
approx      compare  fft     __init__.html  numerical  vector
array.html  extras   filter  linalg         poly
```
and both html files are virtually empty, they have only the head. I don't have a `numpy` directory. I don't know, whether the two issues are linked.

As far as I understand, you use [build-cp.sh](https://github.com/v923z/micropython-ulab/blob/master/build-cp.sh) to compile the firmware for `circuitpython`, so I am a bit baffled by the fact that I can't locally reproduce the issue that you report here.

closed 2022-01-08 by v923z