v923z/micropython-ulab #467: [FEATURE REQUEST] add numpy.save

v923z, 2022-01-02:
As described in https://github.com/v923z/micropython-ulab/issues/466, saving data in a standard binary format could be beneficial. 
`numpy.save` without the `allow_pickle` keyword argument could be a solution: https://numpy.org/doc/stable/reference/generated/numpy.save.html

The file format specifications can be found under https://numpy.org/doc/stable/reference/generated/numpy.lib.format.html#module-numpy.lib.format

@labodezao

labodezao, 2022-01-02:
So maybe the numpy.load could be also intersting :)
Thanks a lot for all you do for porting this to opensource !
❤️

v923z, 2022-01-03:
> So maybe the numpy.load could be also intersting :)

That is a bit harder, because then you have to make sure that you can interpret, what someone else saved.

water5, 2022-01-12:
Build not pass?

>[ 97%] Building C object esp-idf/main/CMakeFiles/__idf_main.dir/home/mc3/micropython-ulab/code/numpy/linalg/linalg_tools.c.obj
>[ 97%] Building C object esp-idf/main/CMakeFiles/__idf_main.dir/home/mc3/micropython-ulab/code/numpy/ndarray/ndarray_iter.c.obj
>/home/mc3/micropython-ulab/code/numpy/io/io.c: In function 'io_save':
>/home/mc3/micropython-ulab/code/numpy/io/io.c:96:45: error: format '%ld' expects argument of type 'long int', but argument 3 has type 'size_t' {aka 'unsigned int'} [-Werror=format=]
         offset += sprintf(buffer+offset, "%ld,", ndarray->shape[ULAB_MAX_DIMS - 1]);
                                           ~~^    ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
                                           %d
>/home/mc3/micropython-ulab/code/numpy/io/io.c:99:49: error: format '%ld' expects argument of type 'long int', but argument 3 has type 'size_t' {aka 'unsigned int'} [-Werror=format=]
             offset += sprintf(buffer+offset, "%ld, ", ndarray->shape[ULAB_MAX_DIMS - i]);
                                               ~~^     ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
                                               %d
>/home/mc3/micropython-ulab/code/numpy/io/io.c:101:45: error: format '%ld' expects argument of type 'long int', but argument 3 has type 'size_t' {aka 'unsigned int'} [-Werror=format=]
         offset += sprintf(buffer+offset, "%ld", ndarray->shape[ULAB_MAX_DIMS - 1]);
                                           ~~^   ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
                                           %d
>[ 97%] Building C object esp-idf/main/CMakeFiles/__idf_main.dir/home/mc3/micropython-ulab/code/numpy/numerical.c.obj
>cc1: some warnings being treated as errors
make[3]: *** [esp-idf/main/CMakeFiles/__idf_main.dir/build.make:3785: esp-idf/main/CMakeFiles/__idf_main.dir/home/mc3/micropython-ulab/code/numpy/io/io.c.obj] Error 1
>make[3]: *** Waiting for unfinished jobs....
>[ 97%] Building C object esp-idf/main/CMakeFiles/__idf_main.dir/home/mc3/micropython-ulab/code/numpy/numpy.c.obj
>[ 97%] Building C object esp-idf/main/CMakeFiles/__idf_main.dir/home/mc3/micropython-ulab/code/numpy/poly.c.obj
>make[3]: Leaving directory '/home/mc3/micropython/ports/esp32/build-GENERIC'
>make[2]: *** [CMakeFiles/Makefile2:4683: esp-idf/main/CMakeFiles/__idf_main.dir/all] Error 2
>make[2]: Leaving directory '/home/mc3/micropython/ports/esp32/build-GENERIC'
>make[1]: *** [Makefile:136: all] Error 2
>make[1]: Leaving directory '/home/mc3/micropython/ports/esp32/build-GENERIC'
>make failed with exit code 2
>make: *** [Makefile:34: all] Error 2

v923z, 2022-01-12:
@water5 Thanks for looking into this! I think I know what the problem is. I will fix it later today.

v923z, 2022-01-12:
@water5 Could you, please, check out the latest commit on https://github.com/v923z/micropython-ulab/tree/io? I believe that should fix the issue. Let me know if you still have difficulties with the ESP32.

water5, 2022-01-12:
Build error still.
You mean build success on other boards except ESP32 before fix it?

>[ 97%] Building C object esp-idf/main/CMakeFiles/__idf_main.dir/home/mc3/micropython-ulab/code/numpy/numpy.c.obj
>[ 97%] Building C object esp-idf/main/CMakeFiles/__idf_main.dir/home/mc3/micropython-ulab/code/numpy/poly.c.obj
>/home/mc3/micropython-ulab/code/numpy/io/io.c: In function 'io_load':
>/home/mc3/micropython-ulab/code/numpy/io/io.c:190:7: error: 'endianness' may be used uninitialized in this function [-Werror=maybe-uninitialized]
     if((native_endianness != endianness) && (dtype != NDARRAY_UINT8) && (dtype != NDARRAY_INT8)) {
       ^
>[ 97%] Building C object esp-idf/main/CMakeFiles/__idf_main.dir/home/mc3/micropython-ulab/code/numpy/stats.c.obj
>cc1: some warnings being treated as errors
>make[3]: *** [esp-idf/main/CMakeFiles/__idf_main.dir/build.make:3785: esp-idf/main/CMakeFiles/__idf_main.dir/home/mc3/micropython-ulab/code/numpy/io/io.c.obj] Error 1
>make[3]: *** Waiting for unfinished jobs....
>[ 98%] Building C object esp-idf/main/CMakeFiles/__idf_main.dir/home/mc3/micropython-ulab/code/numpy/transform.c.obj
>make[3]: Leaving directory '/home/mc3/micropython/ports/esp32/build-GENERIC'
>make[2]: *** [CMakeFiles/Makefile2:4683: esp-idf/main/CMakeFiles/__idf_main.dir/all] Error 2
>make[2]: Leaving directory '/home/mc3/micropython/ports/esp32/build-GENERIC'
>make[1]: *** [Makefile:136: all] Error 2
>make[1]: Leaving directory '/home/mc3/micropython/ports/esp32/build-GENERIC'
>make failed with exit code 2
>make: *** [Makefile:34: all] Error 2

v923z, 2022-01-12:
@water5 One of the compilers was a bit picky, I noticed that this commit failed on github, too. I think the latest version looks more promising. Can you check it out?

closed 2022-01-19 by v923z