v923z/micropython-ulab #468: [FEATURE REQUEST] numpy.genfromtxt

water5, 2022-01-03:
[https://numpy.org/doc/stable/reference/generated/numpy.genfromtxt.html](https://numpy.org/doc/stable/reference/generated/numpy.genfromtxt.html)

Just suggestion. :)

v923z, 2022-01-03:
@water5 This function has at least 20 keyword arguments. No way can we support all. So, if you want this to be included, then you have to tell, what the minimal version should be able to handle.

water5, 2022-01-03:
I think the four arguments is minimal requires:
`dtype`, `delimiter`, `usecols`, `max_rows`

If possible, implement these will better:
`unpack`, `skip_header`, `skip_footer`, `names`

v923z, 2022-01-03:
Do you think we could change this to `loadtxt` https://numpy.org/doc/stable/reference/generated/numpy.loadtxt.html ? That 
function is simpler. I believe, the implementation is not terribly difficult, only a bit tedious. I could look into it in a couple of weeks. 
Do you know, whether `genfromtxt`, or `loadtxt` always return a matrix, even if you read out a single column/line?

water5, 2022-01-04:
Yes, `numpy.loadtxt` is a good choice for instead `numpy.genfromtxt`, I overlook this before.

v923z, 2022-01-04:
@water5 OK, thanks!

water5, 2022-01-11:
Is it possible implement `numpy.savetxt` too?
[https://numpy.org/doc/stable/reference/generated/numpy.savetxt.html](https://numpy.org/doc/stable/reference/generated/numpy.savetxt.html)

Although `numpy.savetxt` only support 1D or 2D array, and `numpy.save` will implement coming soon, unless implement `numpy.load` too, 
[https://numpy.org/doc/stable/reference/generated/numpy.load.html](https://numpy.org/doc/stable/reference/generated/numpy.load.html)
[https://github.com/v923z/micropython-ulab/issues/467#issuecomment-1003900940](https://github.com/v923z/micropython-ulab/issues/467#issuecomment-1003900940)

Otherwise, `numpy.loadtxt` and `numpy.savetxt` will implement complete minimum save / load function.

v923z, 2022-01-11:
> Is it possible implement `numpy.savetxt` too? https://numpy.org/doc/stable/reference/generated/numpy.savetxt.html
> 
> Although `numpy.savetxt` only support 1D or 2D array, and `numpy.save` will implement coming soon, unless implement `numpy.load` too, https://numpy.org/doc/stable/reference/generated/numpy.load.html [#467 (comment)](https://github.com/v923z/micropython-ulab/issues/467#issuecomment-1003900940)

I am leaning towards `numpy.save/numpy.load`, because it is easier to implement, and more versatile. It is not restricted to 2D arrays, and consumes less space. 

By the way, `numpy.save` is already implemented in https://github.com/v923z/micropython-ulab/pull/471. If you find any problems with it, let me know!

v923z, 2022-01-12:
@water5 If https://github.com/v923z/micropython-ulab/pull/471 is successfully merged, can we close this issue, or you really want to have `loadtxt`? I believe, it would be significantly harder to implement than `load`.

water5, 2022-01-13:
I think sometimes, `.txt` file have better compatibility, likes non-python application(e.g. Excel-csv, other programming language) output data,
Can we set this issue/feature request lower priority, solve it later, and could enable / disable via config file?

v923z, 2022-01-13:
> I think sometimes, `.txt` file have better compatibility, likes non-python application(e.g. Excel-csv, other programming language) output data, 

This is definitely true.

> Can we set this issue/feature request lower priority, solve it later, and could enable / disable via config file?

I can implement this in a week or so, at least, with the first four keyword arguments. We can polish it later. But I would like to do this in conjunction with `savetxt`.

water5, 2022-01-13:
I looking numpy json like this:
[https://github.com/Crimson-Crow/json-numpy](https://github.com/Crimson-Crow/json-numpy)

Can you have a look, how hard to implement this?

v923z, 2022-01-13:
@water5 I think this is pretty easy. Here is the key: https://github.com/Crimson-Crow/json-numpy/blob/e8129b0ff7b2dbb124a87223275231defecb9043/json_numpy.py#L10-L13

Basically, you need three things: the base64-encoded content of the array. The content you can get by calling
```
a = np.array(...)
a.tobytes()
```
(Note that the array must be dense: [.tobytes()](https://micropython-ulab.readthedocs.io/en/latest/ulab-ndarray.html))

Then you need the string representation of the shape. You can get that by calling
```
a.shape.__repr__
```
and then you need the descriptor of the `dtype`. At the moment, there is no function for that, but you can easily write one in python. You need something like 
```
def dtype_to_descr(a):
   if a.dtype == 'h':
       return '<i8'
   else if a.dtype == 'H':
       return '<i16'
...
```
I don't know, whether the base64 encoding is implemented in `micropython`: https://github.com/Crimson-Crow/json-numpy/blob/e8129b0ff7b2dbb124a87223275231defecb9043/json_numpy.py#L2 If it isn't, let me know, and we can implement the complete function in C.

water5, 2022-01-13:
But line 5, require `numpy.lib.format.dtype_to_descr` and `numpy.lib.format.descr_to_dtype`

[https://numpy.org/doc/stable/reference/generated/numpy.lib.format.descr_to_dtype.html](https://numpy.org/doc/stable/reference/generated/numpy.lib.format.descr_to_dtype.html)
[https://numpy.org/doc/stable/reference/generated/numpy.lib.format.dtype_to_descr.html](https://numpy.org/doc/stable/reference/generated/numpy.lib.format.dtype_to_descr.html)

And line 9, require `numpy.generic`.

v923z, 2022-01-13:
> But line 5, require `numpy.lib.format.dtype_to_descr` and `numpy.lib.format.descr_to_dtype`
> 

`dtype_to_descr` simply returns a short string, as I indicated in the earlier post, describing the `dtype`. So, an `<i8`, `<u16` etc., depending on the `dtype`.

> And line 9, require `numpy.generic`.

That is in a test, you don't actually need that, because you know beforehand that you want to convert an `ndarray`. That is there, so that you don't try to convert a list, or something similar. You can safely skip it. You really need only these three lines:

```
{
'__numpy__': b64encode(obj.tobytes()).decode('ascii'),
'dtype': dtype_to_descr(obj.dtype),
'shape': obj.shape
}
```
which is a simple dictionary, which you can serialise. The only question is, whether you have `b64encode`. We could add that here, if you need it. Actually, this is much easier to implement, than even `savetxt`.

water5, 2022-01-13:
>I don't know, whether the base64 encoding is implemented in micropython: https://github.com/Crimson-Crow/json-numpy/blob/e8129b0ff7b2dbb124a87223275231defecb9043/json_numpy.py#L2 If it isn't, let me know, and we can implement the complete function in C.

Yes, in `ubinascii`, `a2b_base64`, `b2a_base64`.

water5, 2022-01-14:
JSON very useful when transfer via network.

water5, 2022-01-14:
Additional, out of the thread: Can you post pull request, add support `newline` parameter to `ubinascii`?
[https://github.com/micropython/micropython/blob/master/extmod/modubinascii.c#L168](https://github.com/micropython/micropython/blob/master/extmod/modubinascii.c#L168)

refer:
[https://docs.python.org/3/library/binascii.html#binascii.b2a_base64](https://docs.python.org/3/library/binascii.html#binascii.b2a_base64)

Sometimes, the newline char annoying, I'm not skilled in C.

v923z, 2022-01-14:
> Additional, out of the thread: Can you post pull request, add support `newline` parameter to `ubinascii`? https://github.com/micropython/micropython/blob/master/extmod/modubinascii.c#L168
>
> refer: [...]

I think the fix shouldn't be hard, but it seems to me that this is by design: https://github.com/micropython/micropython/pull/3254 You could raise an issue at `micropython`, if you think that this is important, and then I can pick it up from there.

water5, 2022-01-21:
I see a `io-savetxt` branch created, is it finished? can I use that right now?

v923z, 2022-01-21:
> I see a `io-savetxt` branch created, is it finished? can I use that right now?

See it for yourself:)

`savetxt` should be OK, I am still working on `loadtxt`.

water5, 2022-01-21:
Is it hard to add `delimiter` parameter? default/currently is `space` char, sometimes conflict other chars.
Alsl `loadtxt`.

v923z, 2022-01-21:
@water5 I don't quite see what the problem is. Here is the test script, quite a few of the keywords are implemented: https://github.com/v923z/micropython-ulab/blob/io-savetxt/tests/2d/numpy/savetxt.py

water5, 2022-01-21:
Oh, I use wrong parameter sequence.

v923z, 2022-01-27:
@water5 I think https://github.com/v923z/micropython-ulab/pull/492 implements `loadtxt` as requested. Here is the test script till I find the time to write the documentation: https://github.com/v923z/micropython-ulab/blob/io-savetxt/tests/2d/numpy/loadtxt.py

water5, 2022-01-29:
Default/currently `ulab.numpy.loadtxt` `dtype` parameter not implement, unspecific `dtype` will cause uncertain error, 
```python
a = np.arange(100, dtype = np.uint16).reshape((10, 10))
a
```
>array([[0, 1, 2, 3, 4, 5, 6, 7, 8, 9],
>       [10, 11, 12, 13, 14, 15, 16, 17, 18, 19],
>       [20, 21, 22, 23, 24, 25, 26, 27, 28, 29],
>       [30, 31, 32, 33, 34, 35, 36, 37, 38, 39],
>       [40, 41, 42, 43, 44, 45, 46, 47, 48, 49],
>       [50, 51, 52, 53, 54, 55, 56, 57, 58, 59],
>       [60, 61, 62, 63, 64, 65, 66, 67, 68, 69],
>       [70, 71, 72, 73, 74, 75, 76, 77, 78, 79],
>       [80, 81, 82, 83, 84, 85, 86, 87, 88, 89],
>       [90, 91, 92, 93, 94, 95, 96, 97, 98, 99]], `dtype=uint16`)
```python
np.savetxt('s.txt', a)
np.loadtxt('s.txt')
```
>array([[0.0, 1.0, 2.0, 3.0, 4.0, 5.0, 6.0, 7.0, 8.0, 9.0],
>       [10.0, 11.0, 12.0, 13.0, 14.0, 15.0, 16.0, 17.0, 18.0, 19.0],
>       [20.0, 21.0, 22.0, 23.0, 24.0, 25.0, 26.0, 27.0, 28.0, 29.0],
>       [30.0, 31.0, 32.0, 33.0, 34.0, 35.0, 36.0, 37.0, 38.0, 39.0],
>       [40.0, 41.0, 42.0, 43.0, 44.0, 45.0, 46.0, 47.0, 48.0, 49.0],
>       [50.0, 51.0, 52.0, 53.0, 54.0, 55.0, 56.0, 57.0, 58.0, 59.0],
>       [60.0, 61.0, 62.0, 63.0, 64.0, 65.0, 66.0, 67.0, 68.0, 69.0],
>       [70.0, 71.0, 72.0, 73.0, 74.0, 75.0, 76.0, 77.0, 78.0, 79.0],
>       [80.0, 81.0, 82.0, 83.0, `84.00001`, 85.0, 86.0, 87.0, 88.0, `89.00001`],
>       [90.0, 91.0, 92.0, 93.0, `94.00001`, 95.0, 96.0, 97.0, 98.0, `99.00001`]], `dtype=float32`)
```python
f = open('s.txt', 'r')
f.read()
```
>'0.000000 1.000000 2.000000 3.000000 4.000000 5.000000 6.000000 7.000000 8.000000 9.000000\n10.000000 11.000000 12.000000 13.000000 `13.999999` 15.000000 16.000000 17.000000 `18.000001` 19.000000\n20.000000 `21.000001` 22.000000 `22.999999` `24.000001` 25.000000 `26.000001` 27.000000 `27.999999` `29.000001`\n30.000000 `31.000001` 32.000000 `32.999999` `34.000001` 35.000000 `36.000001` 37.000000 `37.999999` `39.000001`\n40.000000 `40.999999` `42.000002` `43.000001` `44.000001` 45.000000 `45.999999` `47.000002` `48.000001` `49.000001`\n50.000000 `50.999999` `52.000002` `53.000001` `54.000001` 55.000000 `55.999999` `57.000002` `58.000001` `59.000001`\n60.000000 `60.999999` `62.000002` `63.000001` `64.000001` 65.000000 `65.999999` `67.000002` `68.000001` `69.000001`\n70.000000 `70.999999` `72.000002` `73.000001` `74.000001` 75.000000 `75.999999` `77.000002` `78.000001` `79.000001`\n80.000000 `81.000003` `81.999998` `83.000001` `84.000005` 85.000000 `86.000003` `86.999998` `88.000001` `89.000005`\n90.000000 `91.000003` `91.999998` `93.000001` `94.000005` 95.000000 `96.000003` `96.999998` `98.000001` `99.000005`\n'

What's wrong and how to solve `ulab.numpy.savetxt` above situation?

v923z, 2022-01-29:
> Default/currently `ulab.numpy.loadtxt` `dtype` parameter not implement, 

But this is specifically mentioned in the [docs](https://micropython-ulab.readthedocs.io/en/latest/numpy-functions.html?highlight=savetxt#savetxt)

> unspecific `dtype` will cause uncertain error,

> What's wrong and how to solve `ulab.numpy.savetxt` above situation?

I guess, you are concerned about `98.000001`, which should be `98.0`. But this is a representation error. You have 32 bits to represent a floating point number, so this is what you get. I can't really do anything about it. If you try the same on the `unix` port with 64-bit resolution, your errors would be significantly smaller.

water5, 2022-01-29:
> I guess, you are concerned about 98.000001, which should be 98.0. But this is a representation error. You have 32 bits to represent a floating point number, so this is what you get. I can't really [...]

OK, I try that on `unix` port the result is better.

--------------

>>Default/currently ulab.numpy.loadtxt dtype parameter not implement,
>
>But this is specifically mentioned in the [docs](https://micropython-ulab.readthedocs.io/en/latest/numpy-functions.html?highlight=savetxt#savetxt)

That just specific origin array `dtype`, when create a new array load from `.txt` file, the `dtype` of new array is `float`, and can't convert to other(e.g. `uint8`, `uint16`).

`numpy.loadtxt` have `dtype` parameter, 
[https://numpy.org/doc/stable/reference/generated/numpy.loadtxt.html](https://numpy.org/doc/stable/reference/generated/numpy.loadtxt.html)

Or / and do that by other way, can we implement `numpy.matrix.astype`?
[https://numpy.org/doc/stable/reference/generated/numpy.matrix.astype.html](https://numpy.org/doc/stable/reference/generated/numpy.matrix.astype.html)
```python
a = np.arange(3, dtype = np.float32)
a
```
>array([0., 1., 2.], dtype=float32)
```python
a = a.astype(np.uint16)
a
```
>array([0, 1, 2], dtype=uint16)

v923z, 2022-01-29:
> OK, I try that on `unix` port the result is better.
>
>
> That just specific origin array `dtype`, when create a new array load from `.txt` file, the `dtype` of new array is `float`, and can't convert [...]

I really don't understand what the problem is. This always works

```
a = np.array([1, 2, 3])
print(a)
>>> array([1.0, 2.0, 3.0], dtype=float64)
b = np.array(a, dtype=np.uint8)
print(b)
>>> array([1, 2, 3], dtype=uint8)
```
i.e., you can always convert the `dtype` of an array, if you pass the array to the constructor, and specify the output `dtype`. 

> `numpy.loadtxt` have `dtype` parameter, https://numpy.org/doc/stable/reference/generated/numpy.loadtxt.html

I know that, but this adds a lot to the firmware size, and you can do the conversion later, anyway. As I pointed out many times, duplicating functionality is not a viable option. If something can already be done, then I don't see the point in implementing it differently.

> Or / and do that by other way, can we implement `numpy.matrix.astype`? https://numpy.org/doc/stable/reference/generated/numpy.matrix.astype.html

This is simply an alias for what I described above. Does this have any benefits over that?

v923z, 2022-01-29:
> `numpy.loadtxt` have `dtype` parameter, https://numpy.org/doc/stable/reference/generated/numpy.loadtxt.html

https://github.com/v923z/micropython-ulab/pull/500 implements the `dtype` keyword argument.

water5, 2022-01-30:
Thanks for implement relate functions.

water5, 2022-01-30:
I think this implement is worth and necessary, actually, most time MicroPython use for microcontroller, not for 64bit system, 
```python
a = np.loadtxt('s.txt')
a
```
>array([[0.0, 1.0, 2.0, 3.0, 4.0, 5.0, 6.0, 7.0, 8.0, 9.0],
>       [10.0, 11.0, 12.0, 13.0, 14.0, 15.0, 16.0, 17.0, 18.0, 19.0],
>       [20.0, 21.0, 22.0, 23.0, 24.0, 25.0, 26.0, 27.0, 28.0, 29.0],
>       [30.0, 31.0, 32.0, 33.0, 34.0, 35.0, 36.0, 37.0, 38.0, 39.0],
>       [40.0, 41.0, 42.0, 43.00001, 44.00001, 45.0, 46.0, 47.0, 48.0, 49.0],
>       [50.0, 51.0, 52.0, 53.0, 53.99999, 55.0, 56.0, 56.99999, 58.00001, 59.00001],
>       [60.0, 61.0, 62.0, 63.00001, 64.0, 65.0, 66.0, 67.0, 68.0, 69.0],
>       [70.0, 71.0, 72.0, 73.00001, 74.0, 75.0, 76.0, 77.00001, 77.99999, 79.00001],
>       [80.0, 81.0, 82.0, 83.0, 84.00001, 84.99999, 86.00001, 87.00001, 88.00001, 89.0],
>       [90.0, 91.00001, 92.0, 93.0, 94.00001, 94.99999, 96.0, 97.0, 97.99999, 99.00001]], dtype=float32)
```python
a = np.array(a, dtype = np.uint16)
a
```
>array([[0, 1, 2, 3, 4, 5, 6, 6, 8, 9],
>       [10, 11, `12`, `12`, `13`, 15, 16, 17, 18, 19],
>       [20, 21, 22, 23, 24, `25`, `25`, 26, 27, 29],
>       [`30`, `30`, 32, 33, `34`, `34`, 36, 37, `38`, `38`],
>       [40, 41, 42, 43, 44, 45, 46, 47, `48`, `48`],
>       [50, 51, 52, `53`, `53`, `55`, `55`, 56, 58, 59],
>       [60, `61`, `61`, 63, `64`, `64`, 66, 67, 68, 69],
>       [69, 71, 72, 73, 74, 75, 76, `77`, `77`, 79],
>       [80, 81, 82, 83, `84`, `84`, 86, 87, `88`, `88`],
>       [90, 91, 92, 93, `94`, `94`, 96, `97`, `97`, 99]], dtype=uint16)

Seems use `dtype` parameter have not above issue, I will try more times later, 
```python
np.loadtxt('s.txt', dtype = np.uint16)
```
> array([[0, 1, 2, 3, 4, 5, 6, 7, 8, 9],
>       [10, 11, 12, 13, 14, 15, 16, 17, 18, 19],
>       [20, 21, 22, 23, 24, 25, 26, 27, 28, 29],
>       [30, 31, 32, 33, 34, 35, 36, 37, 38, 39],
>       [40, 41, [...]

v923z, 2022-01-30:
@water5 On the `unix` port, I don't see the error that you are reporting. Here is my test script

```
a = np.arange(100, dtype=np.float).reshape((10, 10))
print(a)
np.savetxt('loadtxt.dat', a)
print(np.loadtxt('loadtxt.dat', dtype=np.uint16))
```
and here is the output:
```
array([[0.0, 1.0, 2.0, 3.0, 4.0, 5.0, 6.0, 7.0, 8.0, 9.0],
       [10.0, 11.0, 12.0, 13.0, 14.0, 15.0, 16.0, 17.0, 18.0, 19.0],
       [20.0, 21.0, 22.0, 23.0, 24.0, 25.0, 26.0, 27.0, 28.0, 29.0],
       [30.0, 31.0, 32.0, 33.0, 34.0, 35.0, 36.0, 37.0, 38.0, 39.0],
       [40.0, 41.0, 42.0, 43.0, 44.0, 45.0, 46.0, 47.0, 48.0, 49.0],
       [50.0, 51.0, 52.0, 53.0, 54.0, 55.0, 56.0, 57.0, 58.0, 59.0],
       [60.0, 61.0, 62.0, 63.0, 64.0, 65.0, 66.0, 67.0, 68.0, 69.0],
       [70.0, 71.0, 72.0, 73.0, 74.0, 75.0, 76.0, 77.0, 78.0, 79.0],
       [80.0, 81.0, 82.0, 83.0, 84.00001, 85.0, 86.0, 87.0, 88.0, 89.00001],
       [90.0, 91.0, 92.0, 93.0, 94.00001, 95.0, 96.0, 97.0, 98.0, 99.00001]], dtype=float32)

array([[0, 1, 2, 3, 4, 5, 6, 7, 8, 9],
       [10, 11, 12, 13, 14, 15, 16, 17, 18, 19],
       [20, 21, 22, 23, 24, 25, 26, 27, 28, 29],
       [30, 31, 32, 33, 34, 35, 36, 37, 38, 39],
       [40, 41, 42, 43, 44, 45, 46, 47, 48, 49],
       [50, 51, 52, 53, 54, 55, 56, 57, 58, 59],
       [60, 61, 62, 63, 64, 65, 66, 67, 68, 69],
       [70, 71, 72, 73, 74, 75, 76, 77, 78, 79],
       [80, 81, 82, 83, 84, 85, 86, 87, 88, 89],
       [90, 91, 92, 93, 94, 95, 96, 97, 98, 99]], dtype=uint16)
```

What is your platform?

v923z, 2022-01-30:
@water5 Sorry, I think I have misinterpreted your comment. The error was not in the `loadtxt` function, but in the `array` initialiser, when you pass another array as the argument. I fixed it. Could you, please, check out https://github.com/v923z/micropython-ulab/pull/500 once more to see if that fixes the issue?

water5, 2022-01-30:
I use ESP32.
Sorry, supplement, I use the numeric in `.txt` file generate from `double precision float` implement of MicroPython, like these: 
>'0.000000000000000 1.000000000000000 2.000000000000000 3.000000000000000 4.000000000000000 5.000000000000000 6.000000000000000 7.000000000000000 8.000000000000000 9.000000000000000\n10.000000000000000 11.000000000000000 12.000000000000001 13.000000000000000 14.000000000000001 15.000000000000000 16.000000000000000 17.000000000000001 18.000000000000000 19.000000000000001\n20.000000000000000 21.000000000000000 22.000000000000001 23.000000000000002 24.000000000000003 25.000000000000000 26.000000000000000 27.000000000000001 28.000000000000003 29.000000000000003\n30.000000000000000 31.000000000000000 32.000000000000001 33.000000000000002 34.000000000000003 35.000000000000000 36.000000000000000 37.000000000000001 38.000000000000003 39.000000000000003\n40.000000000000000 41.000000000000005 42.000000000000001 42.999999999999998 44.000000000000003 45.000000000000000 46.000000000000005 47.000000000000001 48.000000000000007 49.000000000000003\n50.000000000000000 51.000000000000005 52.000000000000001 53.000000000000007 54.000000000000003 55.000000000000000 56.000000000000005 57.000000000000001 58.000000000000007 59.000000000000003\n60.000000000000000 61.000000000000005 62.000000000000001 63.000000000000007 64.000000000000003 65.000000000000000 66.000000000000005 67.000000000000001 68.000000000000007 69.000000000000003\n70.000000000000000 71.000000000000005 72.000000000000001 73.000000000000007 74.000000000000003 75.000000000000000 76.000000000000005 77.000000000000001 78.000000000000007 79.000000000000003\n80.000000000000000 80.999999999999996 82.000000000000010 83.000000000000007 84.000000000000003 85.000000000000000 85.999999999999996 87.000000000000010 88.000000000000007 89.000000000000003\n90.000000000000000 90.999999999999996 92.000000000000010 93.000000000000007 94.000000000000003 95.000000000000000 96.000000000000014 97.000000000000010 98.000000000000007 99.000000000000003\n'

I think this is MicroPython implement issue, on `single precision float` implement of MicroPython, even though use in `list`, have same problem: 
```python
a = [50.000000000000000, 51.000000000000005, 52.000000000000001, 53.000000000000007, 54.000000000000003, 55.000000000000000, 56.000000000000005, 57.000000000000001, 58.000000000000007, 59.000000000000003]
a
```
>[50.0, 51.0, 52.0, 53.0, 53.99999, 55.0, 56.0, 56.99999, 58.00001, 59.00001]

But, indeed, use `dtype` parameter seems have not above issue: 
```python
np.loadtxt('s.txt', dtype = np.uint16)
```
> array([[0, 1, 2, 3, 4, 5, 6, 7, 8, 9],
> [10, 11, 12, 13, 14, 15, 16, 17, 18, 19],
> [20, 21, 22, 23, 24, 25, 26, 27, 28, 29],
> [30, 31, 32, 33, 34, 35, 36, 37, 38, 39],
> [40, 41, 42, 43, 44, 45, 46, 47, [...]

water5, 2022-01-30:
And, other issue, when use `suppress scientific notation` in `.txt` file: 
>0.000e+00, 1.000e+00, 2.000e+00, 3.000e+00, 4.000e+00, 5.000e+00, 6.000e+00, 7.000e+00, 8.000e+00, 9.000e+00
>1.000e+01, 1.100e+01, 1.200e+01, 1.300e+01, 1.400e+01, 1.500e+01, 1.600e+01, 1.700e+01, 1.800e+01, 1.900e+01
>2.000e+01, 2.100e+01, 2.200e+01, 2.300e+01, 2.400e+01, 2.500e+01, 2.600e+01, 2.700e+01, 2.800e+01, 2.900e+01
>3.000e+01, 3.100e+01, 3.200e+01, 3.300e+01, 3.400e+01, 3.500e+01, 3.600e+01, 3.700e+01, 3.800e+01, 3.900e+01
>4.000e+01, 4.100e+01, 4.200e+01, 4.300e+01, 4.400e+01, 4.500e+01, 4.600e+01, 4.700e+01, 4.800e+01, 4.900e+01
>5.000e+01, 5.100e+01, 5.200e+01, 5.300e+01, 5.400e+01, 5.500e+01, 5.600e+01, 5.700e+01, 5.800e+01, 5.900e+01
>6.000e+01, 6.100e+01, 6.200e+01, 6.300e+01, 6.400e+01, 6.500e+01, 6.600e+01, 6.700e+01, 6.800e+01, 6.900e+01
>7.000e+01, 7.100e+01, 7.200e+01, 7.300e+01, 7.400e+01, 7.500e+01, 7.600e+01, 7.700e+01, 7.800e+01, 7.900e+01
>8.000e+01, 8.100e+01, 8.200e+01, 8.300e+01, 8.400e+01, 8.500e+01, 8.600e+01, 8.700e+01, 8.800e+01, 8.900e+01
>9.000e+01, 9.100e+01, 9.200e+01, 9.300e+01, 9.400e+01, 9.500e+01, 9.600e+01, 9.700e+01, 9.800e+01, 9.900e+01

Will construct a incorrect `shape` of array: 
```python
np.loadtxt('s.txt', dtype = np.float, delimiter = ', ')
```
>array([[0.0, 1.0, 2.0, ..., 8.0, 9.0, 10.0],
>       [11.0, 12.0, 13.0, ..., 19.0, 20.0, 21.0],
>       [22.0, 23.0, 24.0, ..., 30.0, 31.0, 32.0],
>       [33.0, 34.0, 35.0, ..., 41.0, 42.0, 43.0],
>       [44.0, 45.0, 46.0, ..., 52.0, 53.0, 54.0],
>       [55.0, 56.0, 57.0, ..., 63.0, 64.0, 65.0],
>       [66.0, 67.0, 68.0, ..., 74.0, 75.0, 76.0],
>       [77.0, 78.0, 79.0, ..., 85.0, 86.0, 87.0],
>       [88.0, 89.00001, 90.0, ..., 96.0, 97.0, 98.0],
>       [99.00001, 0.0, 0.0, ..., 0.0, 0.0, 0.0]], dtype=float32)

v923z, 2022-01-30:
> Will construct a incorrect `shape` of array:
> 
> ```python
> np.loadtxt('s.txt', dtype = np.float, delimiter = ', ')
> ```

Even if you have `np.loadtxt('s.txt', dtype = np.float, delimiter = ',')`? (Note the missing space in `delimiter`.)

v923z, 2022-01-30:
> I think this is MicroPython implement issue, on `single precision float` implement of MicroPython, even though use in `list`, have same problem:
>
> ```python
> a = [50.000000000000000, [...]
That's correct. It is not quite trivial to correctly print a floating point number. You can see how much goes into this here: https://github.com/micropython/micropython/blob/4d9e657f0ee881f4a41093ab89ec91d03613744d/py/formatfloat.c#L114

water5, 2022-01-30:
>Even if you have np.loadtxt('s.txt', dtype = np.float, delimiter = ',')? (Note the missing space in delimiter.)

Use
```python
np.loadtxt('s.txt', dtype = np.float, delimiter = ', ')
```
and 
```python
np.loadtxt('s.txt', dtype = np.float, delimiter = ',')
```
have same problem, Will construct a incorrect `shape` of array.

water5, 2022-01-30:
And, can you implement `skiprows` parameter too? when data file is huge, that's very useful.

v923z, 2022-01-30:
I think I know what the error was. I have updated the PR.

v923z, 2022-01-30:
> And, can you implement `skiprows` parameter too? when data file is huge, that's very useful.

Yes, that could be added.

water5, 2022-01-31:
>I think I know what the error was. I have updated the PR.

Sorry, I use `.txt` file generate from `Windows platform`, it use `\r\n` for newline, `Linux` and `MicroPython` use `\n`, `CPython` on `Linux` could convert that to `\n` automatically, but `MicroPython` not, I think this is `MicroPython` compatibility issue.

I use last version of `loadtxt-fix` branch, same issue still exist, then I modify the `.txt` file use `\n`, array `shape` correct. so, I guess previous version of `loadtxt-fix` is correct, but I'm not ensure, I have not the previous of firmware, is it need to rollback?

water5, 2022-01-31:
Additional, do you know someone already post relate issue for `MicroPython`?

v923z, 2022-01-31:
> Sorry, I use `.txt` file generate from `Windows platform`, it use `\r\n` for newline, `Linux` and `MicroPython` use `\n`, `CPython` on `Linux` could convert that to `\n` automatically, but [...]

No, we don't need to roll back. Thanks for the update!

> Additional, do you know someone already post relate issue for `MicroPython`?

I don't know, but you could ask of the forum. This issue must come up quite often.

v923z, 2022-01-31:
> Sorry, I use `.txt` file generate from `Windows platform`, it use `\r\n` for newline, `Linux` and `MicroPython` use `\n`, `CPython` on `Linux` could convert that to `\n` automatically, but [...]

Why don't you use `save/load`? That works with binary files, so you shouldn't have this problem with the linebreak/linefeed. On the flip side, you have to read the whole file, you can't skip lines.

v923z, 2022-01-31:
@water5 I have added the `skiprows` keyword in https://github.com/v923z/micropython-ulab/pull/500. I haven't yet updated the docs, though.

v923z, 2022-01-31:
> > Sorry, I use `.txt` file generate from `Windows platform`, it use `\r\n` for newline, `Linux` and `MicroPython` use `\n`, `CPython` on `Linux` could convert that to `\n` automatically, but `MicroPython` not, I think this is `MicroPython` compatibility issue.

In principle, it would not be impossible to treat `\r\n`, but the problem is that these are actually two characters, and the buffer's boundary might be between them, so we would have to carry over the `\r` character to the next iteration. This happens here: 
https://github.com/v923z/micropython-ulab/blob/f5f42c364349a4dae571fe091dd6c61654a6a7c7/code/numpy/io/io.c#L332-L359

We read in 127 characters in one round, and then analyze the content. But it could just happen that `buffer[126] = '\r'`, and then we would have to carry this piece of information over to the next iteration, where `buffer[0]` should be equal to `\n`. This is all a bit hackish, so if it is not prohibitively hard for you, I would just recommend that after generating the files on windows, you either convert your files, or, if you use `numpy`, you could even specify the `newline` character: https://numpy.org/doc/stable/reference/generated/numpy.savetxt.html

water5, 2022-02-01:
>I would just recommend that after generating the files on windows, you either convert your files.

This a choice to solve the issue, actually, `ulab.numpy.loadtxt` enough to solve most basic functions currently, I will close this issue, Thanks.

>or, if you use numpy, you could even specify the newline character.

This way not work, even though use `savetxt` 's `newline` parameter specific newline to use '\n' on `Windows platform`, when open the file on `MicroPython`, still parsing to `\r\n`, I think this depend OS.

But I curious, `numpy` on `CPython` is fine to handle that, whatever `\n` or `\r\n`, how to operate data file on `numpy`, use `Python built-in open function` or directly use `C` ?
[https://github.com/numpy/numpy/blob/v1.22.0/numpy/lib/npyio.py#L1042](https://github.com/numpy/numpy/blob/v1.22.0/numpy/lib/npyio.py#L1042)

v923z, 2022-02-01:
> But I curious, `numpy` on `CPython` is fine to handle that, whatever `\n` or `\r\n`, how to operate data file on `numpy`, use `Python built-in open function` or directly use `C` ? https://github.com/numpy/numpy/blob/v1.22.0/numpy/lib/npyio.py#L1042

It comes from here: https://github.com/numpy/numpy/blob/4adc87dff15a247e417d50f10cc4def8e1c17a03/numpy/lib/_datasource.py#L74

v923z, 2022-02-01:
Closed by https://github.com/v923z/micropython-ulab/pull/500

closed 2022-02-01 by v923z