v923z/micropython-ulab #469: [FEATURE REQUEST]how to install this ulab package in a calculator like Casio fx-CG50

PandaJohnWick, 2022-01-05:
**Describe the solution you'd like**
I already have microPython 1.9.4 in the calculator, how do I install this package in the calculator, especially since numpy is important for matrix operations?
**Additional context**
Add any other context about the feature request here.

v923z, 2022-01-05:
@PandaJohnWick You can't just install `ulab`, you have to compile the firmware for the specific platform, because `ulab` is very tightly integrated into the firmware. I don't know how you compile that for the Casio, but here are the steps for some of the more standard chips: https://github.com/v923z/micropython-ulab#firmware. I would suggest raising the question on the Casio user forum, if there is such a thing.

v923z, 2022-01-14:
Since this issue is not really relevant here, I am closing it.

closed 2022-01-14 by v923z