v923z/micropython-ulab #470: [FEATURE REQUEST] numpy.delete

water5, 2022-01-07:
[https://numpy.org/doc/stable/reference/generated/numpy.delete.html](https://numpy.org/doc/stable/reference/generated/numpy.delete.html)

I think this is a basic function.

v923z, 2022-01-07:
But can't this functionality be reproduced with slicing, or `compress`? https://numpy.org/doc/stable/reference/generated/numpy.compress.html `compress` is implemented in the yet unmerged PR https://github.com/v923z/micropython-ulab/pull/456.

water5, 2022-01-08:
In some use case, `numpy.compress` is better than `numpy.delete`, but when just need to delete several rows/columns of a big array, likes below:
```python
b = np.arange(10000).reshape(100, 100)
```
>array([[   0,    1,    2, ...,   97,   98,   99],
>       [ 100,  101,  102, ...,  197,  198,  199],
>       [ 200,  201,  202, ...,  297,  298,  299],
>       ...,
>       [9700, 9701, 9702, ..., 9797, 9798, 9799],
>       [9800, 9801, 9802, ..., 9897, 9898, 9899],
>       [9900, 9901, 9902, ..., 9997, 9998, 9999]])

```python
np.delete(b, 98, axis = 0)
```
>array([[   0,    1,    2, ...,   97,   98,   99],
>       [ 100,  101,  102, ...,  197,  198,  199],
>       [ 200,  201,  202, ...,  297,  298,  299],
>       ...,
>       [9600, 9601, 9602, ..., 9697, 9698, 9699],
>       [9700, 9701, 9702, ..., 9797, 9798, 9799],
>       [9900, 9901, 9902, ..., 9997, 9998, 9999]])

```python
np.delete(b, 98, axis = 1)
```
>array([[   0,    1,    2, ...,   96,   97,   99],
>       [ 100,  101,  102, ...,  196,  197,  199],
>       [ 200,  201,  202, ...,  296,  297,  299],
>       ...,
>       [9700, 9701, 9702, ..., 9796, 9797, 9799],
>       [9800, 9801, 9802, ..., 9896, 9897, 9899],
>       [9900, 9901, 9902, ..., 9996, 9997, 9999]])

If use `numpy.compress`, need to construct a 1-D array of bools first use for condition, when origin array huge, likes above case, it's not a easy job?

v923z, 2022-01-08:
> In some use case, `numpy.compress` is better than `numpy.delete`, but when just need to delete several rows/columns of a big array, likes below:
> 
> 
> If use `numpy.compress`, need to construct a 1-D array of bools first use for condition, when origin array huge, likes above case, it's not a easy job?

This is a good point, I completely overlooked that. We can definitely add `delete`. There are a couple of other functions in the pipeline, but I should be able to pull this off in a week.

v923z, 2022-01-10:
@water5 Could you, please, check out https://github.com/v923z/micropython-ulab/pull/473? I think the code should work (the tests passed). I haven't implemented indexing via slices, so you can do something like this:

```
np.delete(a, 1, axis=0)
np.delete(a, [1, 2, 3, 4], axis=0)
np.delete(a, (1, 2, 3, 4), axis=0)
np.delete(a, range(5), axis=0)
```
(Actually, any iterable should do in place of the indices, so you can simulate indices via the `range` function.)

Also note that the indices are copied once, because I have to sort them in increasing order. That means that `4*len(indices)` bytes are required to store the values temporarily. I don't know if that is acceptable, but I don't actually see another option, since there is no guarantee that the indices are sorted.

water5, 2022-01-11:
When just delete 1 row/column, 
must use `np.delete(a, [row/column], axis = 0)` or `np.delete(a, (row/column, ), axis = 0)` or `np.delete(a, [5, ], axis = 0)`, 

`np.delete(a, row/column, axis = 0)` or `np.delete(a, (row/column), axis = 0)` not work.

```python
a = np.arange(100).reshape((10, 10))
np.delete(a, 5, axis = 0)
```
>not work, 
>
>array([[0, 1, 2, 3, 4, 5, 6, 7, 8, 9],
>       [10, 11, 12, 13, 14, 15, 16, 17, 18, 19],
>       [30, 31, 32, 33, 34, 35, 36, 37, 38, 39],
>       [40, 41, 42, 43, 44, 45, 46, 47, 48, 49],
>       [50, 51, 52, 53, 54, 55, 56, 57, 58, 59],
>       [60, 61, 62, 63, 64, 65, 66, 67, 68, 69],
>       [70, 71, 72, 73, 74, 75, 76, 77, 78, 79],
>       [80, 81, 82, 83, 84, 85, 86, 87, 88, 89],
>       [90, 91, 92, 93, 94, 95, 96, 97, 98, 99]], dtype=int16)
----------------
```python
np.delete(a, (5), axis = 0)
```
>not work, 
>
>array([[0, 1, 2, 3, 4, 5, 6, 7, 8, 9],
>       [10, 11, 12, 13, 14, 15, 16, 17, 18, 19],
>       [30, 31, 32, 33, 34, 35, 36, 37, 38, 39],
>       [40, 41, 42, 43, 44, 45, 46, 47, 48, 49],
>       [50, 51, 52, 53, 54, 55, 56, 57, 58, 59],
>       [60, 61, 62, 63, 64, 65, 66, 67, 68, 69],
>       [70, 71, 72, 73, 74, 75, 76, 77, 78, 79],
>       [80, 81, 82, 83, 84, 85, 86, 87, 88, 89],
>       [90, 91, 92, 93, 94, 95, 96, 97, 98, 99]], dtype=int16)
-------------
```python
np.delete(a, (5, ), axis = 0)
```
>work, 
>
>array([[0, 1, 2, 3, 4, 5, 6, 7, 8, 9],
>       [10, 11, 12, 13, 14, 15, 16, 17, 18, 19],
>       [20, 21, 22, 23, 24, 25, 26, 27, 28, 29],
>       [30, 31, 32, 33, 34, 35, 36, 37, 38, 39],
>       [40, 41, 42, 43, 44, 45, 46, 47, 48, 49],
>       [60, 61, 62, 63, 64, 65, 66, 67, 68, 69],
>       [70, 71, 72, 73, 74, 75, 76, 77, 78, 79],
>       [80, 81, 82, 83, 84, 85, 86, 87, 88, 89],
>       [90, 91, 92, 93, 94, 95, 96, 97, 98, 99]], dtype=int16)
---------------
```python
np.delete(a, [5], axis = 0)
```
>work, 
>
>array([[0, 1, 2, 3, 4, 5, 6, 7, 8, 9],
>       [10, 11, 12, 13, 14, 15, 16, 17, 18, 19],
>       [20, 21, 22, 23, 24, 25, 26, 27, 28, 29],
>       [30, 31, 32, 33, 34, 35, 36, 37, 38, 39],
>       [40, 41, 42, 43, 44, 45, 46, 47, 48, 49],
>       [60, 61, 62, 63, 64, 65, 66, 67, 68, 69],
>       [70, 71, 72, 73, 74, 75, 76, 77, 78, 79],
>       [80, 81, 82, 83, 84, 85, 86, 87, 88, 89],
>       [90, 91, 92, 93, 94, 95, 96, 97, 98, 99]], dtype=int16)
---------------
```python
np.delete(a, [5, ], axis = 0)
```
>work, 
>
>array([[0, 1, 2, 3, 4, 5, 6, 7, 8, 9],
>       [10, 11, 12, 13, 14, 15, 16, 17, 18, 19],
>       [20, 21, 22, 23, 24, 25, 26, 27, 28, 29],
>       [30, 31, 32, 33, 34, 35, 36, 37, 38, 39],
>       [40, 41, 42, 43, 44, 45, 46, 47, 48, 49],
>       [60, 61, 62, 63, 64, 65, 66, 67, 68, 69],
>       [70, 71, 72, 73, 74, 75, 76, 77, 78, 79],
>       [80, 81, 82, 83, 84, 85, 86, 87, 88, 89],
>       [90, 91, 92, 93, 94, 95, 96, 97, 98, 99]], dtype=int16)
---------------
When delete multi rows/columns, 
use `np.delete(a, (3, 5, 7), axis = 0)` or `np.delete(a, [3, 5, 7], axis = 0)` both work, 
```python
np.delete(a, (3, 5, 7), axis = 0)
```
>work, 
>
>array([[0, 1, 2, 3, 4, 5, 6, 7, 8, 9],
>       [10, 11, 12, 13, 14, 15, 16, 17, 18, 19],
>       [20, 21, 22, 23, 24, 25, 26, 27, 28, 29],
>       [40, 41, 42, 43, 44, 45, 46, 47, 48, 49],
>       [60, 61, 62, 63, 64, 65, 66, 67, 68, 69],
>       [80, 81, 82, 83, 84, 85, 86, 87, 88, 89],
>       [90, 91, 92, 93, 94, 95, 96, 97, 98, 99]], dtype=int16)
---------------
```python
np.delete(a, [3, 5, 7], axis = 0)
```
>work, 
>
>array([[0, 1, 2, 3, 4, 5, 6, 7, 8, 9],
>       [10, 11, 12, 13, 14, 15, 16, 17, 18, 19],
>       [20, 21, 22, 23, 24, 25, 26, 27, 28, 29],
>       [40, 41, 42, 43, 44, 45, 46, 47, 48, 49],
>       [60, 61, 62, 63, 64, 65, 66, 67, 68, 69],
>       [80, 81, 82, 83, 84, 85, 86, 87, 88, 89],
>       [90, 91, 92, 93, 94, 95, 96, 97, 98, 99]], dtype=int16)
-------------
Is it match design purpose?

v923z, 2022-01-11:
> When just delete 1 row/column, must use `np.delete(a, [row/column], axis = 0)` or `np.delete(a, (row/column, ), axis = 0)` or `np.delete(a, [5, ], axis = 0)`,
> 
> `np.delete(a, row/column, axis = 0)` or `np.delete(a, (row/column), axis = 0)` not work.

Thanks for looking into this. This is indeed an error, I will fix it as soon as possible.

v923z, 2022-01-11:
@water5 Could you, please, check out the latest commit in https://github.com/v923z/micropython-ulab/pull/473?

water5, 2022-01-12:
Thanks, test result correct currently.

By the way, Question about `git` use, 

```shell
git clone https://github.com/v923z/micropython-ulab.git
cd micropython-ulab
git checkout io
```

>Branch 'io' set up to track remote branch 'io' from 'origin'.
>Switched to a new branch 'io'

```shell
git pull
```
>Already up to date.

Why can't refresh files and appear 'Already up to date.'? Which step I miss?
I use download / extract zip file at this time.

v923z, 2022-01-12:
> Thanks, all test result correct currently.

Many thanks for the feedback!

> By the way, Question about `git` use,
>
> ```shell
> git clone https://github.com/v923z/micropython-ulab.git
> cd micropython-ulab
> git checkout io
> ```
>
>
> ```shell
> git pull
> ```
>
>
> Why can't refresh files and [...]

You should usually fetch the content before pulling

```
git fetch
git pull
```

closed 2022-01-12 by v923z

water5, 2022-01-15:
Build error with STM32, but build pass with ESP32.
`$ make BOARD=ADAFRUIT_F405_EXPRESS USER_C_MODULES=../../../micropython-ulab all`

> CC ../../../micropython-ulab/code/numpy/stats.c
> CC ../../../micropython-ulab/code/numpy/transform.c
> ../../../micropython-ulab/code/numpy/transform.c: In function 'transform_delete':
> ../../../micropython-ulab/code/numpy/transform.c:209:9: error: unknown type name 'ssize_t'; did you mean '_ssize_t'?
  209 |         ssize_t value = (ssize_t)mp_obj_get_int(indices);
      |         ^~~~~~~
      |         _ssize_t
> ../../../micropython-ulab/code/numpy/transform.c:209:26: error: 'ssize_t' undeclared (first use in this function); did you mean '_ssize_t'?
  209 |         ssize_t value = (ssize_t)mp_obj_get_int(indices);
      |                          ^~~~~~~
      |                          _ssize_t
> ../../../micropython-ulab/code/numpy/transform.c:209:26: note: each undeclared identifier is reported only once for each function it appears in
> ../../../micropython-ulab/code/numpy/transform.c:209:34: error: expected ',' or ';' before 'mp_obj_get_int'
  209 |         ssize_t value = (ssize_t)mp_obj_get_int(indices);
      |                                  ^~~~~~~~~~~~~~
> ../../../micropython-ulab/code/numpy/transform.c:213:45: error: expected ')' before 'axis_len'
  213 |         if((value < 0) || (value > (ssize_t)axis_len)) {
      |                           ~                 ^~~~~~~~
      |                                             )
> ../../../micropython-ulab/code/numpy/transform.c:222:20: error: expected ';' before 'value'
  222 |             ssize_t value = (ssize_t)mp_obj_get_int(item);
      |                    ^~~~~~
      |                    ;
> ../../../micropython-ulab/code/numpy/transform.c:223:16: error: 'value' undeclared (first use in this function)
  223 |             if(value < 0) {
      |                ^~~~~
> ../../../micropython-ulab/code/numpy/transform.c:226:49: error: expected ')' before 'axis_len'
  226 |             if((value < 0) || (value > (ssize_t)axis_len)) {
      |                               ~                 ^~~~~~~~
      |                                                 )
> make: *** [../../py/mkrules.mk:77: build-ADAFRUIT_F405_EXPRESS/code/numpy/transform.o] Error 1

v923z, 2022-01-16:
@water5 I think a header file was missing. Could you, please, check out https://github.com/v923z/micropython-ulab/tree/delete-fix to see if that fixes the issue?

v923z, 2022-01-16:
@water5 I see that you have closed https://github.com/v923z/micropython-ulab/issues/484. Does this mean that the issue has been resolved?

water5, 2022-01-16:
OK, thanks, `STM32`, `RP2` build success.
But `ATSAMD` build error below:
`make BOARD=ADAFRUIT_ITSYBITSY_M4_EXPRESS USER_C_MODULES=../../../micropython-ulab all`

> CC ../../../micropython-ulab/code/utils/utils.c
> CC ../../../micropython-ulab/code/ulab.c
> CC ../../lib/littlefs/lfs1.c
> CC ../../lib/littlefs/lfs1_util.c
> AS ../../shared/runtime/gchelper_m3.s
> LINK build-ADAFRUIT_ITSYBITSY_M4_EXPRESS/firmware.elf
> arm-none-eabi-ld: build-ADAFRUIT_ITSYBITSY_M4_EXPRESS/code/ulab_tools.o: in function `ulab_tools_mp_obj_is_scalar':
> ulab_tools.c:(.text.ulab_tools_mp_obj_is_scalar+0x30): undefined reference to `mp_type_complex'
> arm-none-eabi-ld: build-ADAFRUIT_ITSYBITSY_M4_EXPRESS/code/ndarray.o: in function `ndarray_set_complex_value':
> ndarray.c:(.text.ndarray_set_complex_value+0x20): undefined reference to `mp_obj_get_complex'
> arm-none-eabi-ld: ndarray.c:(.text.ndarray_set_complex_value+0x3c): undefined reference to `mp_type_complex'
> arm-none-eabi-ld: build-ADAFRUIT_ITSYBITSY_M4_EXPRESS/code/ndarray.o: in function `ndarray_get_item':
> ndarray.c:(.text.ndarray_get_item+0x12): undefined reference to `mp_obj_new_complex'
> arm-none-eabi-ld: build-ADAFRUIT_ITSYBITSY_M4_EXPRESS/code/ndarray.o: in function `ndarray_assign_elements':
> ndarray.c:(.text.ndarray_assign_elements+0x42): undefined reference to `mp_obj_get_complex'
> arm-none-eabi-ld: build-ADAFRUIT_ITSYBITSY_M4_EXPRESS/code/ndarray.o: in function `ndarray_from_mp_obj':
> ndarray.c:(.text.ndarray_from_mp_obj+0xc6): undefined reference to `mp_obj_get_complex'
> arm-none-eabi-ld: ndarray.c:(.text.ndarray_from_mp_obj+0xe4): undefined reference to `mp_type_complex'
> arm-none-eabi-ld: build-ADAFRUIT_ITSYBITSY_M4_EXPRESS/code/numpy/carray/carray.o: in function `carray_conjugate':
> carray.c:(.text.carray_conjugate+0x5e): undefined reference to `mp_obj_get_complex'
> arm-none-eabi-ld: carray.c:(.text.carray_conjugate+0x72): undefined reference to `mp_obj_new_complex'
> arm-none-eabi-ld: carray.c:(.text.carray_conjugate+0x9c): undefined reference to `mp_type_complex'
> arm-none-eabi-ld: build-ADAFRUIT_ITSYBITSY_M4_EXPRESS/code/numpy/create.o: in function `create_linspace':
> create.c:(.text.create_linspace+0x50): undefined reference to `mp_obj_get_complex'
> arm-none-eabi-ld: create.c:(.text.create_linspace+0x5a): undefined reference to `mp_obj_get_complex'
> arm-none-eabi-ld: create.c:(.text.create_linspace+0xe4): undefined reference to `mp_obj_new_complex'
> arm-none-eabi-ld: create.c:(.text.create_linspace+0x170): undefined reference to `mp_type_complex'
> arm-none-eabi-ld: build-ADAFRUIT_ITSYBITSY_M4_EXPRESS/code/numpy/fft/fft_tools.o: in function `fft_kernel':
> fft_tools.c:(.text.fft_kernel+0xa2): undefined reference to `sinf'
> arm-none-eabi-ld: fft_tools.c:(.text.fft_kernel+0xb2): undefined reference to `sinf'
> arm-none-eabi-ld: build-ADAFRUIT_ITSYBITSY_M4_EXPRESS/code/numpy/numerical.o: in function `numerical_sum_mean_std_ndarray':
> numerical.c:(.text.numerical_sum_mean_std_ndarray+0xd4): undefined reference to `roundf'
> arm-none-eabi-ld: build-ADAFRUIT_ITSYBITSY_M4_EXPRESS/code/numpy/vector.o: in function `vector_acos':
> vector.c:(.text.vector_acos+0x8): undefined reference to `acosf'
> arm-none-eabi-ld: build-ADAFRUIT_ITSYBITSY_M4_EXPRESS/code/numpy/vector.o: in function `vector_acosh':
> vector.c:(.text.vector_acosh+0x8): undefined reference to `acoshf'
> arm-none-eabi-ld: build-ADAFRUIT_ITSYBITSY_M4_EXPRESS/code/numpy/vector.o: in function `vector_asin':
> vector.c:(.text.vector_asin+0x8): undefined reference to `asinf'
> arm-none-eabi-ld: build-ADAFRUIT_ITSYBITSY_M4_EXPRESS/code/numpy/vector.o: in function `vector_asinh':
> vector.c:(.text.vector_asinh+0x8): undefined reference to `asinhf'
> arm-none-eabi-ld: build-ADAFRUIT_ITSYBITSY_M4_EXPRESS/code/numpy/vector.o: in function `vector_atan':
> vector.c:(.text.vector_atan+0x8): undefined reference to `atanf'
> arm-none-eabi-ld: build-ADAFRUIT_ITSYBITSY_M4_EXPRESS/code/numpy/vector.o: in function `vector_atanh':
> vector.c:(.text.vector_atanh+0x8): undefined reference to `atanhf'
> arm-none-eabi-ld: build-ADAFRUIT_ITSYBITSY_M4_EXPRESS/code/numpy/vector.o: in function `vector_cos':
> vector.c:(.text.vector_cos+0x8): undefined reference to `cosf'
> arm-none-eabi-ld: build-ADAFRUIT_ITSYBITSY_M4_EXPRESS/code/numpy/vector.o: in function `vector_erf':
> vector.c:(.text.vector_erf+0x8): undefined reference to `erff'
> arm-none-eabi-ld: build-ADAFRUIT_ITSYBITSY_M4_EXPRESS/code/numpy/vector.o: in function `vector_erfc':
> vector.c:(.text.vector_erfc+0x8): undefined reference to `erfcf'
> arm-none-eabi-ld: build-ADAFRUIT_ITSYBITSY_M4_EXPRESS/code/numpy/vector.o: in function `vector_gamma':
> vector.c:(.text.vector_gamma+0x8): undefined reference to `tgammaf'
> arm-none-eabi-ld: build-ADAFRUIT_ITSYBITSY_M4_EXPRESS/code/numpy/vector.o: in function `vector_lgamma':
> vector.c:(.text.vector_lgamma+0x8): undefined reference to `lgammaf'
> arm-none-eabi-ld: build-ADAFRUIT_ITSYBITSY_M4_EXPRESS/code/numpy/vector.o: in function `vector_log2':
> vector.c:(.text.vector_log2+0x8): undefined reference to `log2f'
> arm-none-eabi-ld: build-ADAFRUIT_ITSYBITSY_M4_EXPRESS/code/numpy/vector.o: in function `vector_sin':
> vector.c:(.text.vector_sin+0x8): undefined reference to `sinf'
> arm-none-eabi-ld: build-ADAFRUIT_ITSYBITSY_M4_EXPRESS/code/numpy/vector.o: in function `vector_tan':
> vector.c:(.text.vector_tan+0x8): undefined reference to `tanf'
[... cut ...]
> arm-none-eabi-ld: vector.c:(.text.vector_sqrt+0x94): undefined reference to `mp_obj_new_complex'
> arm-none-eabi-ld: vector.c:(.text.vector_sqrt+0x112): undefined reference to `atan2f'
> arm-none-eabi-ld: vector.c:(.text.vector_sqrt+0x122): undefined reference to `cosf'
> arm-none-eabi-ld: vector.c:(.text.vector_sqrt+0x134): undefined reference to `sinf'
> arm-none-eabi-ld: vector.c:(.text.vector_sqrt+0x1cc): undefined reference to `mp_type_complex'
> make: *** [Makefile:152: build-ADAFRUIT_ITSYBITSY_M4_EXPRESS/firmware.elf] Error 1

v923z, 2022-01-16:
@water5 I might be wrong, but I believe, the ATSAMD doesn't have complex support, so you have to set ULAB_SUPPORTS_COMPLEX to 0: https://github.com/v923z/micropython-ulab/blob/4f828f188db4fb22c9a11b5e38fcd49f2ca41bf0/code/ulab.h#L36

water5, 2022-01-16:
[I already modified:
`#define ULAB_SUPPORTS_COMPLEX               (0)`
still rise error below:

> CC ../../../micropython-ulab/code/utils/utils.c
> CC ../../../micropython-ulab/code/ulab.c
> CC ../../lib/littlefs/lfs1.c
> CC ../../lib/littlefs/lfs1_util.c
> AS ../../shared/runtime/gchelper_m3.s
> LINK build-ADAFRUIT_ITSYBITSY_M4_EXPRESS/firmware.elf
> arm-none-eabi-ld: build-ADAFRUIT_ITSYBITSY_M4_EXPRESS/code/numpy/fft/fft_tools.o: in function `fft_kernel':
> fft_tools.c:(.text.fft_kernel+0xa2): undefined reference to `sinf'
> arm-none-eabi-ld: fft_tools.c:(.text.fft_kernel+0xb2): undefined reference to `sinf'
> arm-none-eabi-ld: build-ADAFRUIT_ITSYBITSY_M4_EXPRESS/code/numpy/numerical.o: in function `numerical_sum_mean_std_ndarray':
> numerical.c:(.text.numerical_sum_mean_std_ndarray+0xc8): undefined reference to `roundf'
> arm-none-eabi-ld: build-ADAFRUIT_ITSYBITSY_M4_EXPRESS/code/numpy/vector.o: in function `vector_acos':
> vector.c:(.text.vector_acos+0x8): undefined reference to `acosf'
> arm-none-eabi-ld: build-ADAFRUIT_ITSYBITSY_M4_EXPRESS/code/numpy/vector.o: in function `vector_acosh':
> vector.c:(.text.vector_acosh+0x8): undefined reference to `acoshf'
> arm-none-eabi-ld: build-ADAFRUIT_ITSYBITSY_M4_EXPRESS/code/numpy/vector.o: in function `vector_asin':
> vector.c:(.text.vector_asin+0x8): undefined reference to `asinf'
> arm-none-eabi-ld: build-ADAFRUIT_ITSYBITSY_M4_EXPRESS/code/numpy/vector.o: in function `vector_asinh':
> vector.c:(.text.vector_asinh+0x8): undefined reference to `asinhf'
> arm-none-eabi-ld: build-ADAFRUIT_ITSYBITSY_M4_EXPRESS/code/numpy/vector.o: in function `vector_atan':
> vector.c:(.text.vector_atan+0x8): undefined reference to `atanf'
> arm-none-eabi-ld: build-ADAFRUIT_ITSYBITSY_M4_EXPRESS/code/numpy/vector.o: in function `vector_atanh':
> vector.c:(.text.vector_atanh+0x8): undefined reference to `atanhf'
> arm-none-eabi-ld: build-ADAFRUIT_ITSYBITSY_M4_EXPRESS/code/numpy/vector.o: in function `vector_cos':
> vector.c:(.text.vector_cos+0x8): undefined reference to `cosf'
> arm-none-eabi-ld: build-ADAFRUIT_ITSYBITSY_M4_EXPRESS/code/numpy/vector.o: in function `vector_erf':
> vector.c:(.text.vector_erf+0x8): undefined reference to `erff'
> arm-none-eabi-ld: build-ADAFRUIT_ITSYBITSY_M4_EXPRESS/code/numpy/vector.o: in function `vector_erfc':
> vector.c:(.text.vector_erfc+0x8): undefined reference to `erfcf'
> arm-none-eabi-ld: build-ADAFRUIT_ITSYBITSY_M4_EXPRESS/code/numpy/vector.o: in function `vector_gamma':
> vector.c:(.text.vector_gamma+0x8): undefined reference to `tgammaf'
> arm-none-eabi-ld: build-ADAFRUIT_ITSYBITSY_M4_EXPRESS/code/numpy/vector.o: in function `vector_lgamma':
> vector.c:(.text.vector_lgamma+0x8): undefined reference to `lgammaf'
> arm-none-eabi-ld: build-ADAFRUIT_ITSYBITSY_M4_EXPRESS/code/numpy/vector.o: in function `vector_log2':
> vector.c:(.text.vector_log2+0x8): undefined reference to `log2f'
> arm-none-eabi-ld: build-ADAFRUIT_ITSYBITSY_M4_EXPRESS/code/numpy/vector.o: in function `vector_sin':
> vector.c:(.text.vector_sin+0x8): undefined reference to `sinf'
> arm-none-eabi-ld: build-ADAFRUIT_ITSYBITSY_M4_EXPRESS/code/numpy/vector.o: in function `vector_tan':
> vector.c:(.text.vector_tan+0x8): undefined reference to `tanf'
> arm-none-eabi-ld: build-ADAFRUIT_ITSYBITSY_M4_EXPRESS/code/numpy/vector.o: in function `vector_around':
> vector.c:(.text.vector_around+0x70): undefined reference to `roundf'
> arm-none-eabi-ld: build-ADAFRUIT_ITSYBITSY_M4_EXPRESS/code/numpy/vector.o: in function `vector_arctan2':
> vector.c:(.text.vector_arctan2+0x98): undefined reference to `atan2f'
> make: *** [Makefile:152: build-ADAFRUIT_ITSYBITSY_M4_EXPRESS/firmware.elf] Error 1
](url)

v923z, 2022-01-16:
It seems to me that all math functions are disabled. I think you enable them here: https://github.com/micropython/micropython/blob/3dc9a42bc2a048153be742f96bd3d7f0a030ea84/ports/samd/boards/ADAFRUIT_ITSYBITSY_M4_EXPRESS/mpconfigboard.h

v923z, 2022-01-16:
But this is obviously not a `ulab` issue: you already have problems with `modmath`, which comes from `micropython`. You should disable `ulab` altogether, and just try to compile `micropython`. Your math library is disabled somewhere. Once you figured that out, you can switch on `ulab`, and then it should work. Also, you should `make clean` before compiling again.

water5, 2022-01-16:
OK, tested, 
`/micropython/ports/samd/boards/ADAFRUIT_ITSYBITSY_M4_EXPRESS/mpconfigboard.h`
only 
```C
#define MICROPY_PY_CMATH                			(1)
```

could enable, 

-------------------

```C
#define MICROPY_PY_MATH                 			(0)
```
can't enable, so, is it possible use `ulab` on `ATSAMD` port?

v923z, 2022-01-16:
Somewhere on `circuitpython` I have seen that on the SAMD boards `ulab` is not enabled, so it could be that they are not capable. Those are really small MCUs: https://github.com/micropython/micropython/tree/master/ports/samd.

water5, 2022-01-16:
OK, `ATSAMD` port on `MicroPython` is very new, maybe improving.
`ItsyBitsy M4 Express (ATSAMD51G19A)` already enabled `ulab` on `CircuitPython`.
[https://circuitpython.org/board/itsybitsy_m4_express/](https://circuitpython.org/board/itsybitsy_m4_express/)