v923z/micropython-ulab #472: [BUG] np.array(a, dtype=np.complex)

wired8, 2022-01-10:
**Describe the bug**
A clear and concise description of what the bug is. Give the `ulab` version

Master latest

**To Reproduce**
```python
 a = [4, 3, 1, (2-2j), (2+2j), (2-1j), (2+1j), (2-1j), (2+1j), (1+1j), (1-1j)]
 a = np.array(a, dtype=np.complex)
```
**Output**
array([(4+0j), 3.0+0.0j, 1.0+0.0j, ..., 1.0+1.0j, 1.0-1.0j, 2.0-1.0j], dtype=complex)

**Expected**
arrary([4.+0.j 3.+0.j 1.+0.j 2.-2.j 2.+2.j 2.-1.j 2.+1.j 2.-1.j 2.+1.j 1.+1.j 1.-1.j]

Note the ...

Looks like duplicate complex values is the cause?

v923z, 2022-01-10:
I think this is an issue with the prinout, and the the array as such. You can check this by first setting the printout threshold, which would then result in the correct output: https://micropython-ulab.readthedocs.io/en/latest/ulab-ndarray.html#customising-array-printouts
```
np.set_printoptions(threshold=100)
```
However, this is still a bug, I agree. I will try to fix it by early tomorrow.

wired8, 2022-01-10:
Yes!  That works.  Thanks @v923z

closed 2022-01-12 by v923z