v923z/micropython-ulab #476: [FEATURE REQUEST]

water5, 2022-01-14:
`numpy.lib.format.dtype_to_descr`
[https://numpy.org/doc/stable/reference/generated/numpy.lib.format.dtype_to_descr.html](https://numpy.org/doc/stable/reference/generated/numpy.lib.format.dtype_to_descr.html)

`numpy.lib.format.descr_to_dtype`
[https://numpy.org/doc/stable/reference/generated/numpy.lib.format.descr_to_dtype.html](https://numpy.org/doc/stable/reference/generated/numpy.lib.format.descr_to_dtype.html)

v923z, 2022-01-14:
@water5 I think I would handle this in `snippets`. Could you, please, check out https://github.com/v923z/micropython-ulab/tree/json, and try the script in https://github.com/v923z/micropython-ulab/tree/json/snippets. 

This works with `hexlify` at the moment. If you are OK with this, I can add the reverse function with `unhexlify`. Unfortunately, `a2b_base64` does not work as expected. I have raised an issue: https://github.com/micropython/micropython/issues/8175

water5, 2022-01-14:
> This works with hexlify at the moment. If you are OK with this, I can add the reverse function with unhexlify. Unfortunately, a2b_base64 does not work as expected. I have raised an issue: micropython/micropython#8175

You should use `ubinascii.b2a_base64`?
```python
from ubinascii import b2a_base64, a2b_base64
b2a_base64(bytearray([1, 2, 3]))
```
> b'AQID\n'

And `CPython` side, between `base64.b64encode` and `binascii.a2b_base64` have little difference.
`CPython` and `MicroPython` two side should use same library, but `MicroPython` only implement `ubinascii`.

-----------

Notice:
On `MicroPython`, `ubinascii.b2a_base64` including a newline char at the end by default, and can't choose by parameter.
[https://docs.micropython.org/en/latest/library/binascii.html#binascii.b2a_base64](https://docs.micropython.org/en/latest/library/binascii.html#binascii.b2a_base64)

On `CPython`, have `newline` parameter, `newline=True` by default.
[https://docs.python.org/3/library/binascii.html#binascii.b2a_base64](https://docs.python.org/3/library/binascii.html#binascii.b2a_base64)

v923z, 2022-01-14:
> You should use `ubinascii.b2a_base64`?

Oh, thanks! Could you, please, check out the latest commit?

v923z, 2022-01-14:
@water5 I have added the reverse function, https://github.com/v923z/micropython-ulab/blob/json/snippets/json_to_ndarray.py. If it fulfills your requirements, I would merge this branch into master.

water5, 2022-01-14:
Test OK, Thanks.

closed 2022-01-14 by v923z

water5, 2022-01-14:
```python
a = np.array([1, 2, 3], dtype = np.int16)
b = ndarray_to_json(a)
b
```
> '{"dtype": "<i2", "__numpy__": "010002000300", "shape": [3]}'
------------
json_to_ndarray() output incorrect.
```python
json_to_ndarray(b)
```
> {'dtype': '<i2', '__numpy__': '010002000300', 'shape': [3]}
> array([1, 2, 3], dtype=int16)

v923z, 2022-01-14:
What is the expected result? You have to set `b64=True`: https://github.com/v923z/micropython-ulab/blob/8154accf1b22661376f4320226c4a1958569d45f/snippets/ndarray_to_json.py#L47

water5, 2022-01-14:
json_to_ndarray() output included input content currently.

v923z, 2022-01-14:
So does the function work now as expected?

water5, 2022-01-14:
Output included the JSON contents also.

v923z, 2022-01-14:
It is really hard to say anything definite, if you don't tell what the expected behaviour is. I think this produces the required output:

```
dtypes = (np.uint8, np.int8, np.uint16, np.int16, np.float)

for dtype in dtypes:
    ndarray = np.array([1, 2, 3], dtype=dtype)
    print(ndarray_to_json(ndarray))
```

```
{"dtype": "|u1", "__numpy__": "AQID\n", "shape": [3]}
{"dtype": "|i1", "__numpy__": "AQID\n", "shape": [3]}
{"dtype": "<u2", "__numpy__": "AQACAAMA\n", "shape": [3]}
{"dtype": "<i2", "__numpy__": "AQACAAMA\n", "shape": [3]}
{"dtype": "<f8", "__numpy__": "AAAAAAAA8D8AAAAAAAAAQAAAAAAAAAhA\n", "shape": [3]}
```
Is this not correct?

v923z, 2022-01-14:
See also https://github.com/Crimson-Crow/json-numpy/blob/e8129b0ff7b2dbb124a87223275231defecb9043/json_numpy.py#L10-L14

water5, 2022-01-15:
I think that unnecessary.

[https://github.com/v923z/micropython-ulab/blob/master/snippets/json_to_ndarray.py#L62](https://github.com/v923z/micropython-ulab/blob/master/snippets/json_to_ndarray.py#L62)

v923z, 2022-01-16:
@water5 That's right. I have removed the printout.