v923z/micropython-ulab #482: [FEATURE REQUEST] Implement array element dtype support string limitted.

water5, 2022-01-14:
Maybe implement array element dtype support string limitted(except lowest dimension, `ULAB_MAX_DIMS` - 1), then convert char's to ASCII value?

v923z, 2022-01-14:
@water5 I am not quite sure what exactly this should do. Could you add more details, or add an example that explains it?

water5, 2022-01-14:
```python
a = np.array(['abc', 3, 'E3t5'])
```
> array(['abc', '3', 'E3t5'], dtype='<U11')

v923z, 2022-01-14:
I think this is pretty much impossible in the present structure, and I don't know what the use case for such a construct would be. If you really need it, you can pre-process the array in `python` before passing it to `np.array`. If you go down that path, you can add your code to the [snippets](https://github.com/v923z/micropython-ulab/tree/master/snippets).

closed 2022-01-15 by water5