v923z/micropython-ulab #489: [BUG] signal.sosfilt

wired8, 2022-01-18:
When using a correct zi value the following error is thrown:
ValueError: zi must be of shape (n_section, 2)

**To Reproduce**
```python
from ulab import numpy as np
from ulab import scipy as spy

x = np.array([0, 1, 2, 3, 4, 5, 6, 7, 8, 9])
sos = np.array([[1, 2, 3, 1, 5, 6], [1, 2, 3, 1, 5, 6], [1, 2, 3, 1, 5, 6]],dtype=np.float)
zi = np.array([[1, 2], [3, 4], [5, 6]],dtype=np.float)
y, zo = spy.signal.sosfilt(sos, x, zi=zi)
print(y,zo)

>>> ValueError: zi must be of shape (n_section, 2)
```

**Expected behavior**
```
[ 9.000000e+00 -4.700000e+01  2.240000e+02 -9.870000e+02  4.129000e+03
 -1.654900e+04  6.414900e+04 -2.419370e+05  8.921210e+05 -3.228165e+06] [[   37242.    74835.]
 [ 1026187.  1936542.]
 [10433318. 18382017.]
```

**Additional context**
Compiled with 2 dimensions.

v923z, 2022-01-18:
Thanks, I'll look into it.

v923z, 2022-01-19:
@wired8 Could you, please, check out https://github.com/v923z/micropython-ulab/pull/491?

closed 2022-01-21 by v923z