v923z/micropython-ulab #493: Cannot build latest MicroPython v1.18 w/ ulab

teuler, 2022-01-22:
Hi
I tried to build the Pico port of MicroPython (v1.18, pulled today) together with ulab (latest version, pulled today). The targeted board is the Nano RP2040 Connect. The building process failed with the following message:
```
/usr/lib/gcc/arm-none-eabi/9.2.1/../../../arm-none-eabi/bin/ld: CMakeFiles/firmware.dir/home/tom/ulab/code/numpy/io/io.c.obj: in function `io_save':
io.c:(.text.io_save+0x42): undefined reference to `mp_builtin_open'
/usr/lib/gcc/arm-none-eabi/9.2.1/../../../arm-none-eabi/bin/ld: CMakeFiles/firmware.dir/home/tom/ulab/code/numpy/io/io.c.obj: in function `io_load':
io.c:(.text.io_load+0x34): undefined reference to `mp_builtin_open'
collect2: error: ld returned 1 exit status
mcd
```
I guess this may be related to the latest optimizations in the 1.18 release of MicroPython.
A work-arround is to set `ULAB_NUMPY_HAS_LOAD` and `ULAB_NUMPY_HAS_SAVE` both to `0` - then in builds perfectly fine.

I am using Ubuntu 20.04 running in the Windows 11 Linux subsystem. The build command is:
```
make -j 8 BOARD=ARDUINO_NANO_RP2040_CONNECT USER_C_MODULES=$BUILD_DIR/../ulab/code/micropython.cmake
```
Thanks.

v923z, 2022-01-22:
> I guess this may be related to the latest optimizations in the 1.18 release of MicroPython. A work-arround is to set `ULAB_NUMPY_HAS_LOAD` and `ULAB_NUMPY_HAS_SAVE` both to `0` - then in builds perfectly fine.

I have no experience with the RP2040, but my first question would be, whether you can do something like this:

```
with open('somefile.txt', 'r') as fin:
    print(fin.read())
```
As far as I understand, the `mp_builtin_open` that the compiler is complaining about is the function handling the context manager in the snippet above.

According to https://github.com/micropython/micropython/search?q=mp_builtin_open, `mp_builtin_open` seems to be a public function on all ports. At least, I don't see a `static` qualifier anywhere. But it might still be that the RP2040 does not use it at all.

teuler, 2022-01-22:
The following code works:
```
>>> f = open('data.txt', 'w')
>>> f.write("Hello")
5
>>> f.close()
>>> f = open('data.txt', 'r')
>>> print(f.read())
Hello
>>> 
```

teuler, 2022-01-22:
PS: I successfully build MPy 1.17 w/ ulab a few days ago.

v923z, 2022-01-22:
@teuler I can build the `unix` port with `v.1.18`, so I believe, the issue is really with the `rp2` port. I'll try to figure out why this doesn't work.

teuler, 2022-01-22:
Thanks.
In the mean time, I built the `ESP32` port (`UM_TINYPICO`) with MPy v1.18 and it works as well with all default constants in `ulab.h`. I am trying now to build the vanilla `rp2` port ...

v923z, 2022-01-22:
> I also tried to build the vanilla `rp2` port with the same errors. So I guess it is indeed an `rp2` port issue.

Do you need those two functions urgently? I would really like to get to the bottom of this. We could try to roll out a hackish patch, but I would prefer to do this properly, and we might need input from upstream, so I can't commit to a definite timescale.

teuler, 2022-01-22:
No, I don't need them. I just wanted to report that incompatibility. But thanks.

v923z, 2022-01-22:
> No, I don't need them. I just wanted to report that incompatibility.

Thanks for that, much appreciated!

v923z, 2022-01-24:
@dpgeorge I don't want to open an issue on https://github.com/micropython/micropython/issues, because I feel that this doesn't quite belong there. However, we are still facing a difficulty, and I was wondering, whether you could comment on how to solve it. On many ports, this works

```c
mp_obj_t stream = mp_builtin_open(2, open_args, (mp_map_t *)&mp_const_empty_map);
const mp_stream_p_t *stream_p = mp_get_stream(stream);

... 
stream_p->write(stream, buffer, offset, &error);
stream_p->ioctl(stream, MP_STREAM_CLOSE, 0, &error);
```
but there are some, where it cannot be linked. The relevant source is in
https://github.com/v923z/micropython-ulab/blob/6dfdc44c4e43eb588f4224b5be170c4619eeb1a1/code/numpy/io/io.c#L233-L374

Are there plans to add `mp_builtin_open` to the `rp2040` port? If the answer is no, what would be a possible workaround? If there isn't any, is it possible to handle this at the pre-processing stage, and simply drop the `io_save` function, if `mp_builtin_open` is not available on a particular port?

Writing/reading numerical data to/from files has been a recurring request, and it would be great, if we didn't have to exclude the functionality.

dpgeorge, 2022-01-25:
Thanks for raising this issue.  It is a minor incompatibility with the rp2 port.  Now that I look at it, probably `mp_builtin_open` can be completely removed from all ports because it's not really needed, and not guaranteed to exist.  But I'd need to think more about that.

In the meantime I think the best fix is to change `mp_builtin_open(...)` to:
```c
mp_builtin_open_obj.fun.kw(...)
```
An alternative would be:
```c
#if MICROPY_VFS
mp_vfs_open(...)
#else
mp_builtin_open(...)
#endif
```

v923z, 2022-01-25:
@dpgeorge Many thanks for the feedback! I will try to go with your first suggestion. I'll ping you once more, if we still have difficulties.

v923z, 2022-01-25:
@teuler Thomas, could you, please, check out https://github.com/v923z/micropython-ulab/pull/496? I could locally compile it, but it would be great to have an extra set of eyes before merging. Thanks!

v923z, 2022-01-25:
@dpgeorge I couldn't pull this off with your first suggestion. This works, however https://github.com/v923z/micropython-ulab/blob/77a856e84ebc09e07ba245334d5c6cc0b318535f/code/numpy/io/io.h#L14-L18 Again, thanks for chipping in!

dpgeorge, 2022-01-25:
Why didn't `mp_builtin_open_obj.fun.kw()` work?

v923z, 2022-01-26:
Most probably I forgot to run `make clean` before compiling. It's OK now. I have updated the code. Sorry for the noise.

vijaysekark, 2022-01-27:
Hi, I was able to build successfully by replacing mp_builtin_open(...) with  mp_builtin_open_obj.fun.kw(...).

Please update the repo to close this issue. 

[image]

v923z, 2022-01-27:
Thanks for the feedback!

closed 2022-01-27 by v923z

teuler, 2022-01-27:
@v923z Sorry for the delay. I can confirm that also the firmware for the `ARDUINO_NANO_RP2040_CONNECT` is build correctly.

v923z, 2022-01-27:
@teuler No problem, and thanks for bringing up the issue!