v923z/micropython-ulab #494: [FEATURE REQUEST] numpy.unique

water5, 2022-01-23:
[https://numpy.org/doc/stable/reference/generated/numpy.unique.html](https://numpy.org/doc/stable/reference/generated/numpy.unique.html)

v923z, 2022-01-23:
While it is not the most performant way, I think you can already get this functionality by
```
a = np.array()
set(a.tolist())
```

water5, 2022-01-24:
But `set` not support `ndarray` to `list` result.
```python
a = np.arange(9).reshape((3, 3))
a
```
>array([[0, 1, 2],
>       [3, 4, 5],
>       [6, 7, 8]], dtype=int16)
```python
a = a.tolist()
a
```
>[[0, 1, 2], [3, 4, 5], [6, 7, 8]]
```python
set(a)
```
>Traceback (most recent call last):
>  File "\<stdin\>", line 1, in \<module\>
>TypeError: unsupported type for \_\_hash\_\_: 'list'

Only support 1D: 
```python
a = np.arange(9)
a
```
>array([0, 1, 2, 3, 4, 5, 6, 7, 8], dtype=int16)
```python
set(a)
```
>{0, 1, 2, 3, 4, 5, 6, 7, 8}

water5, 2022-01-24:
Although we can do `set` operation after `numpy.reshape`: 
```python
a = a.reshape((9, ))
set(a)
a = a.reshape((3, 3))
```
But the `return_index` parameter result will pointless.

v923z, 2022-01-24:
I am trying to understand in what situation this function would be required.