v923z/micropython-ulab #497: build problem on Mac OSX

ryanGT, 2022-01-28:
I am having a build problem that I am pretty sure is my fault, but I don't know how to fix it.  I have used numpy for many years, but my C skills are pretty weak and I have never cross-compiled anything.  I am trying to build micropython with ulab using my Mac.  I think I have the necessary tools from homebrew using the command 

brew install --cask gcc-arm-embedded

I executed this command from the STM32 ports folder after cloning micropython and ulab from git:

make BOARD=PYBV11 USER_C_MODULES=../../../ulab all

I believe everything compiles ok, but I get this message when linking:

CC build-PYBV11/pins_PYBV11.c
LINK build-PYBV11/firmware.elf
arm-none-eabi-ld: build-PYBV11/code/numpy/io/io.o: in function `io_save':
io.c:(.text.io_save+0xd0): undefined reference to `sprintf'
arm-none-eabi-ld: io.c:(.text.io_save+0x124): undefined reference to `sprintf'
arm-none-eabi-ld: io.c:(.text.io_save+0x1dc): undefined reference to `sprintf'
make: *** [build-PYBV11/firmware.elf] Error 1

Any help would be appreciated.

Thanks,

Ryan

ryanGT, 2022-01-28:
So, I believe that sprintf should be defined in stdio, which is included in io.c:

#include <string.h>
#include <stdio.h>

Does that mean I have a bad version of stdio.h installed somewhere or is something not linking correctly to a stdio library or something?

ryanGT, 2022-01-28:
FYI, I followed the same build process on a Raspberry Pi 4 and got the same error regarding sprintf.

v923z, 2022-01-28:
@ryanGT I have noticed recently that some compilers have difficulties with `sprintf`, so https://github.com/v923z/micropython-ulab/pull/492/commits/d56dd2f9212377139fe632cdf396e1f3587086ef implements our own `sprintf` for these cases. Could you, please, check out the [io-savetxt ](https://github.com/v923z/micropython-ulab/tree/io-savetxt) branch, and see if you can compile that?

Also, you can find our CI here: https://github.com/v923z/micropython-ulab/blob/master/.github/workflows/build.yml That should tell you what is required on Macs. The branch that I referred to passes with this script.

ryanGT, 2022-01-28:
Thanks for the help.  Unfortunately, I am getting the same error message for the io-savetxt branch, which really doesn't make sense.

I am a python programmer with weak C skills.  I don't know how to undo a failed make.  So, I deleted both micropython and ulab repos on my Mac and recloned them.  I then went into the newly cloned ulab repo and did 

git checkout -b io-savetxt

git branch
* io-savetxt
  master

I then went to the ports/stm folder in micropython and executed 

make BOARD=PYBV11 USER_C_MODULES=../../../ulab all

The first time through, it complains about a lack of stm32lib:

stm32lib submodule not found, fetching it now...
(cd ../.. && git submodule update --init lib/stm32lib)
Submodule 'lib/stm32lib' (https://github.com/micropython/stm32lib) registered for path 'lib/stm32lib'
Cloning into '/Users/kraussry/git/other_peoples_repos/micropython/lib/stm32lib'...
Submodule path 'lib/stm32lib': checked out 'c1b7a415778247dd3881a1f3d11ea02190cbc94d'

and the build fails with hal related issues:

CC ../../lib/littlefs/lfs2.c
CC ../../lib/littlefs/lfs2_util.c
CC build-PYBV11/pins_PYBV11.c
LINK build-PYBV11/firmware.elf
arm-none-eabi-ld: cannot find build-PYBV11/lib/stm32lib/STM32F4xx_HAL_Driver/Src/stm32f4xx_hal.o: No such file or directory
arm-none-eabi-ld: cannot find build-PYBV11/lib/stm32lib/STM32F4xx_HAL_Driver/Src/stm32f4xx_hal_adc.o: No such file or directory
arm-none-eabi-ld: cannot find build-PYBV11/lib/stm32lib/STM32F4xx_HAL_Driver/Src/stm32f4xx_hal_adc_ex.o: No such file or directory
....

and many related errors.

Rerunning the make command causes many hal related things to compile:

CC ../../lib/stm32lib/STM32F4xx_HAL_Driver/Src/stm32f4xx_hal.c
CC ../../lib/stm32lib/STM32F4xx_HAL_Driver/Src/stm32f4xx_hal_adc.c
CC ../../lib/stm32lib/STM32F4xx_HAL_Driver/Src/stm32f4xx_hal_adc_ex.c
CC ../../lib/stm32lib/STM32F4xx_HAL_Driver/Src/stm32f4xx_hal_cortex.c
CC ../../lib/stm32lib/STM32F4xx_HAL_Driver/Src/stm32f4xx_hal_dma.c
....

but the build fails with the same error messages:

CC ../../lib/stm32lib/STM32F4xx_HAL_Driver/Src/stm32f4xx_hal_i2s_ex.c
AS ../../lib/stm32lib/CMSIS/STM32F4xx/Source/Templates/gcc/startup_stm32f405xx.s
CC ../../lib/stm32lib/CMSIS/STM32F4xx/Source/Templates/system_stm32f4xx.c
LINK build-PYBV11/firmware.elf
arm-none-eabi-ld: build-PYBV11/code/numpy/io/io.o: in function `io_save':
io.c:(.text.io_save+0xd0): undefined reference to `sprintf'
arm-none-eabi-ld: io.c:(.text.io_save+0x124): undefined reference to `sprintf'
arm-none-eabi-ld: io.c:(.text.io_save+0x1dc): undefined reference to `sprintf'
make: *** [build-PYBV11/firmware.elf] Error 1

I use git mostly to share my code with my students.  My skills are a bit weak there are well.  I have never used github's CI.  Your message made it sound like I could run something to test that I have the necessary tools.  Can I run build.yml from the command line?  Or do I need to fork the repo and do something on the github web interface?  I tried reading about CI on github's docs but didn't see how to force a CI command from the command line.

Sorry for my newbie C issues.

Thanks,
Ryan

v923z, 2022-01-28:
>  I don't know how to undo a failed make

Either you do what you did below, or you run `make clean`. This alone might not always clean up the mess, because `make` does not keep track of what you tried to compile for, so it might try to attempt the wrong artifacts folder. What you should do in those cases is get rid of the folder that begins with `build_some_board`. This directory you can find in the port's folder, where you are currently compiling. 

> So, I deleted both micropython and ulab repos on my Mac and recloned them. I then went into the newly cloned ulab repo and did
> 
> git checkout -b io-savetxt
> 
> git branch
> 
>     * io-savetxt
>       master

What happens, if you try 
```
git checkout io-savetxt
git fetch
git pull
```

`git checkout -b branch_name` creates a new branch, from wherever you are at that point, most probably `master` in this case.

> Rerunning the make command causes many hal related things to compile:

Right. The reason for this is that micropython uses a lot of sub-modules, which first have to be fetched, and then compiled. But that happens now automatically, when you re-run the make. 

> CC ../../lib/stm32lib/STM32F4xx_HAL_Driver/Src/stm32f4xx_hal.c CC ../../lib/stm32lib/STM32F4xx_HAL_Driver/Src/stm32f4xx_hal_adc.c CC [...]

No, but that file simply lists the commands that you need to run for creating the output. Unfortunately, the CI is for the `unix` port, since that is the only one that we can actually check. All other ports would require an piece of hardware.

> Or do I need to fork the repo and do something on the github web interface? 

That you could do, but that is not going to help you too much, because you could only check, whether the `unix` port can be built. By the way, if you clone the repository, you can trigger the CI by simply editing and then saving the README.md, or any other file.

> 
> Sorry for my newbie C issues.

Absolutely no problem.  I hope this helps. Let us know if you are still having difficulties. 

PS: I would really like to set up a web site, from where pre-compiled firmware could be downloaded for most hardware platforms. I need a couple of weeks to sort out the details. And before you become desperate, if you simply need the firmware for the pyboard, I could send that by e-mail.

ryanGT, 2022-01-28:
Success.  Thanks for your patience.  I see my mistake (PEBKAC).  I have used git for several years, but almost exclusively on projects where I am the only developer, so I rarely branch.  This command:

git checkout -b io-savetxt

creates a new branch with the right name, but no new content (it is the same as the main/master I just cloned).

The correct git command is

git checkout io-savetxt

which checks out the branch from the server.  Fetching and pulling gets my new checked out branch ready.

Here is the output from my newly flashed pyboard:

MicroPython v1.18-56-g517e82eb6 on 2022-01-28; PYBv1.1 with STM32F405RG
Type "help()" for more information.
>>> import ulab
>>> ulab.__version__
'4.4.0-2D-c'
>>> from ulab import numpy as np
>>> mat = np.zeros((3,3))
>>> mat
array([[0.0, 0.0, 0.0],
       [0.0, 0.0, 0.0],
       [0.0, 0.0, 0.0]], dtype=float32)

Thanks again,
Ryan

v923z, 2022-01-28:
@ryanGT Ryan,

> Success. Thanks for your patience. 

Thanks for the feedback, I am glad it worked out for you. If you are satisfied, please, close the issue!

ryanGT, 2022-01-28:
My problem is solved.  I am closing the issue.

closed 2022-01-28 by ryanGT