v923z/micropython-ulab #498: Running ulab examples in the web browser

urish, 2022-01-29:
First of all, thanks for bring numpy and scipy to embedded Python. It's so cool to have all these math abilities now available in micropython!

I'm working on an online simulator for the ESP32, and one of the users asked to add ulab support. So I went ahead and compiled the latest ulab (4.3.2) with the last micropython (1.18.0), and made it available for the users of the simulator:

https://wokwi.com/arduino/projects/322114140704342610

The idea is that anyone can open the link, modify the code, press the green "Play" button, and see the result immediately. Or they can just run the code and poke around in the MicroPython REPL. Then they can save the modified code, and share it with others. So it's a great way to get familiar with ulab without having to flash a custom firmware to your ESP32.

How can the online simulation be useful for the ulab community?

v923z, 2022-01-29:
@urish Thanks for sharing your work! If you intend to keep your project alive for longer periods of time, I would definitely like to add a link in the [README](https://github.com/v923z/micropython-ulab/blob/master/README.md). I can also make this issue sticky, so that it would always be at the top of the issue page. 

> How can the online simulation be useful for the ulab community?

I cannot really comment on this question. To be honest, I have only very vague ideas as to how people use this code in their own projects. But if I were you, I would mention it somewhere on the [micropython forum](https://forum.micropython.org/). You would get significantly more visibility there.

urish, 2022-01-29:
Thank you! A link from the README would be cool. 

Also, good idea with the MicroPython forum. I shared the simulator there in the past, but perhaps the ulab support calls for another post there.

p.s. I created a Dockerfile that builds the ESP32 firmware (latest micropython + ulab). I think it shouldn't be too hard to turn it into a GitHub action that will create nightly builds. If you are interested, I can look into this.

v923z, 2022-01-29:
> Thank you! A link from the README would be cool.

OK, will do.

> p.s. I created a Dockerfile that builds the ESP32 firmware (latest micropython + ulab). I think it shouldn't be too hard to turn it into a GitHub action that will create nightly builds. If you are interested, I can look into this.

I have been experimenting with that [here](https://github.com/v923z/micropython-builder), and I really want to set up something that regularly compiles the firmware for many platforms. Not nightly builds, but perhaps every three days or so. That in itself is simple. The question is, what you do with the files afterwards. While the files can in principle be hosted on github, you cannot store them there for high-volume downloading, they have to be hosted on an external service. That is what `micropython`, and `circuitpython` also do.

v923z, 2022-01-29:
@urish Uri, I inserted a short paragraph [here](https://github.com/v923z/micropython-ulab/blob/master/README.md#esp32-based-boards). If you are not happy with it for whatever reason, you can always open a PR, and we can take it from here.

urish, 2022-01-29:
Looks good, thanks!

> While the files can in principle be hosted on github, you cannot store them there for high-volume downloading, they have to be hosted on an external service

It'd be very useful to have automated builds.

I've actually seen quite few popular projects using the Releases feature of GitHub to manage their downloads. Here are some examples:

- [cloudflared](https://github.com/cloudflare/cloudflared/releases) by CloudFlare
- [CircuitPython Library Bundle](https://github.com/adafruit/Adafruit_CircuitPython_Bundle/releases/latest)

There's a way to run an action whenever you push a new tag, so it automatically attaches the binaries to the right tag. I set this up for [wokwigw](https://github.com/wokwi/wokwigw/blob/main/.github/workflows/release.yml) a few days ago.

v923z, 2022-01-29:
> > While the files can in principle be hosted on github, you cannot store them there for high-volume downloading, they have to be hosted on an external service
> 
> It'd be very useful to have automated builds.

I am working on it. I want to do everything via bash scripts, so that one could also download the script, and run it on a PC. yaml files are not good for that. But there are a couple of things to consider. E.g., cloning `micropython` takes about 3 minutes, and it should be done only once, not for each platform.

> 
>     * [cloudflared](https://github.com/cloudflare/cloudflared/releases) by CloudFlare
> 
>     * [CircuitPython Library Bundle](https://github.com/adafruit/Adafruit_CircuitPython_Bundle/releases/latest)

These are organisations, who pay for services. I am not sure I can just publish whatever artifacts for downloads. This repository is private. Your repository also belongs to an organisation, so you probably know what that entails.

> There's a way to run an action whenever you push a new tag, so it automatically attaches the binaries to the right tag. I set this up for [wokwigw](https://github.com/wokwi/wokwigw/blob/main/.github/workflows/release.yml) a few days ago.

I don't want to tie this to releases for at least two reasons. First, we have to clone two entirely separate repositories, whose release schedules might be completely out of sync. Second, neither `micropython`, nor `ulab` has frequent releases (6 months/1 month on average), but that doesn't mean that we don't merge bug fixes, which should become part of the binary as soon as possible. As I said, I would bind the compilation to a chron job, and run every couple of days.

urish, 2022-01-29:
> I don't want to tie this to releases for at least two reasons.

Yes, that makes a lot of sense.

> I am not sure I can just publish whatever artifacts for downloads

As far as I can tell, this is the case. For instance, the Wokwi organization uses the free tier. GitHub [says](https://docs.github.com/en/repositories/releasing-projects-on-github/about-releases): "There is no limit on the total size of a release, nor bandwidth usage."

v923z, 2022-01-29:
> As far as I can tell, this is the case. For instance, the Wokwi organization uses the free tier. GitHub [says](https://docs.github.com/en/repositories/releasing-projects-on-github/about-releases): [...]

Oh, I was just looking for that, thanks! Now, when we are at it, would you have it as part of `ulab`, or in a complete separate repository? Originally I thought that I would have to buy external web services, and in that case it would have made sense to set up a separate repository, but in light of what you have said, I am not so sure anymore. I am still leaning towards a separate repository, though.

urish, 2022-01-29:
If you keep it in a separate repo, then make sure to make it obvious for people to know where they can find the releases. When I first looked at ulab, I immediately went to the "releases" section to see if there's a prebuilt firmware I can downloaded. Only then I looked at the README and scanned for "download" and "release", and a few minutes later I discovered the docs.

This is where I was expecting the link to the docs to be:

[image]

But it links back to the repo...

v923z, 2022-01-29:
> If you keep it in a separate repo, then make sure to make it obvious for people to know where they can find the releases. 

OK.

> When I first looked at ulab, I immediately went to the "releases" section to see if there's a prebuilt firmware I can downloaded. Only then I looked at the README and scanned for "download" and "release", and a few minutes later I discovered the docs.

Sorry, I see that it was confusing.

> This is where I was expecting the link to the docs to be:

I've fixed it. Thanks for bringing this up!

tejalbarnwal, 2022-07-25:
[image]
Hey, though ulab allows for arrays with dimension 4, am not able to even make an array of dimension 3
Could you please look into the same?
@v923z @urish

v923z, 2022-07-25:
I think @urish takes the case `ULAB_MAX_DIMS = 2`. The error message that you see is simply wrong (it always displayed 4, no matter what `ULAB_MAX_DIMS` was), it was fixed only very recently. So, if the firmware hasn't since been updated, then it is still wrong there.

v923z, 2022-07-25:
@tejalbarnwal Actually, there was a hint in your output: `you are running ulab version 4.3.2-2D-c`, which is somewhat old, but definitely two-dimensional: https://micropython-ulab.readthedocs.io/en/latest/ulab-intro.html#the-ulab-version-string