v923z/micropython-ulab #509: Implementation in Thonny IDE

Nmsngym, 2022-02-23:
Hello! I'm new in Micropython and I am programming a Fuzzy Logic Controller in my project. As I search numpy library for my micropython using Thonny IDE, I saw that Ulab is used. But I do not know how to implement your work in my program, like how to call it or does it have an installable library in the packages? Thank you so much.

v923z, 2022-02-23:
I don't think `ulab`, or `micropython` for that matter, has anything to do with the IDE. `ulab` must be part of the firmware that you are running on the microcontroller. Once you can open the board's python console, (you would most probably connect to the board via the serial console, or perhaps, webserial) you would use `ulab` as any standard python libray: https://github.com/v923z/micropython-ulab/#usage

So, in this regard, there is nothing to implement.

v923z, 2022-03-07:
Since there is not too much activity on this issue, I am closing it.