v923z/micropython-ulab #511: Cant Find ESP32-Wroom as Target?

Gruftgrabbler, 2022-02-24:
Hello I am owning an Adafruit Feather ESP32 where I wanted to install micropython including ulab.

I managed to compile the GENERIC esp32 firmware in (esp32/boards/GENERIC) on my ubuntu computer.
But unfortunately after flashing the firmware to my esp32 board I couldn't get it running.

I looked in every board in the board folder for something that sounds like esp32 wroom or something like that but couldn't find it.
Am I stupid or is there really no available firmware?

Thank you and
Cheers
Gruftgrabbler

v923z, 2022-02-24:
The Adafruit Feather is among the boards, I believe:

```bash
boards$ ls
deploy_c3.md     GENERIC_OTA        manifest.py             sdkconfig.spiram_sx
deploy.md        GENERIC_S2         manifest_release.py     sdkconfig.usb
deploy_s2.md     GENERIC_S3         sdkconfig.240mhz        SIL_WESP32
ESP32_S2_WROVER  GENERIC_S3_SPIRAM  sdkconfig.base          UM_FEATHERS2
GENERIC          GENERIC_SPIRAM     sdkconfig.ble           UM_FEATHERS2NEO
GENERIC_C3       LOLIN_S2_MINI      sdkconfig.nimble_core0  UM_TINYPICO
GENERIC_C3_USB   LOLIN_S2_PICO      sdkconfig.nimble_core1  UM_TINYS2
GENERIC_D2WD     M5STACK_ATOM       sdkconfig.spiram
```
Alternatively, you can download the `circuitpython` firmware from Adafruit's website. That already includes `ulab`.

Gruftgrabbler, 2022-02-24:
thank you for your reply.

Unfortunately I already spend months in developing into a project which uses micropython on the raspberry pi pico.
Since I want to test the code on my Adafruit feather esp32 I need to get ulab running on it.

However I tried to flash some circuit python builds but none of them worked on my specific board -.-

Back to subject:

I tried GENERIC and GERNIC_SPIRAM and looking into the board.json of each board makes me believe that the feather isn't included

v923z, 2022-02-25:
> Unfortunately I already spend months into developing a project which uses micropython on the raspberry pi pico. Since I want to test the code on my Adafruit feather esp32 I need to get ulab running on it.

What makes you think that you can't run your code with `circuitpython`? `circuitpython` is still `python`, so I don't see the problem. It simply has a number of libraries included in the firmware, but the syntax is the same.

> I tried GENERIC and GERNIC_SPIRAM and looking into the board.json of each board makes me believe that the feather isn't included

But this is not a `ulab` issue. If you can't compile the firmware, then you should raise the issue at `micropython`.

Gruftgrabbler, 2022-02-25:
> What makes you think that you can't run your code with circuitpython? circuitpython is still python, so I don't see the problem. It simply has a number of libraries included in the firmware, but the syntax is the same.

I could try it that a good point. However I couldn't find any firmware matches at the circuit python [download page](https://circuitpython.org/downloads). All of them require a uf2 boot loader which I can't install because the feather esp32 doesn't have a boot select button. 

> But this is not a ulab issue. If you can't compile the firmware, then you should raise the issue at micropython.
Thank you, I am gonna look there to see if there is anything useful :)

For now I tried almost everything what I can.

Gruftgrabbler, 2022-02-25:
Oh my gosh.

The error was of cause in front of the computer and not in the computer ...

I used the esp32-cmake.sh to compile the firmware but later on I *only* flashed the .bin file to the esp32 instead of just copy and pasting the given flash command. lol

So now I have a working esp32 with ulab on board. thank you! :) ^^

closed 2022-02-25 by Gruftgrabbler