v923z/micropython-ulab #512: Error building firmware for ESP32

fanrenkong, 2022-02-25:
I only recently started learning micropython and was lucky enough to discover the ulab module. I followed the steps exactly to build the firmware for my esp32 board, but when I in the last step, I got the following error:

/Volumes/CODE/ESP32/micropython/micropython/ports/esp32/modmachine.c:40:10: fatal error: esp32/clk.h: No such file or directory
 #include "esp32/clk.h"
          ^~~~~~~~~~~~~
compilation terminated.
/Volumes/CODE/ESP32/micropython/micropython/ports/esp32/main.c:38:10: fatal error: soc/cpu.h: No such file or directory
 #include "soc/cpu.h"
          ^~~~~~~~~~~
compilation terminated.
/Volumes/CODE/ESP32/micropython/micropython/ports/esp32/gccollect.c:37:10: fatal error: soc/cpu.h: No such file or directory
 #include "soc/cpu.h"
          ^~~~~~~~~~~
compilation terminated.

I have tried several times and don't know what is the problem, can you give me some help?

v923z, 2022-02-25:
Can you compile the firmware without `ulab`? The error that you quote above has nothing to do with `ulab`. Also, have you checked out the scripts in https://github.com/v923z/micropython-ulab/tree/master/build? Do they fail?

fanrenkong, 2022-02-25:
Thanks for you suggestion, I figured out that the problem should occur when installing esp-idf, by following the instructions carefully again, I successfully compiled the firmware.

closed 2022-02-25 by fanrenkong