v923z/micropython-ulab #513: [BUG] Numpy Module has no delete method

Gruftgrabbler, 2022-03-01:
**Describe the bug**

On my raspberry pi Pico I can't use the uLab function delete which should be implemented regarding to the [documentation](https://micropython-ulab.readthedocs.io/en/latest/numpy-functions.html?highlight=delete#delete).

Example:

```python
>>> from ulab import numpy as np
>>> a = np.array(range(25), dtype=np.uint8).reshape((5,5))

>>> print('a:\n', a)

a:
 array([[0, 1, 2, 3, 4],
       [5, 6, 7, 8, 9],
       [10, 11, 12, 13, 14],
       [15, 16, 17, 18, 19],
       [20, 21, 22, 23, 24]], dtype=uint8)
>>> print('\naxis = 0\n', np.delete(a, 2, axis=0))

Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
AttributeError: 'module' object has no attribute 'delete'
```

**To Reproduce**
Copy And paste the code into a REPL shell

**Expected behavior**
I expected the same output of the documentation

v923z, 2022-03-01:
I don't know what you mean, when saying "my raspberry pi Pico I can't use the uLab". Where does the firmware come from? From https://github.com/v923z/micropython-builder/releases, or piromoni, or somewhere else, or you compile it? What is the version of the `ulab` that you are using? I have just tried on the `unix` port, `delete` is there.

If you are compiling yourself, you set the functions in the header https://github.com/v923z/micropython-ulab/blob/master/code/ulab.h.

v923z, 2022-04-14:
This issue seems to be inactive, so closing now.

closed 2022-04-14 by v923z