v923z/micropython-ulab #515: [BUG] Numpy Indexing appears to write data incorrectly

fadesibert, 2022-03-16:
**Describe the bug**
numpy appears to write array data incorrectly when accessed by index. Particularly, when flipping data, data is mangled/duplicated.

```python
from ulab import numpy as np
print(np.__version__)
```
ulab version: '5.0.4-2D-c'
numpy has no __version__

**To Reproduce**
```
a = np.array([[1,2,3,4,5],[2,3,4,5,6],[3,4,5,6,7]], dtype=np.uint8)

>>> a[::,1::2]
array([[2, 4],
       [3, 5],
       [4, 6]], dtype=uint8)
>>> a[::-1, 1::2]
array([[4, 6],
       [3, 5],
       [2, 4]], dtype=uint8)
>>> a[::,1::2] = a[::-1, 1::2]
>>> a
array([[1, 4, 3, 6, 5],
       [2, 3, 4, 5, 6],
       [3, 4, 5, 6, 7]], dtype=uint8)
```

**Expected behavior**
Alternate columns should be flipped. Instead, they are flipped and copied. Expected output of a after transformation is with the alternating columns as in pre-transform a[::-1, 1::2]
```
array([
       [1, 4, 3, 6, 5],
       [2, 3, 4, 5, 6],
       [3, 2, 5, 4, 7]], dtype=uint8)
```

**Additional context**
Behaviour is correct in CPython. Observed in Unix Port and ESP32 port, with -O3 and -O0

fadesibert, 2022-03-16:
workaround suggests that there is an issue with copy in place?
```
>>> out = np.zeros(a.T.shape, dtype=np.uint8)
>>> for i in range(a.T.shape[0]):
...     if not i % 2:
...         out[i] = a.T[i]
...     out[i] = a.T[i][::-1]
... 
>>> out
array([[3, 2, 1],
       [4, 3, 2],
       [5, 4, 3],
       [6, 5, 4],
       [7, 6, 5]], dtype=uint8)

```

v923z, 2022-03-16:
@fadesibert Thanks for the bug report, I'll look into it.

v923z, 2022-03-16:
> ```python
> from ulab import numpy as np
> print(np.__version__)
> ```
> 
> ulab version: '5.0.4-2D-c' numpy has no **version**

In `micropython`, `numpy` and `scipy` don't have a version number, only `ulab`.

> ```python
> import ulab
> print(ulab.__version__)
> ```

closed 2022-05-17 by v923z