v923z/micropython-ulab #520: [OPTIMISATION] Only clear ndarray data once when it's allocated

dpgeorge, 2022-04-18:
In MicroPython's memory API there are `m_new()` and `m_new0()` macros.  The latter will zero out the new memory before returning it.

MicroPython by default builds with `MICROPY_GC_CONSERVATIVE_CLEAR` enabled, which means all GC heap mallocs are zero'd out by default (to prevent stray pointers), and thus `m_new()` and `m_new0()` are the same thing.

In `ndarray_new_ndarray()` I see that there's a `memset()` called on memory returned by `m_new()`, and so this will probably be doing an unnecessary clear of the array data.

For performance reasons, I suggest using `m_new0()` instead of `m_new()`, and removing the `memset()`.  Or guarding the `memset()` in `#if !MICROPY_GC_CONSERVATIVE_CLEAR` ... `#endif`.

v923z, 2022-04-18:
@dpgeorge Many thanks for the suggestion! These are really trivial changes in the code, so I should be able to incorporate them in the next couple of days. I believe, `ndarray_new_ndarray` is not the only place, where this happens. 

On a related issue, some people suggested to get rid of `m_del`, and let the garbage collector deal with the RAM. On the one hand, I see the point, namely, that it adds to the code size, but on the other hand, too much memory might be locked up for too long. If you have any comments on the subject, I would be glad to hear them. I could even do this conditionally. (Adding `m_del`, not the listening.)

dpgeorge, 2022-04-19:
> I believe, `ndarray_new_ndarray` is not the only place, where this happens.

Yes I saw some other ones as well.

> On a related issue, some people suggested to get rid of `m_del`, and let the garbage collector deal with the RAM.

In MicroPython core, the code usually attempts to call `m_del` when possible, ie when a pointer is known to go out of scope.  This does help to free up RAM sooner rather than later and delays a GC cycle by a bit.  IMO it is worth it to call `m_del` whenever possible.  It's pretty cheap in terms of code size.

closed 2022-04-22 by v923z