v923z/micropython-ulab #523: Instructions for adding ulab to Lego Spike Prime

nw0428, 2022-05-16:
I am trying to add ulab to Lego's spike prime product. This should be possible because it uses an [STM32F413](https://www.st.com/en/microcontrollers-microprocessors/stm32f413-423.html). Is it possible to compile and add ulab without flashing the entire system?

v923z, 2022-05-16:
No, I don't think so. `ulab` is tightly integrated into the `micropython` firmware. It is really not like `python` on a PC, where you can install arbitrary libraries after the fact. I know that there are plans for making `micropython` more modular in this regard, but I have no idea, where those plans stand. 

But why is compilation and flashing a problem?

v923z, 2023-06-21:
Since there hasn't been any traffic on this thread for a while, and since this is not really a `ulab` issue, closing now.

closed 2023-06-21 by v923z