v923z/micropython-ulab #526: Cannot save and load numpy array on OpenMV4

heleidsn, 2022-05-17:
Hi, I am doing a small project with OpenMV, thanks for your effort to give us NumPy features for Microcontrollers.

However, I cannot save and load Numpy array with OpenMV which have ulab v4.0.0 with the code below. But I can find docs in the ulab book. 

The issues also raised [here](https://github.com/openmv/openmv/issues/1642). 

Could you please tell me what's wrong with it? Thanks.

### Code
``` python
import ulab
from ulab import numpy as np

a = np.array(range(25)).reshape((5, 5))
print(a)
print(ulab.__version__)
np.save('a.npy', a)
```

### Error
AttributeError: 'module' object has no attribute 'save'

v923z, 2022-05-17:
The version number only tells you, which methods _might_ be available, but everything might still be switched off. and `openmv` might have just done that, for reasons of flash space. You could find out, what you have following the description here: [finding-out-what-your-firmware-supports](https://micropython-ulab.readthedocs.io/en/latest/ulab-intro.html#finding-out-what-your-firmware-supports)

v923z, 2022-05-17:
I have just figured out, `load` and `save` was introduced in 4.3: https://github.com/v923z/micropython-ulab/blob/master/docs/ulab-change-log.md You have to pick the right version of the documentation.

heleidsn, 2022-05-17:
Hi @v923z, thanks for pointing out this, I will build the openmv firmware with the latest ulab. 

Many thanks and the issue will be closed.

closed 2022-05-17 by heleidsn