v923z/micropython-ulab #528: Unable to install micropython for rp2, cmake issues

MicahHaack, 2022-06-10:
I have been trying to follow the instructions for installing micropython for the rp2 (using ubuntu on windows subsystem), using the provided bash file found here: (https://github.com/v923z/micropython-ulab/blob/master/build/rp2.sh)

It works well until the last couple commands for cmake, where I get the following errors:

[image]

I have installed all build essentials, cmake, make, gcc, and g++, and all are up to date. I'm not sure how to fix this.

v923z, 2022-06-10:
I noticed that the latest version of `micropython` throws the compile process off. As a temporary fix, you can roll back your local `micropython` version by two, or three days. I'll try to find out, what causes the issue.

v923z, 2022-06-10:
@Gadgetoid Philip, do you happen to know what is going on here?

Gadgetoid, 2022-06-10:
This just isn't finding "make" or GCC- I'll try and run a build with the latest version and see what happens.

v923z, 2022-06-10:
https://github.com/v923z/micropython-builder didn't produce artifact in the latest run, and that's why I think that something is amiss. I just haven't had time to find out the details.

Gadgetoid, 2022-06-10:
Completely *different* error. Good fun:

```
<command-line>: error: expected identifier or '(' before numeric constant
/home/phil/Development/pico/micropython-latest/ports/rp2/build-PICO/genhdr/moduledefs.h:87:57: note: in expansion of macro 'MODULE_ULAB_ENABLED'
   87 | extern const struct _mp_obj_module_t ulab_user_cmodule, MODULE_ULAB_ENABLED;
      |                                                         ^~~~~~~~~~~~~~~~~~~
/home/phil/Development/pico/micropython-latest/py/objmodule.c:169:5: error: macro "MP_ROM_PTR" passed 2 arguments, but takes just 1
  169 |     MICROPY_REGISTERED_MODULES
      |     ^~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /home/phil/Development/pico/micropython-latest/py/mpstate.h:35,
                 from /home/phil/Development/pico/micropython-latest/py/runtime.h:29,
                 from /home/phil/Development/pico/micropython-latest/py/bc.h:30,
                 from /home/phil/Development/pico/micropython-latest/py/objmodule.c:32:
/home/phil/Development/pico/micropython-latest/py/obj.h:339: note: macro "MP_ROM_PTR" defined here
  339 | #define MP_ROM_PTR(p) (p)
      | 
In file included from /home/phil/Development/pico/micropython-latest/py/objmodule.c:40:
/home/phil/Development/pico/micropython-latest/ports/rp2/build-PICO/genhdr/moduledefs.h:89:62: error: 'MP_ROM_PTR' undeclared here (not in a function)
   89 | #define MODULE_DEF_MP_QSTR_ULAB { MP_ROM_QSTR(MP_QSTR_ulab), MP_ROM_PTR(&ulab_user_cmodule, MODULE_ULAB_ENABLED) },
      |                                                              ^~~~~~~~~~
/home/phil/Development/pico/micropython-latest/ports/rp2/build-PICO/genhdr/moduledefs.h:145:5: note: in expansion of macro 'MODULE_DEF_MP_QSTR_ULAB'
  145 |     MODULE_DEF_MP_QSTR_ULAB \
      |     ^~~~~~~~~~~~~~~~~~~~~~~
/home/phil/Development/pico/micropython-latest/py/objmodule.c:169:5: note: in expansion of macro 'MICROPY_REGISTERED_MODULES'
  169 |     MICROPY_REGISTERED_MODULES
      |     ^~~~~~~~~~~~~~~~~~~~~~~~~~
make[3]: *** [CMakeFiles/firmware.dir/build.make:1302: CMakeFiles/firmware.dir/home/phil/Development/pico/micropython-latest/py/objmodule.c.obj] Error 1
make[3]: *** Waiting for unfinished jobs....
make[2]: *** [CMakeFiles/Makefile2:1559: CMakeFiles/firmware.dir/all] Error 2
make[1]: *** [Makefile:84: all] Error 2
make: *** [Makefile:27: all] Error 2
```

I don't have a clue what's going awry here!

Gadgetoid, 2022-06-10:
Failing with our modules, too, looks like something has broken or changed in MicroPython. I'll see if I can figure out what.

AFAICT this is a totally separate issue to the mysterious missing tools that @MicahHaack is seeing, however.

v923z, 2022-06-10:
@Gadgetoid Thanks for the update!

@MicahHaack Can you compile `micropython` without `ulab`, using the script in https://github.com/micropython/micropython/blob/master/.github/workflows/ports_rp2.yml?

Gadgetoid, 2022-06-10:
@v923z ah I found the culprit - https://github.com/micropython/micropython/commit/efe23aca7154d21a9e8f283313c7a1ac29e05d86

MicahHaack, 2022-06-10:
> @Gadgetoid Thanks for the update!
> 
> @MicahHaack Can you compile `micropython` without `ulab`, using the script in https://github.com/micropython/micropython/blob/master/.github/workflows/ports_rp2.yml?

Is there a bash script somewhere? I am unfamiliar with docker and am having trouble using this.

v923z, 2022-06-10:
> Is there a bash script somewhere? I am unfamiliar with docker and am having trouble using this.

The bash script is linked in the workflow file. You don't need docker, or anything like that. You should be able to run the script locally.

MicahHaack, 2022-06-10:
> > Is there a bash script somewhere? I am unfamiliar with docker and am having trouble using this.
> 
> The bash script is linked in the workflow file. You don't need docker, or anything like that. You should be able to run the script locally.

[image]

Do you mean like this? I don't see any output so I'm not sure if this is right.

v923z, 2022-06-10:
No, I think you need 

```bash
source tools/ci.sh && ci_rp2_setup
source tools/ci.sh && ci_rp2_build
```

v923z, 2022-06-10:
OK, now you can pull in https://github.com/v923z/micropython-ulab/pull/529, and with that you should be able to compile `ulab`. I will merge the PR, once I sorted out the `circuitpython` issue.

MicahHaack, 2022-06-10:
[image]

Okay, after cloning that and running the make command I get this output. Does this look correct?

v923z, 2022-06-10:
> Okay, after cloning that and running the make command I get this output. Does this look correct?

Well, it says that the firmware has been built, so we must assume that this is correct.