v923z/micropython-ulab #530: [BUG]Can't boot on esp32c3 on micropython v1.91

coinight, 2022-07-01:
I had tried to make the firmware with ulab several times.But none is working.Mpy can be boot without ulab,but when i added USER_C_MODULES = ./ulab/code/micropython.cmake to the makefile,the compile succeed but boot failed.So i wonder how to solve the problem.
Thanks.
[image]

coinight, 2022-07-01:
the version of espidf is v4.4

v923z, 2022-07-01:
@coinight I don't have this board, but my first guess would be that the firmware might be too big for this particular chip. What you could do is reduce the number of dimensions, and switch off as many functions/features as possible: https://micropython-ulab.readthedocs.io/en/latest/ulab-intro.html#customising-the-firmware. That should significantly reduce the firmware size. If you strip everything, and leave only the FFT, `ulab` adds around 5 kB.

coinight, 2022-07-03:
Thanks a lot.I don't know this issue before.I will try your advise recently. Thanks again :)

v923z, 2022-07-14:
@coinight Can we close the issue, or do you still have difficulties with the compilation?

closed 2022-07-18 by coinight