v923z/micropython-ulab #541: circuitpython and micropython tests deliver differing results

v923z, 2022-07-26:
https://github.com/micropython/micropython/commit/f9cbe6bc47dd4f5b8e85178caecd6f0de22b4c34 modifies the formatting of certain floats, while `circuitpython` keeps the old implementation. This difference leads to differing outputs on the test scripts.

@jepler @tannewt @dhalbert What would be the preferable fix for this till `circuitpython` pulls in the new implementation? I would not like to have two tiers of tests, but I also undrestand that `ulab` should be tested against `circuitpython`, too.

dhalbert, 2022-07-26:
I think we could cherry-pick the formatting fix for 8.0.0, and ignore it for 7.3.x. It will not be an issue until we try to update ulab.

v923z, 2022-07-26:
@dhalbert Thanks for the feedback! I could take it out of the CI for the time being. We just have to make sure that we put it back, once you merged the float formatting. Would it be possible to add this to the 8.0.0 milestone, so that we don't forget about it?

dhalbert, 2022-07-26:
Oh, I didn't realize it was being tested in your CI. I wonder if you can make the tests be rather more forgiving (limiting significant digits), or are there formatting changes all over the place?

v923z, 2022-07-26:
This issue pops up, whenever the result of whatever operation is an integer float. There are several test scripts that produce that. Here is an example: https://github.com/v923z/micropython-ulab/runs/7528072036?check_suite_focus=true

dhalbert, 2022-07-26:
In that example, is `988.0000000000001` the value that CircuitPython currently produces or that the newly updated MicroPython produces?

v923z, 2022-07-26:
This is the value produced by `circuitpython`. `micropython` now prints just `998.0`.

v923z, 2022-07-26:
I haven't looked at the implementation in detail, but the commit message is 
```
Formerly, py/formatfloat would print whole numbers inaccurately with
nonzero digits beyond the decimal place.  This resulted from its strategy
of successive scaling of the argument by 0.1 which cannot be exactly
represented in floating point. ```

dhalbert, 2022-07-26:
The new fix is a good one, I just wanted to double-check.

dpgeorge, 2022-07-26:
In evaluating that improvement to formatfloat, I did run it on the ulab test suite.  IMO the output of ulab tests is much better with this formatfloat fix.

v923z, 2022-07-27:
I definitely agree, this is a significant improvement. Thanks for merging it!

v923z, 2022-08-03:
The issue is chalked up in https://github.com/adafruit/circuitpython/issues/6649, so this can be closed.

closed 2022-08-03 by v923z