v923z/micropython-ulab #543: Uploading RP2 firmware.uf2 seemingly does not work

MicahHaack, 2022-07-28:
After following the rp2 compilation steps and generating the firmware.uf2 file, upon uploading the file to my PICO, no available COM port is found.

One exception to the steps for rp2 compilation was needing to update submodules with this command:
git -submodule update --init

After that update command the firmware claimed to have built properly. However, uploading the firmware does not allow me to access the PICO at all.

v923z, 2022-07-28:
I think we should try to understand, whether the issue has anything to do with `ulab`. What happens, if you compile `micropython` without any user modules? Can you access the PICO then? Are the steps listed in this repository faulty, or the same happens, if you follow the instructions of https://github.com/micropython/micropython/tree/master/ports/rp2?

MicahHaack, 2022-07-28:
I just compiled only with micropython using the steps provided above, and after uploading the uf2 file I can see the COM port again.

v923z, 2022-07-28:
Can it be that with `ulab`, the firmware is just too big? You can reduce the number of dimensions, and take out functions to reduce the size: https://micropython-ulab.readthedocs.io/en/latest/ulab-intro.html#customising-the-firmware

MicahHaack, 2022-07-28:
I went to recompile the ulab version to check the size again and for some reason I think it works now. Shrug.
[image]
I am able to talk to my PICO and import the ulab modules. I am unsure why it didn't work before.

v923z, 2022-07-28:
It could be that didn't run `make clean` or something.

closed 2022-07-28 by v923z

v923z, 2022-07-29:
@MicahHaack It has just occurred to me that you can actually download the firmware from here: https://github.com/v923z/micropython-builder/releases