v923z/micropython-ulab #547: [BUG]

uraich, 2022-09-19:
**Describe the bug**

Latest changes in micropython break ulab (using latest version)

**To Reproduce**
Try to recompile latest micropython version with ulab included

**Additional context**

[image]

jimmo, 2022-09-19:
@uraich sorry you're a step ahead of me. I was going to send a PR to ulab tomorrow to fix this.

Did you see the build troubleshooting error that now gets emitted at the end of the compile?

jimmo, 2022-09-19:
Raised https://github.com/v923z/micropython-ulab/issues/548 with more info.

uraich, 2022-09-19:
Well, I am not always this fast! Thanks for the prompt answer. I can wait a few days until ulab will work with the new version of micropython.

v923z, 2022-09-19:
@uraich It seems to that you run into difficulties only with the `optimize` sub-module. You can undefine those functions, if you don't need them, and I believe, after that you should be able to compile.

uraich, 2022-09-19:
@jimmo said he will create a PR to correct those problems. I can easily wait a few days. I have an ESP32 with a recent micropython and ulab that works just fine. 
Thank you very much for your quick answer.

v923z, 2022-09-21:
Closed via https://github.com/v923z/micropython-ulab/pull/549

closed 2022-09-30 by v923z