v923z/micropython-ulab #548: FYI: ulab needs to be updated to upstream MicroPython mp_obj_type_t changes.

jimmo, 2022-09-19:
We just merged https://github.com/micropython/micropython/pull/8813

This will require some changes in ulab. I am happy to send a PR to do this, but won't get time until tomorrow.

Here's the overview of what needs to change:
https://github.com/micropython/micropython/wiki/Build-Troubleshooting#define_const_obj_type

v923z, 2022-09-19:
@jimmo It would actually be great, if you could take care of this, but if you are tight with time, just let me know, and I can pick this up tomorrow.

iabdalkader, 2022-09-19:
@jimmo @v923z This (breaking?) change *seems* required to build ulab with micropython upstream (or > 1.19) correct ? If so, I'd suggest that you tag/release the current ulab code/state before merging any breaking changes, for those of us that rebase on stable micropython release (the latest is 1.19) and will continue to use it for a while.

v923z, 2022-09-19:
@iabdalkader https://github.com/v923z/micropython-ulab/commit/00139bdd9684bd9d10b0d89fa8fbb0906f2d05af brings in a breaking change. I could tag the latest commit, no problem, but will that be OK with you?

iabdalkader, 2022-09-19:
Yes, anything on or after https://github.com/v923z/micropython-ulab/commit/00139bdd9684bd9d10b0d89fa8fbb0906f2d05af is fine really, I just need the `MP_REGISTER_MODULE` macro fixed, that's the only change required in 4.0.0 to update to 1.19. Hopefully nothing else needs fixing, if so, will update the submodule to whatever commit that works. Thanks!

v923z, 2022-09-19:
@iabdalkader Here you go: https://github.com/v923z/micropython-ulab/releases/tag/5.1.1

iabdalkader, 2022-09-19:
Builds fine with 1.19 thanks!

v923z, 2022-09-21:
Closed via https://github.com/v923z/micropython-ulab/pull/549

closed 2022-09-21 by v923z