v923z/micropython-ulab #552: [BUG]:   Build error with DTYPE_OBJECT flag for esp32

A622266, 2022-10-09:
Build process does not complete for esp32 if flag for DTYPE_OBJECT is set in ulab.h:
#define ULAB_HAS_DTYPE_OBJECT               (1)

Error following execution of make command:
/($BUILD_DIR)/ulab/code/ulab.c:182:5: error: expected declaration specifiers or '...' before 'ulab_dtype_type'
     ulab_dtype_type,
     ^~~~~~~~~~~~~~~
No other flags from the default ulab.h file were set for this build.

jimmo, 2022-10-09:
Sorry looks like I made a mistake in https://github.com/v923z/micropython-ulab/pull/549 and didn't catch this because it's not part of the default build configuration. See #553

A622266, 2022-10-09:
Yes I tested the fix by modifying ulab.c as indicated.  I was able to complete the build process with the flag set.  Since you made a PR, this issue can be closed.

PR #553 (2022-10-10): code/ulab.c: Fix missing comma in ulab_dtype_type definition.
Fixes #552. @A622266
closed 2022-10-09 by A622266